PlummersSoftwareLLC/NightDriverStrip #90: Unexpected behavior when effects are disabled via webserver

xcalibur839, 2021-11-18:
## Bug report

**Problem**

- Disabling an effect that is currently running does not immediately stop the effect, it will continue to run until DEFAULT_EFFECT_INTERVAL has elapsed
- If you only have one effect in AllEffects[] and you disable that effect, the LEDs will appear to freeze in whatever their final state was after DEFAULT_EFFECT_INTERVAL has elapsed. This does not happen when there are multiple effects in AllEffects[] that are all disabled, due to the fading between effects that sets all LEDs to Off/Black (i.e. the final state of the LEDs is Off/Black)

**Steps**

1.  Enable the webserver and Wi-Fi in globals.h and configure secrets.h
2.  Determine the IP address of the ESP32 and connect to the webserver with a web browser
3.  Disable all of the listed effects (or call the /disableEffect endpoint using some other means)

**Example**

curl -d 'effectIndex=0' 192.168.0.25/disableEffect

**Notes**

As of submitting this issue, the checkboxes do not currently call the /disableEffect endpoint. [PR #89](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/89) was created to fix that issue, but unless and until that PR is merged, the provided curl example can be used with the correct IP for your device to manually call the /disableEffect endpoint.

xcalibur839, 2021-11-23:
Working on [PR #93](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/93) has changed my view on this issue. For example, clicking on a disabled effect from the web page will immediately play that effect even though it is disabled. However, it will not play automatically as part of the CheckEffectTimerExpired()/NextEffect() loop.

So perhaps the first part of this issue is not actually an issue, but is working as intended. Specifically:
> Disabling an effect that is currently running does not immediately stop the effect

The effect is still disabled as far as the EffectManager auto play is concerned.

xcalibur839, 2021-11-30:
I think the biggest reason this behavior is unexpected is the lack of visual feedback for the currently running effect. When unchecking an effect in the list, there currently isn't any indication that unchecking the effect did anything until all of the effects are unchecked.

[PR #98](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/98) attempts to address this issue by providing visual feedback that unchecking an effect did register, as well as the current timeframe for the actively running effect. This behavior is very similar to something like Windows Media Player, where the currently playing song is selected, a progress bar shows how much time is remaining, and the play queue can be modified without stopping playback of the current song.

rbergen, 2021-12-06:
@xcalibur839 With #93 and #98 merged, I'm wondering what is left of this issue?

xcalibur839, 2021-12-06:
@rbergen I think you're correct and this is no longer an issue! Closing now, thank you for everything!

closed 2021-12-06 by xcalibur839