PlummersSoftwareLLC/NightDriverStrip #94: Builds started failing due to a change in

rbergen, 2021-11-25:
NightDriverStrip builds, including those in CI, started failing today. This is due to a change in version 0.9.7 of [Adafruit BusIO](https://github.com/adafruit/Adafruit_BusIO) that gets pulled into the dependency tree; it was version 0.9.6 in the last CI builds that succeeded.

The [upstream issue](adafruit/Adafruit_BusIO#69) points to not having the latest ESP32 Arduino framework, while Platform IO states everything is already up to date. [An issue](platformio/platform-espressif32#680) for this has also been opened on the PlatformIO project.

Looking at the state of affairs, I think the best choice is to sit this one out, for now.

davepl, 2021-11-26:
In order to get my checkin today to built, I specified the full path to the current BusIO until they propagate the fix, and that seems to work,

rbergen, 2021-11-27:
It looks like Bus IO 1.9.8 got picked up in the dependency tree, because I just triggered a CI run that succeeded. That means I'm closing this issue.

closed 2021-11-27 by rbergen