PlummersSoftwareLLC/NightDriverStrip #126: /getEffectList returns faulty JSON when there are many effects

laufersteppenwolf, 2022-03-13:
## Bug report

**Problem**

The problem is the following line:
https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/992a5413ff08d1ed185dae9b7e729720916ce511/include/spiffswebserver.h#L87

This initializes a new AsyncJsonResponse with 1024 bytes of JSON buffer. But if there are many effects, this buffer runs out and creates the following JSON:

` {"currentEffect":0,"millisecondsRemaining":258835,"effectInterval":300000,"enabledCount":24,"Effects":[{"name":"RainbowFill Effect","enabled":true},{"name":"RainbowFill Effect","enabled":true},{"name":"Sample Effect","enabled":true},{"name":"Sample Effect","enabled":true},{"name":"Palette Effect","enabled":true},{"name":"DoublePaletteEffect Effect","enabled":true},{"name":"Meteor Effect","enabled":true},{"name":"Magenta Twinkle Stars","enabled":true},{"name":"Blue Sparkle Stars","enabled":true},{"name":"Red Twinkle Stars","enabled":true},{"name":"Lava Stars","enabled":true},{"name":"Rainbow Twinkle Stars","enabled":true},{"name":"Little Blooming Rainbow Stars","enabled":true},{"name":"Big Blooming Rainbow Stars","enabled":true},{"name":"Neon Bars","enabled":true},{"name":"VU Effect","enabled":true},{"name":"VU Effect","enabled":true},{"name":"VU Effect","enabled":true},{"name":"VU Effect","enabled":true},{}]}`

Manually setting the buffer to e.g. 2048 bytes fixes the issue, although this is just a matter of time until this buffer size is gonna be too small again.

What are your thoughts on how to handle this? Is it worth the effort and calculating the buffer size beforehand, or would you just set the buffer size to a reasonable big size?

**Steps**

1.  Build the Brooklynroom config with the webserver enabled
2.  Make the web request
3.  

**Example**

The Brooklynroom config already comes with this issue

**Notes**

My temporary "fix" was to change this line
`AsyncJsonResponse * response = new AsyncJsonResponse();`
to
`AsyncJsonResponse * response = new AsyncJsonResponse(false, 2048U);`

rbergen, 2022-03-14:
If I read the documentation of `DynamicJsonDocument` correctly, what may be behind this is that at some point, support was dropped for dynamically growing the buffer as required by the document's contents. I think increasing the buffer size to a value > 1024 makes sense in any case. Maybe that value should be 2048, maybe it should be something else. In my opinion, this should be combined with a check if buffer overflow happens while adding the effects. This would require a transition from the use of chained `operator[]` invocations to adding `"effect"` `JsonDocument` instances to the main document using `add()`, in the effects adding loop. If a memory overflow situation is detected (by `add()` returning `false`), there are two things I can imagine happening:
- `GetEffectListText` restarting itself from scratch, but with a larger buffer. This is inefficient if overflow occurs, but doesn't require changes at the API consumer's side. We could consider indicating somewhere (logging? response?) that overflows start occurring, and use that as a trigger to review the buffer size.
- Returning partial results. `GetEffectListText` would then have to indicate if the effect list it returns includes the last effect in the overall list. The caller would need to check for this flag, and then be able to indicate it wants to "skip" the first `n` effects when calling the endpoint if an incomplete list is returned. The benefit of this approach is that memory use is predictable, the downside is that multiple `GetEffectListText` calls may be necessary.

laufersteppenwolf, 2022-03-14:
Just gave it a try, works fine

You have my +1 for the PR

PR #127 (2022-03-17): Detect and address JSON buffer overflow in CSPIFFSWebServer::GetEffectListText()
## Description

This fixes #126. Testing of this has been done by @laufersteppenwolf.

Also, it actively untracks `include/secrets.h`. When pulling `main` after this PR is merged, this can lead to a one-time occurrence of either:
1. the file being deleted locally
2. a merge conflict occurring. This is actually the likely scenario if the "default" secrets.h was modified.

In either case, if a new `secrets.h` is subsequently created locally then it will be covered by `.gitignore` and left alone by git.

## Contribut [... cut]
closed 2022-03-17 by davepl