PlummersSoftwareLLC/NightDriverStrip #193: I bought One of the Heltec-Wifi Kit32 Its the on that popped up when i clicked the amazon link in Video Description however its not a v2 its a v3 with oled

gothicdude1044, 2023-01-07:
## Bug report

**Problem**

The Code is not compatible with latest heltec v3 wifi kit 32 it fails trying to upload filesystem Pops up an error this chip is ESP32-S3 Not ESP-32

**Steps**

1.  BUild FS via Platform io Heltec v2 demo 
2.  Try to upload
3.  

**Example**

**Notes**

gothicdude1044, 2023-01-08:
Can you give me an Example ive tried using
[env:heltecv2demo]                     ; V2 board has 8M flash
board = heltec_wifi_kit_32_v2
board_build.mcu = esp32s3
monitor_speed = 115200
upload_speed  = 921600
upload_port   = COM9
monitor_port  = COM9
build_flags   = -DDEMO=1
                -DUSE_SCREEN=1
                -DARDUINO_HELTEC_WIFI_KIT_32=1                
                -Dregister=                         ; Sinister:  redefine 'register' so FastLED can use that keyword under C++17
                -std=gnu++17
                -Ofast 
which i pulled from the 
[env:panlee]
board         = esp32-s3-devkitc-1
board_build.mcu = esp32s3

smerkadidit, 2023-01-16:
I ran into this same issue and have been trying to figure out the setting changes to make it work.

smerkadidit, 2023-01-20:
Dave, I found out the V3 is now supported in ESP32 but not in platformio yet. 

I linked the libraries from github that do support it and got to this point code wise... which builds and uploads fine now although i did have to disable the audio function... 

unfortunately, I also found an issue with the updated arduino-esp32 framework (https://github.com/espressif/arduino-esp32)  in the pins_arduino.h file (the pinout is not correct, its copied from the old v2 chip and not updated for the v3 pinout) so I put in a issue about that and im trying to figure out the mapping. 

I am pretty new to this platform, so I was wondering if maybe you could provide some help/input/feedback on what I have done so far and maybe help me figure out the proper mapping in the pins_ arduino.h file for the V3?

Here is the issue post I put in, and i pasted the pins_arduino.h file i was trying to make work here. 

https://github.com/espressif/arduino-esp32/issues/7737#issuecomment-1398298658

Here is the code I used to make the board itself work and compile with nightdriver/platformio

[heltec_v3_strip]
lib_deps = 
	${common.lib_deps}
	https://github.com/Baptou88/platform-espressif32.git

[env:heltec_v3_strip]
platform = espressif32
platform_packages =
  ; use upstream Git version
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = heltec_wifi_kit_32_v3
monitor_speed = 115200
upload_speed = 921600
upload_port = COM9
build_flags = 
	-DLEDSTRIP=1
	-DUSE_SCREEN=1
	-DARDUINO_HELTEC_WIFI_KIT_32=1
	-DUNITY_INCLUDE_DOUBLE
	-DUNITY_DOUBLE_PRECISION=1e-12
	-D_IR_ENABLE_DEFAULT_=false
	-DDECODE_NEC=true
	-DUSE_PSRAM=1
	-DBOARD_HAS_PSRAM=1
	-mfix-esp32-psram-cache-issue
	-std=gnu++17
	-Dregister=
	-Os

davepl, 2023-01-20:
Wow, unfortunately you’re above my knowledge level on the platform SDK!  I’ve never tried to remap pins, so you’re in uncharted territory for me!

If you do get a successful config, please do be sure to check in a config section or send it to me if you want me to do it (but I don’t have a V3 to test with).  It’d be nice to get it in there!

Cheers,
Dave

> On Jan 20, 2023, at 8:45 AM, smerkadidit ***@***.***> wrote:
> 
> I am pretty new to this platform, so I was wondering if maybe you could provide some help/input/feedback on what I have done so far and maybe help me figure out the proper mapping in the pins_ arduino.h file for the V3?

mellowmunki, 2023-01-20:
I would be more than happy to test this on a V3 as I also recently bought an M5 Plus and a Heltec Wifi Kit 32 V3 and was also attempting to build the scripts for the V3 last night with no luck.

I was able to communicate and build on the V3 using ArduinoIDE by adding the following into the Additional Boards Manager URLs in Preferences: `https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json`
This then allowed me to install the "Heltec ESP32 Series Dev-boards" package by Heltec Automation(TM) in the Boards Manager and was then able to select the board from Tools > Board > Heltec ESP32 Series Dev-boards > WiFi Kit 32(V3).

Unfortunately it seems that the Heltec ESP32 Series Dev-boards package is not currently available via PlatformIO but I wonder if they have the correct pinout compared to the `espressif/arduino-esp32.git`

Either way I would be more than happy to test anything you may have @smerkadidit and will attempt to build your above example later today for testing.

smerkadidit, 2023-01-20:
> I would be more than happy to test this on a V3 as I also recently bought an M5 Plus and a Heltec Wifi Kit 32 V3 and was also attempting to build the scripts for the V3 last night with no luck.
>
> I [...]

The code and changes to run in nightdriver works for sure, I did test that aspect. Just make sure to disable the audio flags too i didnt get to troubleshooting that part.

Its more less the pinout portion, and if you look at the link I posted some of the progress i made on figuring it out (i believe everything i posted in that pins_arduino.h section is correct but i need to test it)    I posted links to the schematics and pinouts for the old and new boards as well as what needs to be updated. 

I took alot of programming, electronics, and sys engineering courses in college but havent used that part of my brain in a bit lol this is my first big undertaking in trying to make something like this work. Ill have plenty of free time though, cause like Dave ill be diving into this heavily after my shoulder surgery tuesday with all the free time being layed up

davepl, 2023-01-20:
Now I’m a bit lost, as the Heltec Wifi Kit 32 was the first module I ever used!  So I *know* that works!  Which variant are you having issues with?  

- Dave

smerkadidit, 2023-01-20:
Oh its surely the Heltec Wifi Kit 32 v3 that im working with. 

It seems as if they just copied the v1 or v2 pins_arduino.h file for the v3 and thats what I still need to iron out. 

I got the framework and esp32 files working with that board outside of that part and and the audio flag (for the V3) so it will now build/upload to the v3 but since the pins file itself is still wrong, the chip wont function properly. Ill be testing my updated pins file when I get home and keep you posted on the progress.

smerkadidit, 2023-01-20:
> 

@mellowmunki the pins_arduino.h file for the heltec_wifi_kit_32_v3 is incorrect in the  espressif/arduino-esp32.git file also, it looks to be just a copy of the v2 and since the pinouts change it doesnt work right, which is the issue I raised there and then began to try fixing it by looking at the schematics and pinout

mellowmunki, 2023-01-21:
@smerkadidit I tested your code and it does build, although I had to uninstall the Espressif 32 platform, then run the Build with your code so that it pulled down the files from `https://github.com/Baptou88/platform-espressif32.git`.

I noticed you had raised the issue on the espressif/arduino-esp32 repository, you are doing a great job at trying to work out all the pinouts if this is all so new to you! As you say, the pinouts do seem to be all out of whack as there is no output to the OLED or to the 144 LED Strip I have plugged in when uploading the `heltec_v3_strip` code you created.

Wishing you all the best for your upcoming surgery!

Massive thanks go out to @davepl for creating and sharing this project, it is a lot of fun and I hope to be able to one day provide some sort of input into the project. I am VERY new to coding but not new to tech so hope to pick it up quite quickly.
I stumbled on Dave's Garage early December 2022 via the "What do LEDs sound like?" video followed by "NEW LED Project: Learn Arduino, ESP32 and More!". I immediately did as anyone would having watched a Dave video... liked & subscribed :) I then fell into the "Microsoft Worm-hole" and thoroughly enjoyed the history of Dave and his past projects and work. I then got stuck in to the LED/ESP32 videos and had been pining for a board to play with since Christmas, just this week I picked up a M5StickC Plus and the Heltec WiFi Kit 32 V3.

I have so far only managed to get Spectrum and the amazing new Laserline (thank you Dave!) working on the M5Stick but hope to get my head around how to make the other routines work too by going back over Dave's videos to refresh my memory on how to configure the settings for each board for each routine.

The Heltec WiFi Kit 32 V3 has proven to be a tricky one to master and I have only been able to so far get it to function with the Arduino IDE software with the `https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json` url added to the "Additional boards manager URLs" in Preferences and then installed the "Heltec ESP32 Series Dev-boards" package by Heltec Automation(TM) in the Boards Manager.
Then I was able to select a working board configuration via Tools > Board > Heltec ESP32 Series Dev-boards > WiFi Kit 32(V3).

I mention this in case it helps Dave or anyone else when it comes to debugging/configuring this board.

A side note about the packages available...
esp32 from `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json` includes the V3 board and works with FastLed examples within Arduino IDE.
Heltec ESP 32 Series Dev-boards from `https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json` also includes the V3 board and works with FastLed examples within Arduino IDE, but, also adds an extra "Heltec-Example" menu in the Examples section of Arduino IDE which has a Factory Test which enables and works with the screen. I was not able to find the same test with the standard esp32 package.

`https://dl.espressif.com/dl/package_esp32_index.json` does not contain the V3 board at all.

I hope some of this info is handy in some way. I did try adding a git link to the Heltec ESP 32 Series Dev-boards in PlatformIO.ini in VS Code under lib_deps but it fails to build due to a missing json file.

For reference the board that I purchased was this one: [Diymore ESP32 Development Board WIFI Kit 32 ESP32 WIFI Wireless with 0.96 inch OLED Display CP2102 Type C](https://amzn.eu/d/b0570DM)

mellowmunki, 2023-01-22:
I have some good news in that I have managed to get the V3 board working and running Dave's Demo script. I have not yet had a chance to test others but wanted to let you know that thanks to the pinout adjustments by @smerkadidit I was able to get the board working :)

I decided to fork the `https://github.com/espressif/arduino-esp32.git` to my own repo and then applied the pinout changes to the `pins_arduino.h` within my forked version. I was then able to apply my own repo version via the `platform-packages` variable instead of the espressif/arduino-esp32.git that had the incorrect pinouts.

If you would like to test then please do so, I have only tested the demo so far but here is my working code for the platformio.ini
```
; Libs needed to link with an Heltec Wifi Kit 32 V3
[wifi_kit_32_v3]
lib_deps = ${common.lib_deps}
	https://github.com/Baptou88/platform-espressif32.git
```
```
[env:heltec_v3_demo]
platform = https://github.com/Baptou88/platform-espressif32.git
board = heltec_wifi_kit_32_v3
platform_packages = 
  framework-arduinoespressif32@https://github.com/mellowmunki/arduino-esp32.git
monitor_speed = 115200
upload_speed = 921600
upload_port = COM7
build_flags = -DDEMO=1
	-std=gnu++17
	-Dregister=
	-Ofast
lib_deps = ${wifi_kit_32_v3.lib_deps}
```
I have no idea if this is the most optimised way to have coded the board variables but it seems to build, upload and run on my V3 board.

I have also tested Dave's script from [04-FastLED with Arduino Tutorial Introduction](https://www.youtube.com/watch?v=aub9PecrbuM) and with OLED_CLOCK, OLED_DATA and OLED_RESET pins set to match those in your revised pinout the script builds and uploads to the V3 board without issue and the screen correctly displays FPS and the connected LED strip lights up correctly too.

smerkadidit, 2023-01-25:
[image]

any ideas why this youtube lib is causing me errors?

mellowmunki, 2023-01-27:
> [image]
> 
> any ideas why this youtube lib is causing me errors?

I don't know what you have set in your platformio.ini file but my guess is that you need to add the following to the `lib_deps` section: `https://github.com/Baptou88/platform-espressif32.git`

The default espressif32 library doesn't contain the files to support the esp32s3 so you will need to add the above to pull the correct files.

mellowmunki, 2023-01-27:
Curiosity got the better of me so I decided to setup a VM and create a clean VS Code and PlatformIO environment to ensure my installation wasn't pulling from pre-installed libraries that had been pulled via other scripts.

The following is code that when inserted into Dave's platformio.ini file from scratch seems to compile and build cleanly.
I did have to disable 2 of the build flags and also had to change the last flag from `0s` to `0fast`
(I have not had a chance to test uploading as I am not near an ESP32 or LED strip currently to be able to test upload).

```
; Libs needed to link with an Heltec Wifi Kit 32 V3
[wifi_kit_32_v3]
lib_deps = ${common.lib_deps}
		   https://github.com/espressif/arduino-esp32.git
```
```
; Ledstrip script for Heltec WiFi Kit 32 V3 boards
[env:ledstrip]
platform      = https://github.com/Baptou88/platform-espressif32.git    ; Override the default platform files as it does not contain the V3
board         = heltec_wifi_kit_32_v3
platform_packages = 
  framework-arduinoespressif32@https://github.com/mellowmunki/arduino-esp32.git   ; Override the default framework as it does not contain the V3
monitor_speed = 115200
upload_speed  = 921600
upload_port   = COM7
build_flags   = -DLEDSTRIP=1
                -DUSE_SCREEN=1
                -DARDUINO_HELTEC_WIFI_KIT_32=1
                -std=gnu++17
#                -DUNITY_INCLUDE_DOUBLE
#                -DUNITY_DOUBLE_PRECISION=1e-12      ; Make doubles real 8 byte doubles
                -Dregister=                         ; Sinister:  redefined 'register' so FastLED can use that keyword under C++17
                -Ofast
lib_deps      = ${wifi_kit_32_v3.lib_deps}

```
Hope that helps @smerkadidit

amosk, 2023-04-02:
I was able to get it compile and upload using @mellowmunki code (thank you!) by changing _common_ to _base_ and removing _upload_port_.

`
; Libs needed to link with an Heltec Wifi Kit 32 V3
[wifi_kit_32_v3]
lib_deps = ${base.lib_deps}
		   https://github.com/espressif/arduino-esp32.git
`

Side note: my AV flagged github.com/mellowmunk/arduino-esp32/tree/master/tools/gen_esp32part.exe. 
https://www.virustotal.com/gui/file/b5d5e0e6e5fe8d2ec9509a53b719d00059f29bcf9f15d4bc199c48bf92df5fef/detection