PlummersSoftwareLLC/NightDriverStrip #202: TTGO section in Effects.cpp

DanMan32, 2023-01-23:
## Bug report

**Problem**
When using environment TTGO, compilation of effects.cpp generates an error.
Line calling SpectrumAnalyzerEffect is missing int parameter specifying number of spectrum columns

**Notes**
I was able to fix the issue in my cloned copy but not familiar enough on how to update Git repository without clobbering any changes since my cloning.

rbergen, 2023-01-23:
Thanks for the report. May I suggest you do try to create a PR to fix this? When you've done it once, you're also setup to make other contributions, if that becomes opportune.

The best way to go about it is working from a fork, and potentially apply your change in a separate bugfix branch in that. The GitHub documentation Quickstart section contains a pretty easy to follow discussion of the concepts and the steps to take: <https://docs.github.com/en/get-started/quickstart/fork-a-repo>.

rbergen, 2023-02-03:
Fixed with commit [a22e57cdc54cf7a029dbf22b175acc787aeacd99](https://github.com/PlummersSoftwareLLC/NightDriverStrip/commit/a22e57cdc54cf7a029dbf22b175acc787aeacd99).

closed 2023-02-03 by rbergen