PlummersSoftwareLLC/NightDriverStrip #255: Splash screen does not time out

davepl, 2023-04-29:
## Bug report

**Problem**
After flashing my board, the splash logo does not time out in a few seconds. If I erase_flash and then reprogram, it fixes it.  But it recurs.

**Steps**
1. Upload firmware, use it 
2.  ???
3.  Upload firmware
4. Note that splash screen doesn't time out. 

I wonder if perhaps the full interval is being stored for it?  There's some special logic that allows an effect to set a max duration, and perhaps that's being ignored?

If there's a way for me to fetch the JSON from SPIFFs, let me know.  If not, consider adding a "/dumpjson" REST call to do just that in order to ease future debugging?

rbergen, 2023-04-29:
Adding an endpoint to retrieve the effects config JSON is a good idea. I'll do that.

In the meantime, if the problem recurs, could you share a screenshot of the device web interface's effects page? I mean something like this: 
[image]

It will already tell what the effect interval and enabled state of each of the effects is.

rbergen, 2023-04-30:
@davepl What could have happened is that you pressed the ON button on the remote. That sets the effect interval to 0 as per [line 217 of remotecontrol.h](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/4e37e5c30a6be92a1bfb71fc23a6b90c16f72330/include/remotecontrol.h#LL217C13-L217C13). The effect interval is then stored in the effect config JSON. 

If the effect interval is 0, the effect is shown indefinitely, and any max duration returned by the effect is indeed ignored. This is done by the [`GetInterval()` function in effectmanager.h](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/4e37e5c30a6be92a1bfb71fc23a6b90c16f72330/include/effectmanager.h#LL461C1-L468C6).

In my view there are two ways to fix this:

1. Not setting the effect interval to 0 when processing a press of ON on the IR remote.
2. Not ignoring an effect's max duration when the effect interval is 0.

As the current behavior was not implemented by me, I'm not sure which option to pick.

On top of that, it would probably be better if the current effect index is also persisted to and restored from JSON; that would make that the device resumes on the same effect it was on before the most recent flash/reboot. I will implement this when adding the aforementioned JSON dump endpoint.

PR #254 (2023-05-07): Refactor webserver and device config, change WiFi connect (wait) logic
## Description

This:

- Refactors the webserver and device config class to reduce duplication of code
- Modifies the WiFi connect (wait) logic, to allow for connections taking longer to settle
- Shows the splash screen effect at startup, regardless of the effect list loaded from JSON or by default
- Incorporates the changes of #257, and thus also fixes #255
- Made the Weather effect update task persistent, which fixes a memory leak
- Uses PSRAM for JsonDocument instances
- Preloads static device stats in the webserver, pre [... cut]
commit a9714eeb (2023-05-07): Fix #255
closed 2023-05-07 by rbergen