PlummersSoftwareLLC/NightDriverStrip #256: Fix and remove weather images

davepl, 2023-04-29:
## Bug report
We need a good set of open-source weather images from somewhere, or someone needs to draw a consistent set.  Then remove and unused images from bmp/data

VeniceInventors, 2023-05-28:
> will this project work for weather icon replacements?

Not directly, the icons from your link actually are monochrome glyphs from TrueType fonts, the weather images used in this project are color bitmaps.

I'll create a set of bitmaps and post it here soon.

Fingez, 2023-05-29:
[16 X 16 Weather Icons.zip](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/11593880/16.X.16.Weather.Icons.zip)
 
I had a crack at making some

davepl, 2023-05-29:
I think these are great, but I was wondering if you'd consider doing a second set with color for the daytime, and we could use these for evening?  I think they look great, but want to take advantage of the RGB screen as well!

Fingez, 2023-05-30:
[16 x 16 Weather Icons colorized.zip](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/11595262/16.x.16.Weather.Icons.colorized.zip)

I originally thought the 'Monochrome' would help in making the icons more clear, but I was clearly wrong.  The subtle color certainly helps differentiate the snow from the rain for example.  This set has inconsitencies between each icon, but I left them that way for now to see if anyone prefers the feel of a particular one, I can then go and fix the others to match better.

davepl, 2023-05-31:
I like them a lot, but any chance you’d consider using color?  Or if you’re willing to do both, we could use these as the “dark” night theme and the colored ones for daytime.  That’d make the sun yellow, the snow light blue, and so on…

Let me know what you think!
-
 Dave

VeniceInventors, 2023-06-01:
[weather.zip](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/11624133/weather.zip)
Here a set I made in Paint.net with some extra icons. 
I'm not sure how they'll look on LEDs, hopefully there's no flickering from PWM with lower intensity shades.

Fingez, 2023-06-01:
> I like them a lot, but any chance you’d consider using color? Or if you’re willing to do both, we could use these as the “dark” night theme and the colored ones for daytime. That’d make the sun yellow, the snow light blue, and so on… Let me know what you think! - Dave
> […](#)

I attached the colored set on last message, unless you mean you would like them even more colorful? There is only a few colors you can relate to weather though. :D

mggates39, 2023-11-06:
I think I will take the are that @VeniceInventors posted here a couple of months ago and get them into the effect.  I will see if I can get day and night too as those images are mostly there too.

Marshall

mggates39, 2023-11-06:
Can someone pull this branch https://github.com/mggates39/NightDriverStrip/tree/feature/New-Weather-Icons and see if it looks okay.  It is going to be this weekend before I have anything cobbled together to try to look at it.

rbergen, 2023-11-13:
@mggates39 I only just noticed this, which is the reason it took a while before I responded. 

I have checked out the branch and looked at the code, which already raises an interesting question, which is what is the definition of "tomorrow". This question is triggered by the fact you propose adding night-time icons, which I intuitively feel doesn't make sense for the "tomorrow" scenario. That is to say that I think the "tomorrow" part of the panel should show the weather for tomorrow **day time**, but I'm not sure exactly what time of the day to pick/aim for. It also means the current logic to extract the weather for tomorrow is quite possibly flawed: it picks the first weather entry that has tomorrow's date attached to it, which may well be a slot in the middle of the night. 

On the other hand, showing night-time icons for the _current_ weather does seem sensible.

I'm going to have to let this simmer for a bit to figure out what time of day is the most logical one to show tomorrow's weather for. I'll report back here when I've come to an opinion about this.

rbergen, 2023-11-13:
@mggates39 Ok, so after giving it some thought and looking at the OpenWeatherMap API, I'd say the following:

- Let's pick the weather information for tomorrow's weather on or as closely after "tomorrow 3PM" (YYYY-MM-DD 15:00:00) as possible. Generally speaking, that should put us close to the maximum temperature for the day, which I think is what most people will be interested in.  
  Doing this is a bit tricky, as we want the weather for around 3PM _local time_ and the timestamps in the OpenWeatherMap responses are in UTC. That means we need to figure out what "tomorrow 3PM local time" is in UTC `time_t` seconds. I think the best way to do this is using `time()` and `localtime()` to get the current local time, add a day and set the time to 3PM in the `tm` struct that yields, and `mktime()` that into the `time_t` timestamp we need. When we have that we can pick the first weather element in the OpenWeatherMap forecast response that has a `dt` field greater than or equal to that value.  
  This automatically means that all weather images for tomorrow's weather _should_ be daytime images.
- If we're actually adding night-time images to the set then I think we should just change the `weatherIcons` map to use `String` keys - those being the `"<int>d"` or `"<int>n"` values we find in the OpenWeatherMap responses' `icon` property. It'll make the icon data lookup much more straightforward.

I'd propose we make these changes before I start looking at the actual appearance of the weather image set; at least for me it'll be the comprehensive visualization that I'll base my conclusions on.

mggates39, 2023-11-13:
@rbergen That sounds like a solid plan.  I will start on that tonight.  

- My initial take on the the API was that it returned the high/lo for the day, but it appears to be for the 3 hour block.  Maybe we scan the whole set of blocks for the day and get the absolute min and max for the day.
- I have all the day/night images up now.  I am not familiar with the map that is currently being used so I will have to study it before I go to a string index.  I tried once and it failed to compile hence what I did with the offset in the current code to get it running.

Also, the hardware I plan on using only has 4MB of  Flash and even with the web server disabled I can't get it to fit so I have not been able to test it yet.

rbergen, 2023-11-14:
> I got it down to 4MB and can run it now.

Glad to hear that!

> I have it going through all the record for the next day. I do that by taking the UTC epoch time as local time to compare to the date of tomorrow. I will probably average out the weather icons to get a good one for the day and force it to day time.

I like the idea, but I do wonder how you plan to "average out" weather icons? I live in a country where "four seasons in one day" is a thing, so I genuinely wouldn't know how to capture that in one icon. That's part of why I suggested we pick a particular time as "tomorrow's" weather; it may not always represent the day, but at least it's consistent concerning what part of the day it does show.
 
> I should be able to get that buttoned up in the next day or so. I am going through Service Now Admin Training at work and it is taking up some of my spare time this week.

It's a small world once again. My current CEO used to be the Service Now CMO for the EMEA region.

mggates39, 2023-11-15:
My thought was to count the number of each of the icons, just using the number ignoring the day/night indicator and get the one with the most hits.  Though  always taking the icon for the noontime block would be easier.

rbergen, 2023-11-15:
Yeah, that might work. We could just give that a try, monitor what it yields on the matrix for a while and then see if we feel it's so far off we need to reconsider.

mggates39, 2023-11-15:
Okay, I have the 'final'  code up on my branch.  I have pulled out my debugs and cleaned up the comments.  It should be good to go now.  I WILL get my matrix hooked up this weekend to look at it too.

Thank you for all your feedback.

mggates39, 2023-11-16:
Based on the images from https://github.com/PlummersSoftwareLLC/NightDriverStrip/discussions/483#discussioncomment-7590878 of my code running on the prototype, if no one has any objections I will put in a pull request this evening when I get home.

rbergen, 2023-11-16:
As far as I'm concerned there will never be an objection to any PR being opened that is rooted in a genuine intent to contribute something useful. It's also when I - and I am only talking for myself - _really_ start looking at the code (change) that's proposed. Any discussion and polishing that may lead to is just part of the process.

rbergen, 2023-11-23:
The old weather icons have been replaced through the merge of #537. As @mggates39 intends to apply further improvements to the weather visualization I will keep this open to track those. Once all _visualization_ PRs in the (upcoming) sequence have been merged I'll close this, even though further improvements of a different nature to the Weather effect may follow after that.

mggates39, 2023-11-25:
I think the next Pull Request with fix the issue with not properly getting the tomorrow's temperature range and weather icon.  I will also take this opportunity to convert to std::chrono for the time base as well as fix a few debug messages severities.

mggates39, 2023-12-03:
I believe I have made the final Pull Request #558  for this Issue.  I have tied up all my loose ends and put in what I think is a good set of comments for all the methods in the class.  As always, I welcome any and all feedback.

mggates39, 2023-12-08:
@davepl if you are good with these changes, I think this issue can be closed.

rbergen, 2023-12-08:
@mggates39 I'm going to close this now. I thought the merge of #558 would close this automatically, but I now see the phrasing used in the reference from that PR to this issue doesn't use the exact wording GitHub is looking for (in case you're interested: you put the word "issue" between "close" and the issue number, which breaks the link).

Thanks for pointing out this issue is still open.

closed 2023-12-08 by rbergen