PlummersSoftwareLLC/NightDriverStrip #268: Missing tomorrow forecast

davepl, 2023-05-16:
## Bug report
Often times I get today's weather but tomorrow's stays all zeros.  They're separate, but not sure why one would succeed and the other would fail!

When it happens, I get the local town, temp, and today's conditions icon and the high and low for today.  For tomorrow, just zeros.

rbergen, 2023-05-16:
I apologize for the painfully obvious question, but did you happen to catch any logging when this happens? I think I remember this happening when the board time was off (due to NTP synchronization not working), but I'm not entirely sure.

davepl, 2023-05-21:
No, I didn't see anything remarkable in the logs... but it seems to have been resolved.  If it recurs, I'll re-open!

closed 2023-05-21 by davepl