PlummersSoftwareLLC/NightDriverStrip #272: Incorporate effects from Animatrix

davepl, 2023-05-19:
## Bug report

There are a number of interesting matrix effects to be found here:

https://github.com/StefanPetrick/animartrix/blob/main/animation_collection.ino

I think we should port some of the best ones to NightDriver.   They would need to be wrapped in LEDStripEffect, and the oscillators likely need to go in GFXBase.

davepl, 2023-05-24:
Agree 100%... my bad for not having read the CCA license more closely before making the suggestion.  Appreciate you catching it!

Bun-Bun, 2023-05-31:
This is something I would be interested in looking at. There are some neat effects in his library. 

As far as the licensing goes, what is the best way to handle "learn from https://fabiensanglard.net/doom_fire_psx/ - don't copy and paste from it."? Do you give credit for inspiration in comments of the code?

Bun-Bun, 2023-05-31:
Yes, I understand that. But does one give any credit where the inspiration for the mathematics came from? Or just silently take said inspiration?

Bun-Bun, 2023-05-31:
Perfect. That makes sense to me and will do the same.

Bun-Bun, 2023-05-31:
Thank you for that expansion of what you said. Makes sense to me. 

I think the process of bringing the ideas over to this project (as davepl pointed out, it can't be directly included anyway due to the differing structure of the projects) as well as the process of learning and experimentation of the math/ideas would take care of any copyright issues. 

Ethically it makes sense to me to give credit where inspirations came from. See it all the time in the music world but then you do see instances where some are still offended and seek legal action. My coding experience is all from a purely tinkering/hobbyist or 100% private business. Seeing others insights on the licensing aspect is valuable. Personally I'd be glad if someone learned something from my code and was inspired to create something the same or similar and would appreciate credit.

netmindz, 2024-07-28:
If you use my fork https://github.com/netmindz/animartrix then it becomes trivial to wire the effects to different output formats, e.g https://github.com/netmindz/animartrix/blob/OO/examples/ANIMartRIX_FastLED/ANIMartRIX_FastLED.ino or https://github.com/MoonModules/WLED/blob/0562debbe1670fc72971d95da296cc9fb53f5f9a/usermods/usermod_v2_animartrix/usermod_v2_animartrix.h#L91