PlummersSoftwareLLC/NightDriverStrip #273: Subscriber effect blocks draw thread

davepl, 2023-05-21:
## Bug report
When the API request fails, it blocks the main draw thread for about a second, pausing drawing

Not sure if this a priority issue or if there just isn't CPU time for everything that needs to happen, but at a minimum we should "grow" the delay between attempts so that its not several times per minute if it's failing.

rbergen, 2023-06-27:
I've taken another look at the YouTubeSight code, and it basically does an HTTP GET for a straightforward URL. It then picks some values out of the comma-separated list of numbers that it gets back in the response. My first step would be replacing YouTubeSight with code that uses HTTPClient, as I already hinted to. If that doesn't stop the blocks-on-error then I'd look into implementing some form of exponential backoff. 

I'm happy to pick this one up once I again have a board I can test with.

davepl, 2023-06-27:
It would be nice if it just loaded the channel page, scraped the sub count, and didn’t use the API at all, since to track any channel you need a different API key, it seems.  No idea how involved that would be!

- Dave

rbergen, 2023-06-27:
I understand your thinking. A few thoughts off the top of my head:

- Most providers of "free" services include clauses in their T&Cs that prohibit scraping. We'd have to see to what extent that applies to YouTube - and then permanently keep an eye on T&C changes YouTube may make later if they now do allow it.
- Front-end scaping based solutions are brittle, in the sense that if YouTube changes their front-end at any time, our scraping logic may break as well.
- I think the subscriber number in the channel page is "such and such K", not the exact number. This is something I'd have to check, though.
- The channel page (even just the HTML) is quite a bit more data than the API response. This means we may run into memory issues on the ESP32 while we retrieve and parse it. 

I'll be honest: the combination of all this makes that personally, I'm not a fan of the idea.

rbergen, 2023-06-28:
I checked a couple of things, and found the following:

- The YouTube terms of service state that "You are not allowed to access the Service using any automated means (such as robots, botnets or scrapers) except: (a) in the case of public search engines, in accordance with YouTube’s robots.txt file; (b) with YouTube’s prior written permission; or (c) as permitted by applicable law". US legal jurisprudence seems to permit scraping of public information as long as it doesn't include PII (which we wouldn't) and doesn't hurt the operation of YouTube (which I'd consider highly unlikely). So, we should be okay there, in theory.
- The channel page does only include the subscriber count in thousands.

davepl, 2023-06-28:
What about social blade?  They may just interpolate, but seem to give you a live count for other channels…

- Dave

rbergen, 2023-06-28:
Never heard of them, but will look into it when I'm back from Germany. :)

rbergen, 2023-07-02:
So I took a look at Social Blade, and it looks to me like they take whatever the "something something" K is YouTube is showing, and replace the "K" with ",000" - at least for your channel. That we could do ourselves, I think.

That said, I've been chewing on this for a bit over the past week, and I'm just not going to enjoy working on a scraper for any website. I've been involved with it in the work context for a number of years, and I've always found it to be a world of pain in the end. I'll try to fix the effect stutter as I said when I have a board I can test with, but the scraping part I'm going to have to leave to someone else.

davepl, 2023-07-02:
Well, the only good thing about the social blade is that if you watch the live count it “simulates” a live count by interpolating upwards, so at least the display would change over time instead of just showing 437,000 and then 438,000.

If we can fix the pause/stutter without a big change, please do so, otherwise assign an issue to me and I’ll do the scraping!

Thanks,
Dave

rbergen, 2023-07-02:
I feel confident I can fix the pause with a pretty straightforward rewrite of the code that's now in YouTubeSight, but I'm well aware that a statement like that is typical "famous last words" material. 

I'll try, test and hopefully confirm when testing is possible.

Cheers,
Rutger