PlummersSoftwareLLC/NightDriverStrip #277: Flashing on ESP32 S3 doesn't work. invalid header: 0xffffffff

glowingkitty, 2023-05-22:
Just tested flashing the NightDriverStrip firmware via https://plummerssoftwarellc.github.io/NightDriverStrip/
And while the flashing itself works, the firmware cannot be loaded. VScode shows as the serial output only `invalid header: 0xffffffff`

rbergen, 2023-05-22:
@glowingkitty What device and project were you flashing?

glowingkitty, 2023-05-22:
I tried to flash for testing purposes the Panlee project on my esp32 s3 based PCB (selected S3 devkit c1 in the Settings).

glowingkitty, 2023-05-22:
Did you or Someone you know successfully flashed that firmware on an s3 via the web installer and run it without issues? Wonder if it’s a problem specific to the S3. But my best guess so far is a connection to the partition table and wrong manifest.json data.

rbergen, 2023-05-24:
Personally, I haven't tried it as I don't have any S3 devices. Dave may have, but he didn't confirm to me as such. 

I'd say that if the issue was with the partition table, flashing via PlatformIO.ini also shouldn't work. The manifest template is basically a 1-to-1 translation of the partition table, which does work for non-S3 devices. Which makes me think we may have to switch to pre-combined firmware images for S3 devices - which is also what the "Preparing your firmware" section on the [ESP Web tools page](https://esphome.github.io/esp-web-tools/) seems to suggest, depending on how one reads it.

davepl, 2023-05-24:
I have tested S3 in the project, and there are a couple of configs that use it, but my understanding is that the S3 does not support DMA and can’t work with the matrix, so I’ve not looked much further...

glowingkitty, 2023-05-24:
Installing ESP home on my S3 board via their web installer works perfectly fine and the firmware runs perfectly. Still trying to understand what they do differently. Can't replicate it so far.
https://esphome.github.io/esp-web-tools/

Even took the platform.io log now as a base to update the manifest - but still, can't bring the firmware to work correctly. Flashing it via plattformio however is no problem and the flashing itself also works via the web browser. Just the firmware doesn't properly start...
```
"/opt/homebrew/Cellar/platformio/6.1.0/libexec/bin/python3.10" "/Users/.../.platformio/packages/tool-esptoolpy@1.30300.0/esptool.py" --chip esp32s3 --port "/dev/cu.usbmodem1101" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0000 /Users/.../.platformio/packages/framework-arduinoespressif32@3.20004.0/tools/sdk/esp32s3/bin/bootloader_qio_80m.bin 0x8000 "/Users/.../.../.pio/build/glowcore_v3_0/partitions.bin" 0xe000 /Users/.../.platformio/packages/framework-arduinoespressif32@3.20004.0/tools/partitions/boot_app0.bin 0x10000 .pio/build/.../firmware.bin
esptool.py v3.3
```

```
{
  "name": "Software",
  "version": "0.15.1",
  "home_assistant_domain": "software",
  "new_install_skip_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32-S3",
      "parts": [
        { "path": "../bootloader/bootloader_qio_80m.bin", "offset": 0 },
        { "path": "../firmware/partitions.bin", "offset": 32768 },
        { "path": "../firmware/boot_app0.bin", "offset": 57344 },
        { "path": "../firmware/firmware.bin", "offset": 65536 }
      ]
    }
  ]
}
```

glowingkitty, 2023-05-24:
I made more progress and was able to get my firmware to run (after flashing it with the web installer). But the filesystem is broken if I do it that way. If someone can help figuring out the issue, I am happy to share the code/settings of course:
Check out what I posted here:
https://github.com/esphome/esp-web-tools/issues/377

noalear, 2023-05-27:
> I made more progress and was able to get my firmware to run (after flashing it with the web installer). But the filesystem is broken if I do it that way. If someone can help figuring out the issue, I am happy to share the code/settings of course: Check out what I posted here: [esphome/esp-web-tools#377](https://github.com/esphome/esp-web-tools/issues/377)

I purchased two Heltec V3 S3 devices (V2s are discontinued) and am having the same issue you are on both. I havent had any luck getting platformio to work either, but Im new to it so that may just be me doing something wrong. I checked out your esphome thread and I can see this may be part of a larger overall issue with the way S3 is handled. This seems enough above my paygrade that I'm contemplating simply sending these back and just tracking down some solid S2 devices that aren't discontinued. 

@rbergen If you've got an Amazon gift list, or anything to that effect, I will send you one if you are interested.

rbergen, 2023-05-28:
Hi @noalear, first off: thank you very much for your kind offer, that is highly appreciated. I've put a (very short) gift list together for the purpose: https://www.amazon.nl/hz/wishlist/ls/24KZ021WVYZRD.

Bun-Bun2, 2023-05-30:
> I purchased two Heltec V3 S3 devices (V2s are discontinued) and am having the same issue you are on both. I havent had any luck getting platformio to work either, but Im new to it so that may just [...]

The discussion here details what is required to get the Heltec V3 working.

https://github.com/PlummersSoftwareLLC/NightDriverStrip/discussions/265

> I have tested S3 in the project, and there are a couple of configs that use it, but my understanding is that the S3 does not support DMA and can’t work with the matrix, so I’ve not looked much further...
> […](#)

My understanding is the ESP32-S3 does support DMA, it's structured a bit differently than the original ESP32. There are 4x4 RMT channels (tx/rx) whereas the original had just 8 channels. DMA is now called GDMA and there are 5x5 GDMA channels (tx/rx) that can be applied to the various peripherals including RMT channels.  ESP Home and other projects are supporting 4 DMA enabled outputs for driving LEDs and Matrices.

davepl, 2023-06-04:
Just a heads-up:  I'm working on fixing the Feather S3 project - It seems to die allocating memory for the GFXBASE pointers, so I've assigned this bug to me until I get the S3 working again.

noalear, 2023-06-05:
> Hi @noalear, first off: thank you very much for your kind offer, that is highly appreciated. I've put a (very short) gift list together for the purpose: https://www.amazon.nl/hz/wishlist/ls/24KZ021WVYZRD.

I've tried to get that to work, but it says those items are not available in your region. Could you please verify that they are and that the address theyd be shipped to is correct? 

> Just a heads-up: I'm working on fixing the Feather S3 project - It seems to die allocating memory for the GFXBASE pointers, so I've assigned this bug to me until I get the S3 working again.

Thanks Dave! Love your work.

rbergen, 2023-06-05:
@noalear I had indeed forgotten to set the delivery address for the list. That's now been fixed. However, the "delivery address" section of the "manage list" dialog did show this message:
[image]

As these are third-party seller items, it looks like you may still run into problems right this moment. I do appreciate your efforts, and as far as I'm concerned we don't have to rush anyway.

rbergen, 2023-08-04:
I unassigned @davepl because the crashes he was working on were fixed with the merge of #380.

davepl, 2023-09-01:
> S3 had a hard life a few weeks/months ago. At least partially, this was because of flippant changes to the partition table to enable the OTA configuration as the norm on S3 and mesmerizer.
>
> However, at this point, I think there are a couple of us running S3's as development devices (with immunity of partition tables being changed beneath us as that was months ago...) so please consider marking this issue as 'fixed' if current S3 support is building and running for you.
>
> Might it work poorly upgrading from a developer build from a few months ago? Yes, probably. Even IPUs from months ago won't have great upgrade experiences. However, if the upgrade basically works now, instead of coughing up 0xffffffff errors, please consider marking this as fixed as generically, S3 works and we've had reports of success with theHeltec V3. If there's still a problem with YOUR S3, please repond to the above questions and let's figure the out. Otherwise, this seems a candidate for closure.

I run TTGO-S3 all around my house, and so I know the ledstrip_feather configs that use S3 are golden, and you can upgrade in place.

I don’t recall doing anything flippant, but I guess that’s in the eyes of the beholder.  I made it work, I know that.  And for 8M flash too, but where’s my parade?

Anyway, samples size of one and all, but the S3 seems solid in the ledstrip_feather config.  No ladders needed for a long time.

- Dave

rbergen, 2023-09-01:
The issue was opened in relation to the web installer, and I have not yet seen anyone confirm they've been able to flash a project using it. It may well be that it didn't work on the S3 but does now, due to the changes applied through #392. 

The second anybody can confirm a successful flash on an S3 with the web installer, I'll be very happy to close this.

rbergen, 2023-10-28:
Closing this, after I've experienced myself that flashing an ESP32-S3-DevKitC-1 now works.

closed 2023-10-28 by rbergen