PlummersSoftwareLLC/NightDriverStrip #281: SerializeToJSON should also be overriden when creating new effects

dkanavaris, 2023-05-23:
## Bug report

**Problem**

When creating a new effect the SerializeToJSON( ) virtual function should also be implemented. If not when reseting the esp32, either via the reset button or powering off and on,  the new effect still shows up at the effects list and the web server but nothing is drawn on the Leds which remain turned off (black).

**Steps**

1.  Create a new effect and implement the Draw( ) function but not the SerializeToJSON( ) function
2.  Build and Upload to the esp32. The effect works at first but stops working when powering off and then on or when reseting the esp32.
3.  

**Example**

**Notes**

I found that adding these lines to the SerializeToJSON( ) function inside the LEDStripLEDStripEffect solves the issue.
Question is should SerializeToJSON ( ) be implemented for each new effect or this solution works too ? 

 virtual bool SerializeToJSON(JsonObject& jsonObject) override
    {
        StaticJsonDocument<128> jsonDoc;
        
         // Added these 2 extra lines.
        JsonObject root = jsonDoc.to<JsonObject>();
        LEDStripEffect::SerializeToJSON(root);
        
        jsonDoc[PTY_EFFECTNR] = _effectNumber;
        jsonDoc["fn"]         = _friendlyName;

        return jsonObject.set(jsonDoc.as<JsonObjectConst>());
    }

rbergen, 2023-05-23:
@dkanavaris As you can tell by looking at the code of the files in include/effects/matrix and include/effects/strip, the majority of effects included in NightDriverStrip do not override the SerializeToJSON() function included in LEDStripEffect, which must be the base class of any effect that is added. In fact, only those effects that persist and load properties to/from JSON beyond the friendly name and the effect number override SerializeToJSON(). Otherwise, because of the way virtual functions in C++ work, the superclass' SerializeToJSON() method is used anyway.

Adding a new effect to a device on which the effect list has already persisted _can_ lead to the behavior described in #271, which I'll look into documenting more clearly in README.md.

All in all, I think we'll need to take a look at the complete code of your new effect to understand what's going wrong for you. Could you provide a link to the code of your new effect, so we can do so?

dkanavaris, 2023-05-23:
It seems the problem was not the SerializeToJSON() function but, as very well pointed out , the fact that the effect list had persisted. Updating the effects lists seems to have solved my problem.

Thank you for your response!

dkanavaris, 2023-05-23:
It seems the problem was not the SerializeToJSON() function but, as very well pointed out , the fact that the effect list had persisted. Updating the effects lists seems to have solved my problem.

Thank you for your response

closed 2023-05-23 by dkanavaris

rbergen, 2023-05-23:
You're welcome. I've just merged an update to the README in #282 to clarify this.