PlummersSoftwareLLC/NightDriverStrip #285: Update README.md about the addition of new custom effects

dkanavaris, 2023-05-24:
**Problem**
When creating a new custom effect, after adding the strip effect number in ```effects.h```,  the new effect's JSON constructor should also be added at the ```g_JsonEffectFactories``` map in file ```effectfactories.cpp``` so that the effect persists across reboots. If the effect's JSON contructor is not added in the map, when restarting the board the new effect is not present in the ```/effects.cfg```. 

**Notes**
I think that this should be noted in README.md so the steps on how to add a new custom effect are clearer.

davepl, 2023-05-24:
I sense we'll run into this a lot even if there's a readme... perhaps the de-serializer could check the effects table to see if a new effect is present somehow?

Another option would be to include a build date in the JSON and if its a new build, reset?

dkanavaris, 2023-05-24:
Regarding the de-serializer, could a solution be that instead of defining a number for the effect at ```effects.h``` to define the name of of the effect's class  and then dynamically create objects based on the effectName so that the ```g_JsonEffectFactories``` map is no longer needed ?

### Example 
To make it clearer, I mean instead of : 
```
#define EFFECT_STRIP_COLOR_FILL                         17
```
the code becomes
```
#define EFFECT_STRIP_COLOR_FILL                        ColorFillEffect // Name of ColorFillEffect class
```

and then call
```
std::make_shared<EFFECT_STRIP_COLOR_FILL>(jsonObject)
```
Altohugh this requires a lot of work to be able to dynamically created objects based on a effect's name.

I hope what I wrote was clear and did not cause confusion.

rbergen, 2023-05-25:
Providing a way to pick up a new effect when a serialized JSON effect set is already present was already on my mental shortlist of things to work on. Ideally, I would somehow combine the list of effects created in `CreateDefaultEffects()` with the definition of the effect factories. Part of the challenge lies in the dealing with situations where one effect class is used to create multiple variations of one effect as it is done with Spectrum, for instance.

I don't think that just resetting an existing effect JSON file when a new effect is added is sustainable - even when that currently is the thing to do. I think that not being sustainable is particularly true now that we're moving towards enabling the changing of the configuration of individual effects, as well as the order in which they are shown. If we clear the config whenever a new effect is added that will then become a nuisance to the user; they instantly lose all their customization of the config of the existing effects.

As to the latest comment of @dkanarvis, the identifier of an effect class that will come out of the JSON file will always be some basic type (basically either a number or a string). I don't see how that can be used for a construct like the one suggested. Also, the fact that a `#define` must still be set doesn't actually solve the original issue in my view.

In short, there's still some thinking to do on how to go about this. In the meantime, I think an extension to the README to cover the addition of a #define to `effects.h` and a factory to `effectsfactory.cpp` is reasonable and doable. The documentation already includes the addition of the effect to the code of `CreateDefaultEffects()`; that text can easily be extended with a few lines to cover this.

rbergen, 2023-05-25:
I think I found a way to effectively merge the composition of the default effect list and the list of effect factories, as well as make new effects be added to an already serialized set.

I'll first process some of the review points on my latest PR (thanks @davepl!), and then implement this. I'll combine both in the same new PR.

Small heads-up: the numeric `#define` in `effect.h` will stay at least for now, but it will be much more difficult to overlook.

TBC...

rbergen, 2023-05-25:
I've opened PR #287 to implement the changes I announced in my previous comment.

rbergen, 2023-05-26:
Closing this after the merge of #287.

closed 2023-05-26 by rbergen