PlummersSoftwareLLC/NightDriverStrip #289: Feature: Stocks

davepl, 2023-05-28:
It would be nice to have an effect to show a stock price.  Since we do/will soon support multiple copies of an effect with different parameters, I think the effect only needs to handle one stock at a time.

You'd need to sign up for a free-to-use stock ticker service and provide a way to set the API key, ala the Weather example.

mggates39, 2023-06-01:
This one looks like a good way to cut my teeth on the code.

mggates39, 2023-06-07:
I have a dev branch working on this now.  Based it on Subscriber so I had the configuration parameter available to set the ticker code.  Still need to work on the display layout and I would love a good library for SVG since that is what the Stock Services provides for ticker company icons.

rbergen, 2023-06-10:
@mggates39 This is a heads-up that there's a "mild" overhaul of the SettingSpec on the way, aimed at significantly reducing duplication of specs-related data. That also impacts the way effects have to define and register their specific setting specs. If you'd like a preview you can take a look at [`PatternSubscribers.h` in the respective branch in my fork](https://github.com/rbergen/NightDriverStrip/blob/save-memory/include/effects/matrix/PatternSubscribers.h).

I'll update this issue once the upcoming PR has been merged into the main tree.

mggates39, 2023-06-10:
Thank you.  I will take a look at it this evening.  I am at my eldest grandsons highschool graduation today.

rbergen, 2023-06-10:
Congratulations on your grandson's graduation! That's obviously  a very good reason to not look at this today. :)

mggates39, 2023-06-11:
Thank you.  We all had a good time.
Now, haveing looked at the code you indicated, I don't see any issues with my code.  I like how you refactored the setting code.  I will pack port it as soon as I see it hit main.

rbergen, 2023-07-04:
@mggates39 If you are still working on this - or planning to - please take note of issue #341, as I'm closing this issue in favor of that one. Dave's ideas about the effect have evolved a bit, and #341 reflects the current ideas on how it should work.

closed 2023-07-04 by rbergen