PlummersSoftwareLLC/NightDriverStrip #301: UI:  Change from TILES to LIST with checkboxes

davepl, 2023-06-01:
Rather than small tiles for each effect, we should use a "list" format where each effect gets its own row tile. There should be a checkbox on the left side to enable/disable the effect.

Other functionality, such as Activate, can be a link within the tile, perhaps right-aligned.

[image]

Louis-Riel, 2023-06-03:
I wanted to start a convo regarding this. I already have 2 display modes, I will add this layout, and make it the default.

For the effect decorations, I was thinking that it would be nice to have some attributes added to the effect list to be used for decorations. 

I think maybe a picture, and now, looking at the sketch you put up there, a color marquee configuration.

Louis-Riel, 2023-06-04:
This is the initial layout of this model, will refine over time:
Desktop
[image]

Mobile
[image]

Louis-Riel, 2023-06-04:
The light bar could have a configurable set of colors defined at the effect level if we add that. Also, the light bar will later be animated for show, I had issues making that work.

davepl, 2023-06-04:
Cool!  Could the light “bar” also be a grid?  If so, the web page could pull the current contents of the strip or matrix from the ColorData server and display the current contents there….

Louis-Riel, 2023-06-09:
I'm pretty sure the web interface cannot do raw sockets, but here's half a thought. The socket is meant to be used by some other thing running on the network. That thing could be running any framework server thing, and that could also expose a web socket. So the server could send color streams to the esp32, and report back to websocket. So the next thing is that the chip served site could use the UI build sequences, to then send that over the server,over the websocket, for it then play it back on the esp32. The server would then report the new running stream over websocket clients, so there, yes, the web UI could show a representative glow of the light bar.

KeiranHines, 2023-10-29:
I did a quick and simple design change to implement this. In the top right of the 'designer' section now there is a toggle to switch between the old card layout and the new list layout. Clicking anywhere on the list item with popup the settings. 

Any feedback on the design would be great. Other its ready for a PR. 
[image]
[image]

rbergen, 2023-10-29:
@KeiranHines I'm kind of speaking on Dave's behalf - which means I may be wrong - but I think he's looking for a list design that matches the ones shown in earlier comments in this issue (and the issue itself) a little more closely. 

My personal comments are:

- The current list item design looks a bit "cramped" to me; I think it would be easier to use this design if each of the list items has more height.
- I think we should include the settings cog in each of the list items, as we now do on the cards. I don't think opening the settings dialog by clicking anywhere on a list item is very intuitive, and I also don't think it'll be sustainable once the drag-and-drop reordering of effects (as per #300) is implemented.

I'll check with @davepl and ask him to comment directly with his concrete feedback on this iteration of the design.

KeiranHines, 2023-10-29:
Thanks for the feedback. I went a bit back and forth between giving the rows more height and being a more compact layout. I can try a few different heights and paddings post screenshot. 

As for settings. I actually have click and drag to reorder working on this branch. It picks up the drag event separately from a single click for settings. I didn't add the cog initially cause it wasn't in the above designs and I wasn't really sure where I wanted to put it in the row. Again I will add it in and post some more screenshots.

rbergen, 2023-10-29:
Thanks! I think the cogwheel may well not have been in the designs because the whole effect setting dialog hadn't been implemented yet - it certainly wasn't included in this repo's main branch.

davepl, 2023-10-29:
Just thinking out loud, I'd imagine each element could render as a tile rather than as a simple horizontal entry, something like this from another "dark model control panel UI phone" that I googled for:

[image]

But then the question is "what does it include?".  Right now we only have two attributes in the tile (name and enabled).  

```
----
NAME                 [Play Button] 
Description                    Time Remaining
----
```
Of course inactive ones wouldn't have Time Remaining, I suppose.

rbergen, 2023-10-29:
I'm just quickly noting that any description beyond the title (or any other informative property or decoration) would constitute a `LEDStripEffect` property we'd have to add and expose to the web UI via the API. I guess it makes sense to turn it into an effect setting as well, so it can be changed via the UI if a user wants to do so.

KeiranHines, 2023-11-04:
[image]
[image]
A couple of small updates. I have increased the size a bit. I have not yet added the settings cog. I was thinking next to the effect name.

Thoughts?

KeiranHines, 2023-11-04:
And with settings cog: 
[image]
[image]

davepl, 2023-11-04:
Cool!  At what point do you want me to start bugging you about layout?  Like the UI should work on the phone and desktop, and likely the tiles need a max width, that kind of thing.

I haven’t seen anything that can’t be changed later so far as I know, so I’ve been straying out of it!

Cheers,
Dave

PS:  Here’s a good example of the Nanoleaf UI with tiles:

KeiranHines, 2023-11-05:
I have just pushed the latest updates to [this branch](https://github.com/KeiranHines/NightDriverStrip/tree/feature/designerImprovements) It is ready for any all feedback. It should be pretty mobile friendly. I haven't added a max width as the middle section of the row (name section) is set to grow to the available space meaning the rows should always been full width and the name should always be centered. On a large 4k screen or ultrawide this may mean the rows are a bit wide, but on such screen I prefer the grid layout anyway, hence why I left the option for both.

KeiranHines, 2023-11-05:
@robertlipe #491 has been opened. I will address that after I finish up with the drag and drop and list view. Shouldn't be too hard to add.

KeiranHines, 2023-11-05:
[image]
[image]
Another idea with a bit of color and details added. I added the time remaining as either XX.X (h|m|s) depending on most significant time remaining and added a yellow color when less than half the time left, red when less than a quarter.

KeiranHines, 2023-11-05:
@robertlipe  the time here is the remaining effect time. The same time remaining as in the top left, just formatted to be readable instead of just ms. 
I agree the yellow/red are more call to action colors. I couldn't really think of colors I liked more. I just felt it could use a splash of color.
As for linear progress vs circular. The current grid layout had linear. And I considered that. But as you noted. On a large screen, or a short timer it was comically large and raced across my screen so I switched it out for the circular.

rbergen, 2023-11-05:
I'll mainly comment on the color of the countdown timer visualization, to the extent that we really shouldn't use signal colors for that. The neutral ➡️ yellow ➡️ red progression will make me instinctively panic about "how do I stop this?!?", which we all agree is not the point or intention. 

Beyond that, I don't have a very strong personal opinion about linear vs circular progress (both can work), but I do note we have a lot more horizontal space in the list layout than we have vertical, which may plea for a linear progress indicator after all. And yes, at the same time it should not become so big as to cause eyeball strain while tracking it. Maybe it's possible to put a few designs for horizontal indicators together, so we can make up our minds by looking at them?

davepl, 2023-11-05:
I’ve been unable to do a filesystem upload… are there any special tricks, like holding down the BOOT button, or anything required these days?

- Dave

Serial port /dev/cu.usbserial-8411301
Connecting......

A fatal error occurred: Invalid head of packet (0x01): Possible serial noise or corruption.
*** [uploadfs] Error 2
================================================================================================================================================== [FAILED] Took 9.44 seconds ==================================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
mesmerizer     FAILED    00:00:09.442

(Pretty sure it’s no serial noise, I can flash fine…)

KeiranHines, 2023-11-05:
I haven't had any issues building/uploading. I will try tonight and report back if I see similar.

davepl, 2023-11-05:
Figured it out… serial monitor was open in the background, which prevents it from working!

I did a sync from the main tree and built and uploaded flash and filesystem, but don’t get the tiled version yet.  Or did I jump the gun, and this isn’t in main yet?

Thanks
Dave

KeiranHines, 2023-11-05:
Not on main yet. It's on https://github.com/KeiranHines/NightDriverStrip/tree/feature/designerImprovements

KeiranHines, 2023-11-06:
Examples with a linearProgress (Circular still there for comparison). Each one has a slightly different vertical position. 
[image]
[image]
[image]

rbergen, 2023-11-07:
@KeiranHines Thanks! My vote goes to #3 as well. And now that I've seen it, I think I prefer it over the circular progress indicator after all - the latter now comes across to me as if it's pushed into a corner.

KeiranHines, 2023-11-08:
I have updated my branch for this work to use option 3 over the circular progress. Once my current PR is merged (#494) I will open a PR for this unless there is any other feedback before opening it.

PR #511 (2023-11-14): Feature/designer improvements
## Description
Fixes #301: Adding a list view as well as the existing grid view for effects
Fixes #300: Drag and Drop to reorder effects is now possible on grid and list layout
Added a "show disabled" button to hide effects you have disabled. 
[image]

## Contributing requirements
* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein i [... cut]
closed 2023-11-14 by rbergen