PlummersSoftwareLLC/NightDriverStrip #303: UI: UI Comes up blank

davepl, 2023-06-01:
When I launch a browser and go to the mesmerizer IP, I get the admin control panel but I don't really see anything until I click on the home button.  It should default to showing the effect list.

MPMcIntyre, 2023-06-02:
Hi Dave, could you possibly provide a screen shot of the browser console (F12 to open the console in your browser, then F5 to refresh the page), it might give an indication of the error. I think it has to do with not defering the react script tag.

rbergen, 2023-06-02:
@MPMcIntyre I don't think there's an error (or at least, my console doesn't show any), it's just how the UI currently works.

What the issue raises is that the web UI comes up like this when you load it:
[image]

That is: the page loads and the navigation controls are there, but the majority of screen real estate is empty. 

What I think Dave would like is the effect tiles to be shown straight away, which now requires a click on the home icon at the left:
[image]

(Actually, Dave is also suggesting in #301 to change the tiles to a more list-like format, but that's a different issue.)

I hope this clarifies the issue somewhat?

MPMcIntyre, 2023-06-02:
@rbergen Ah okay, yes that clarifies it thanks!

davepl, 2023-06-02:
Hope this is what you’re looking for:

￼

Louis-Riel, 2023-06-03:
A big change to the UI is coming soon. This next PR will a solid base to continue building the site on. This will include a migration to typescript, upgraded build system, and a re-giggying of the UI.

We have alot of UI design work to do, which will include the screw flow, and the look. We need help with good designers with that human touch. 

I will try to push something over the weekend, that will probably not yet include the typescript migration, just the reworked UI, for folks to get thinking about cool design ideas

Louis-Riel, 2023-06-04:
The effect list is now displayed by default when opening the page.

Louis-Riel, 2023-06-04:
[image]

davepl, 2023-06-21:
Fixed in the latest build I tried from Louis!

closed 2023-06-21 by davepl