PlummersSoftwareLLC/NightDriverStrip #306: UI: Mesmerizer Monitor, change to width & height vs count

zephyrr, 2023-06-04:
## Bug report
Dave suggested adding a new issue for this; see # 302 for the discussion

**Problem**
In the query for color information (ColorDataPacket), the caller will receive the led count, but must independetly know the width & height (which can be checked against the count).

Return the width and height in the query response, instead of the count (which can be computed as width * height).
use height == 1 for a linear layout

typedef struct
{
uint32_t header;
uint32_t width;
uint32_t height; // use 1 for linear strip
CRGB colors[NUM_LEDS]; // Array of LED_COUNT CRGB values, == width * height
} ColorDataPacket;

// width and height could be uint16_t instead if you prefer

You may then want to modify the Unity example accordingly.

**Steps**

**Example**

**Notes**

davepl, 2023-06-04:
I checked this in, it’s waiting on CI and PR.

typedef struct
{
    uint32_t  header;         
    uint32_t  width;   
    uint32_t  height;   
    CRGB      colors[NUM_LEDS];  // Array of LED_COUNT CRGB values
} ColorDataPacket;

rbergen, 2023-10-28:
I'm closing this, as the count ➡️ width x height conversion has been implemented in 5301534aa344b94b2e7546046fca6b0d0c4fa587. The request for the preview web UI is documented in #302, which is still open.

closed 2023-10-28 by rbergen