PlummersSoftwareLLC/NightDriverStrip #309: Work Item: Refactor screen code

davepl, 2023-06-07:
The Screen has has grown from 2 to "too many" different screens supported, and everything is done with #ifdef.

Ideally, the base Screen class would inherited by new classes like M5Screen and TFTScreen and OLEDScreen that implement their methods, like SetTextColor, by talking to the appropriate display.  

In other words, refactor the screen code to use encapsulation and inheritance rather than ifdefs...

Another consideration is to make it support the GFXBase drawing, ie: Screen inherits from GFXBase.  That way we'd be able to add the screen as an output channel one day.  But it's lower priority.

davepl, 2023-06-08:
Likely class hierarchy could be:

Screen : GFXBase
+ OLEDScreen : Screen
+ TFTScreen : Screen
+ M5Screen : Screen
+ LCDScreen : Screen

SeanPGit, 2023-06-08:
Would there need to be separate inheritance or sub-classes for "M5 Stick" and "M5 Stack", the latter having a much larger screen. Maybe that's also why I ran into weird problems when I first compiled for the latter, and me not recognizing the former was the intended target; now I have both but need to get back into these projects after a bit of a hiatus!

davepl, 2023-06-08:
I think they can share the same class, but some of the setup might be different between them!

davepl, 2023-06-11:
This is done and checked into the LEDViewer branch of my davepl/NightDriverStrip fork.  Will propagate to main soon.

closed 2023-06-12 by davepl

rbergen, 2023-06-13:
Fix merged through #324.

davepl, 2023-06-13:
I think the headers and objects are the same, and the screen is even the same resolution, but its a different link lib, which is why you’ll see SPECTRUM (M5Plus) and SPECTRUMLITE (M5) projects.  Steal the SPRECTRUMLITE as a starting point if needed.