PlummersSoftwareLLC/NightDriverStrip #323: Work Item: Partition table

davepl, 2023-06-12:
The chip we're using on the Mesmerizer has 8GB of PSRAM and 8GB of flash (or more, depending on which chip you have).  But our partition table only works for 4GB.  I've tried to create an 8GB version but can't set it to boot, so need someone to take a look!

Two main goals:
1) Enable OTA by creating the two APP partitions and then re-enable ENABLE_OTA for mesmerizer
2) Get access to all 8GB of PSRAM if possible (could be the Arduino version doesn't handle it, but can't prove it)

davepl, 2023-06-19:
> There are choices on how we use the extra 8 (extra 4?) MB of PSRAM. References:
> https://thingpulse.com/esp32-how-to-use-psram/
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#configuring-external-ram
> What's the best use for it in the project?
>
> Treat it as yet another pool to be manually managed, via ps_malloc and ps_free? (yuk)
> Integrated it into the standard malloc/free heap, but incur the weirdness of buffers presumably being wildly different speeds depending on whether they happened to land from the internal SRAM or the external SPI bus - remembering that the "S" is "Serial"?
> Put .bss (zero-initillized globals which, in a normal c++ apps, are pretty uncommon) and .noinit (globals, but not reset across reboots - again, uncommon in most apps) in this special area?
>
> It looks like some of these options depend upon which revision of the chip is installed, which might not be a problem in a manufactured board with a tightly-controlled BOM, but probably unpleasant in an open-source project running on random boards.
>
> At least the last of these will require different linker scripts, changing the memory maps and access times and making binaries built incompatible with the others unless we add funky runtime remapping. The second of these will require linking against a newlib that was built with compiler workarounds for chip defects for some revisions.
>
> I have no idea how common each of these revisions are in the real world.
>
> I'm willing to help a bit on this, but need some guidance from someone that knows how extra RAM could best be used as from a pure systems design view, there's not a clear, obviously "right" answer from those options; th [... cut ...]
> I'm guessing that extra flash has an obvious use with A/B partition tables to allow for hot-swapping OTAs and the rest going to a LittleFS or SPIFFS or whatever filesystems. It could hold fonts or .gifs or something...
>
> Using it effectively may tie us more tightly to ESP32 hardware by making more use of esp-idf features and code, making it more difficult for users of Arduino, STM, or the various RISC-V parts. Is that troublesome?

So far the target has been to allocate up to 500 frames of buffer.  If PSRAM were tight, it could be less, but since it’s abundant… 500.  And then we allocate most everything we can from PSRAM preferentially to use it up first.  

There are a couple of places, like decompression, where I copy to a RAM buffer first because the process is random access which is slow over SPIRAM.  But so far, everything runs well using PSRAM.

There’s a RESERVE_MEMORY setting in the builds as well.  If USE_PSRAM is NOT defined, that’s how much RAM will be left aside during buffer allocation.  If USE_PSRAM is defined, it’s how much PSRAM will be preserved.

The most pressing issue though would be to figure out why we can’t seem to partition the larger flash in such a way as to take advantage of it.  So far I’ve been unable to get it to boot with anything other than 4GB partition max.  But for all I know the Arduino version of the espressif code might assume it, I don’t know!

 Dave

rbergen, 2023-07-07:
Completed by #346.

closed 2023-07-07 by rbergen