PlummersSoftwareLLC/NightDriverStrip #333: Settings Issue - Need min, max, etc

davepl, 2023-06-24:
For most settings, like HueSpeed and ScrollSpeed and so on, sometimes only a range of reasonable values makes sense.  I imagine the UI will eventually display a slider for such settings, and it also needs to know a min and max range.

So, I think properties like PTY_HUESPEED may need either system-wide ranges or, better yet, effect-defined ranges.  

No idea how best to handle or implement this, just calling out that it's a likely need!

rbergen, 2023-06-25:
I think we can implement this quite easily using an extension of the `SettingSpec` class. If we add "min" and "max" members to it and then add those to the output of the setting spec endpoints, we're there. Effects can already "publish" their individual setting specs, so we get that one for free.

The only "challenges" I see is how we:

- Deal with the fact that numeric types can be int, size_t or float/double. I'm thinking we'll probably specify min/max as doubles and let the API client interpret that as another type where appropriate/it sees fit.
- Indicate if min and/or max are set at all. This will probably be handled with separate boolean flags. (I'll note just for the record that nullable primitive types would be extremely handy for this. 🙂)

While typing this I'm thinking that overall, I'd probably create a min/max subclass of the basic `SettingSpec` class that can be used if min and/or max need to be specified, and use specialized function templates to generate the API endpoint output. Maybe I'll find the time to add this, today.

rbergen, 2023-06-25:
So, the subclass and specialized template was a nice idea in theory. It was until I had to use dynamic_cast to make it work, and discovered the toolchain doesn't support RTTI. Lesson learned. 
I switched to a more straightforward extension of `SettingSpec`, which also seems to work. I'll open a PR once CI completes successfully on the respective feature branch.

rbergen, 2023-06-26:
Yes, they are. I now realize my comment was actually a reference to an out-of-band conversation I had with Dave a while back. There was no way for you to know that.

That said, thanks for mentioning it!

PR #336 (2023-06-26): Add range to SettingSpec, parallelize CI
## Description

This:

- Pins the espressif32 version to 6.3.0 or over - this stops builds from breaking if the most recent release is < 6.1.
- Parallelizes CI for the vast majority of builds, which brings their overall runtime down to approximately 10 minutes. Only builds that include a push to the web installer Pages site are still sequential.
- Adds optional range specifiers (minimum and maximum value) to the `SettingSpec` class. Credits to @robertlipe for bringing `std::optional` to my attention again. This fixes #333.

## Contributing requirements
 [... cut]
closed 2023-06-26 by davepl