PlummersSoftwareLLC/NightDriverStrip #334: Easy Bug to Fix: Pulsar effect

davepl, 2023-06-25:
The pulsar effect is supposed to, and used to, display little blooming stars that get to be about 16 pixels wide.  Right now they're enormous, extend outside the screen, and crash the system.

Basics:  Fix Pulsar!

davepl, 2023-07-01:
I’m of the opinion that if it draws outside the bounds, we should catch that case and throw a fatal exception, and the effect should get fixed.  The alternatives seem to be eating the pixel and hiding the bug or letting I corrupt… so that’s my though!

> On Jul 1, 2023, at 7:16 AM, Robert Lipe ***@***.***> wrote:
> 
> neighbors

davepl, 2023-07-02:
> Every case that I could find silently drops the pixel, letting it just
> mask/clip at the edges, which would allow you to fill the four corners,
> each with a fourth of a circle. This is how, say, bodmer TFT and Arduino do
> it.
>
> Where is the corresponding code for HUB75? We should examine both drawPixel
> and drawCircle and I may have some bandwidth today for it, but ack isn't
> finding them.
>
> If it's in the hidden file
> (grrrr) libdeps/mesmerizer_anamatrix/SmartMatrix/src/Layer_BackgroundGfx_Impl.h
> there it looks like neither
>
> void SMLayerBackgroundGFX<RGB, optionFlags>::drawCircle
> nor
> void SMLayerBackgroundGFX<RGB, optionFlags>::drawPixel
> test it, but
> void SMLayerBackgroundGFX<RGB, optionFlags>::drawPixel
> does and it's probably the implementation of the previous one
>
> ...but that file doesn't contain 'HUB75' at all and doesn't do the kind of
> bit-whacking that I'd expect to see in a HUB75 driver, No file that I can
> find contains both terms via
> ack -i drawPixel $(ack -l hub75)
>
> Where is the source for the right drawPixel and drawCircle to draw on a
> HUB75 panel?
>
> I really need to get GDB playing nice on Mesmerizer. Again, are you using
> the inbuilt JTAG of the S3 or are you wiring an external pod to that
> mystery connector? There's a libary I see named 'remoteDebug' but I can't
> tell that t=it's a debugger at all, but is rather a place to plop debug
> prints, Is that fair? Should I be trying to extract stacktraces and
> breakpoints out of THAT?

For debugging you need to put jumpers across the 4-5 pairs of jumper pins that are marked as TDI,TDO, and so on.  Then PlatformIO should be able to debug with odd.

The mesmerizer board presents TWO serial ports, one for debugging and one for serial logging.  On the PC, at least, you have to change the driver on one of them with Zadig.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html￼

I believe the checking of the pixel and the throwing of the exception should happen at the lowest level (setPixel?) wherever it’s about to write to the LED buffer. 

- Dave

davepl, 2023-07-02:
> This is where ESP32-S3 is supposedly much more awesome, but I'm still a few
> steps from that point on my S3 board that I'm building up for Strip + HUB75
> development. There, that's all on the SoC itself. "ESP32-S3 has a built-in
> JTAG circuitry
> <https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-builtin-jtag.html>
> and can be debugged without any additional chip. Only an USB cable
> connected to the D+/D- pins is necessary"
>
> So on Mesmerizer, jump TDI->STDI, TCK->13, TDI->12 (two TDI?!?!) TDI->15,
> and TMS->14 ? [ Edit: labeling details aside, yes, I think that's what the
> linked doc is indeed saying, though the bottom TDI is less clear. Is it
> "the four JTAG pins in the cited doc" or "all five pins of Jfoo right there
> together"?
>
> Up on the other end of that connector RX and TX are pre-jumped.
>
> I inventory 2.54 jumpers literally by the hundreds...big pill bottle of
> them from when I had boards made. But this is that 2.0mm pitch connector
> that I've asked about pinouts/schematics for and I don't stock jumpers for
> those.
>
> Are the signals tolerant enough that plain ole Dupont wires making nearby
> loops won't confuse it? I guess if a dangling pod is OK, it probably is,
> but I don't know the background/justification of this connector or where
> these pins really GO.
> [... 22 lines cut ...]
> hitting 'magic' numbers and only in one effect...we think) before? I've run
> the mesmerizer build for hours while working on my own stuff and never
> noticed this crash.
>
> RJL

Yes, you should be able to get away with Dupont connectors!  Cool about the S3, I’ll have to try it!

I have a big pull bottle of 2.00 and 2.54mm now, because the little ones are hard to find!

My guess is that the effect in question has a sound-based amplitude, and that my recent changes to make audio broke it, and I just never happened to display that effect for very long!

- Dave

davepl, 2023-07-03:
> Cool, I'll try to cable it up in the next few days.
>
> S3 looks awesome. (I use C3 regularly...) Unless you're buying a zillion
> and need rock bottom prices and don't care about the extra features, I
> don't know why ESP32-classic is really still a thing. Ten more GPIOs, but
> it's missing a few little things. It may cost a few dimes more, but saves
> you in adding an external CP343 bridge or similar. I'm really hoping
> Espressif pushes a dual-core, low-cost RISC-V soon. Mesmerizer (or at least
> NightDriver) might support such a board soon after I could get my hands on
> one.
>
> I'll get a mixture of 2.00 and 1.27s on the next slow boat I charter. On
> this kind of thing, shipping is so disproportionate to the product that it
> makes sense to batch it.
>
> Yes, I remember this effect having a beat tracker. So maybe if you made
> sweeping changes there recently, that's the root cause.
>
> I know I talk too much, so let me try a super-direct question: is that
> header on Mesmerizer compatible with anything else in the industry? 2.0
> JST/PH connectors of that size don't appear to be common. I don't recognize
> the pinout, but I'm not an ESP32 expert. Is there some pod or cable or
> satellite board or something that I should be looking for to bond with this
> board? I'd like to debug/expand/probe Mesmerizer and if I can buy a
> ready-made cable/board, I'd MUCH rather do that than dremel down and
> superglue a bunch of my own PH's and glom up a DIP.
>
> Increasingly, since we know we're going to have to fix the effect anyway,
> I'm leaning toward just fixing the effect for now because crashes are bad.
> We can chase the higher-order issues of drawCircle and friends scribbling
> out of bounds later. If I code it up, wire only that effect to play for an
> hour sitting next to a radio, and log no reboots, should we consider it
> fixed enough for a submit?
>
> RJL

Just a heads-up:  I’m adding low level exception code to throw if you try to access invalid pixels.  I’ve already founds bugs in a few effects, and am fixing as I go!

- Dave

davepl, 2023-07-04:
Wow, there were a LOT of broken effects writing out of range.  Wonder where they were writing before!  

In any event, I added the safeguard code and for now just debug warn if you draw out of range, will make it more hardcore soon.  I fixed the 10 or so effects that triggered it.

- Dave

davepl, 2023-07-04:
> Awesome. Thanx.
>
> Glad we were at least able to find the problem in the abstract enough for
> you to fix the actual problem.
>
> Kind of ironic to call it out in 'discussion' (and get no discussion) a
> week earlier...

Glad you brought it to the surface!  Based on the effects I fixed (most of which were inherited from Aurora) there were lots of places where it was overwriting. I can’t believe we survived the corruption, frankly!

- Dave

davepl, 2023-07-04:
> Maybe aurora silently clips, by design. (I remember asking if that was the
> intended design here.) It's not an unreasonable design decision to allow,
> say, drawing a fraction of circle near an edge to just be clipped/masked to
> allow the look of a 'bubble' near the edge, for example. It may be visually
> better than my approach of nudging the circle a bit closer to the screen.
>
> And, yes, things hit the fan when that code is ported to another system
> without the same fundamental design. As an example, AIX used to never ever
> ever fail a (reasonable) malloc request, though I can't recall if it was in
> the kernel or user space. The premise was there was "always" more virtual
> address space; there was always SOME page that had backing store that could
> be retired, one more process (except YOUR process. YOUR process is
> precious and we would never kill yours; the system only threw those other,
> more unattractive processes overwater to reclaim memory - there was a whole
> weird mind game that went with it to make you believe it would never happen
> to a process you cared about.) that could be sacrificed.
>
> This was before Linux existed, let alone popularized overcommit. It was
> also even a more thoroughly Orwellian washing of "reality" that made you
> believe that failures only happened to other processed and YOURS would
> never deserve it.
>
> Software on "my" OS used to get grief from developers all the time from
> devs porting code from that world. We had the nerve to fail a malloc and
> their software just never tested for it because...why should they? WE were
> naughty for handling them a signal - that they probably ALSO didn't catch.
>
> Impedance matching of this type can be really messy as the rules are rarely
> written inside the part you're copy/pasting into another framework.

The fundamental difference seems to be a presumption that drawPixel() is allowed to attempt to draw outside its bounds, and it’s caught there.  We could certainly adopt that approach, or even enable it later if we like, but for now I think it hides bugs if we let it eat pixels so for now I prefer that all drawPixel calls be valid.

If it winds up cramping anyone’s style, I can remove the exception and replace it with a bounds check that discards.

- Dave

rbergen, 2023-07-07:
Fixed by #345

closed 2023-07-07 by rbergen