PlummersSoftwareLLC/NightDriverStrip #335: Umbrella env on ESP WROOM32

Thehaus64, 2023-06-26:
## Bug report

**Problem**

I just uploaded Night Driver on to my ESP-WROOM32. the "calm fire" and" medium fire" are green and when I when i tried green twinkle it is red. I am using the WS2812b  BTF-Lighting ones that Dave recommended. I believe it an RGB issue being backwards, however there is no place to change it that i can find in the webserver.
**Steps**

1.  Install the Night Driver software for ESP32
2.  Go to teh web page and click on calm fire
3.  looks like it works but its greens  instead of red and yellows

**Example**

**Notes**

Thehaus64, 2023-06-26:
> I'm a n00b in the code, but have been studying it lately. Let's try some reasoned hit and miss things.
>
> !) Edit platform.ini and see if adding -DTFT_RGB_ORDER=TFT_RGB to the [env:umbrella] block. Rebuild and reflash.
>
>    - Edit include/globals.h
> It looks like it sets COLOR_ORDER EOrder::GRB for "everything", but down in the case for UMBRELLA overrides it to EOrder::RGB . I'd try turning that knob back to GRB which is probably possible just by removing that line.
>
> I hope to not call anyone's baby ugly here, but there are so many layers upon layers of things that redefine things like this and pixel order and whether a "matrix" can mean only a HUB75 or if it can also mean a grid of NeoPixels and such that it's difficult to tell which knobs are effective at which levels. The whole definition of RGB vs BGR on NeoPixel is kind of fuzzy anyway. It's not like there are two wires you can swap; at an electrical and protcol layer, the green intensity is always the first one inside the packet and that's not really negotiable with the electronics.
>
> Neither of those may work (they may both work. Weirdly, if that's the case, the double negative would make it not work because remapping the fixed value twice would cancel out) but they should be easy things to try and easy to undo.
>
> Changing either of these will probably cause the entire program to get mostly rebuilt, so it may take a minute or two, must like your original installation did.
>
> I don't get the feeling that changing things like this or the LED configuration from the web interface is only a few lines of code away. I suspect that untangling the sedimentary layers of code system stop code system is going to take some effort. That's presumably why there's an active call for volunteers. (It's why I've been reading the code.)
>
> Good luck!

Thanks for the reply. I will try that when I get home through visual studio. I was referring to the website where it installs everything for you, then you can’t see or edit the code. I had the same problem when I did it through visual studio a few days ago so I switch the the web upload to see if it was my code that I typed wrong, but had the same green fire look. I wanted to report the bug to help the complete amateur, I’m not saying I know what I’m doing but I do know more than the average computer user. 
I appreciate your help and I’ll let you know if that fixes it so it can be reprogrammed for the website upload version.

Thehaus64, 2023-06-28:
Robert, just so you know that line of code did not help. I still have green fire, I wish I could figure out how to get this to work. I can’t even get the web server to upload all the way today.

Thehaus64, 2023-06-28:
Robert, I found out how and where to fix it. In global.h search for umbrella, then #define COLOR_ORDER       EOrder::GRB
I looks like each effect can be manipulated in here. I was look for a umbrella.h kind of situation. 
I got fire! Now to figure out how to change the pattern over internet. Back to the discussion boards. 
Thanks again for your input! I still can not get the (plummerssoftwarellc.GitHub.Io/NightDriverStrip) to load onto my board. It did one time but now it connects to the board, erases then uploads the code, and right where you set up the web server it says the serial port is being used. I checked and it is the one using it. So I gave up on that approach to easily changing the effects and back to figuring it out on my own. Tomorrow night my mom is having 50 people over so I have to figure it out by then. I want to switch from fire theme to some of the other effects I have written with out having to wire to the board in the middle of dinner. 
Have a great night/day!

rbergen, 2023-07-23:
@Thehaus64 Just checking in if this issue is still current. If not, I'll close it.

rbergen, 2023-10-21:
Closing for lack of OP feedback.

closed 2023-10-21 by rbergen