PlummersSoftwareLLC/NightDriverStrip #341: Feature: Stock Ticker

davepl, 2023-07-03:
We need a stock ticker effect.  It should use an API that is free and for which the user can sign up for their own key, so that everyone is not using the same API key, unless it's unlimited (even bettter!).

Ticker should cycle through the specified stocks at some interval.  It would be nice if it showed current and change, used color to indicate up vs down, and so on.  Look at other stock tickers to see what functionality is included (but don't overtly copy anyone's design!).

rbergen, 2023-07-04:
This looks like a duplicate/evolution of #289. The difference is that #289 describes an effect that shows the stock price for one listing, and this issue describes a ticker that cycles through multiple.

@davepl Would you agree that one stock price/ticker effect would be enough? If so, would I be correct that this issue reflects your current thoughts on how it should work?

davepl, 2023-07-04:
Yes, one should be closed as duplicate, my bad!

rbergen, 2023-07-04:
No worries. I've closed issue #289 in favor of this one.

mggates39, 2023-07-04:
Sweet, I was going to ask if this was going to be planned as I had already thought of a couple of ways to do it.  
My one question is do we want each ticker item to be a separate parameter, or just an array or comma separated list of items?

mggates39, 2023-07-04:
I already have the code for a single item compiling.  I have not tested it yet.  I will start planning for this model now.

davepl, 2023-07-12:
> @mggates39 <https://github.com/mggates39> I just realized I thought of a reply when I read this, but never actually typed it in. Apologies for the delay.
> As I understand it, stock symbols are usually composed of a few capital letters, optionally prefixed with a similar number of letters and a colon to indicate the stock exchange they are listed at: NASDAQ:ABNB for AirBNB, AMS:PHIA for Philips (listed at the Amsterdam exchange). To me, these stock symbols look like like tokens that can quite easily be combined into one comma-separated string/parameter.

You MIGHT need the exchange info but I think most quote services can do it off the ticker only, but you won’t know until you check the actual data service being used.

￼

Like the Apple Stock widget, it would be nice to have the ticker symbol, current price, change in red/green percent, and a little intra-day grap!

- Dave

mggates39, 2023-07-12:
@davepl and @rbergen Comma seperated list it is.  Do you want a maximum number or just grab the necessary number data sets with smart pointers?  The service I am using, https://finnhub.io/docs/api/introduction, works across many exchanges and the data returned includes the exchange name along with the company's full name.  I should have something for review this weekend.  I have not been able to run what I have, but it does compile.

mggates39, 2023-10-09:
@davepl I am at a point where I really need a mesmerizer board to test.  I have a 32x64 LED Matrix but nothing to drive it.  I have kept my fork up to date and can share my[ development branch ](https://github.com/mggates39/NightDriverStrip/tree/dev/mgg_Multi_Ticker) if someone wants to pull the code and run it.  Please send pictures if you do this.

mggates39, 2023-10-10:
Thank you for looking at the code.

- The the ParseTickerSymbols code was just some code I pulled from another example and is not being used yet.  I still need to come up with a way to parse the comma separated list of ticket symbols and turn them into an array of Stock objects.    Right now I just have a hard coded array of three objects for testing.  I am thinking of putting the stock objects in PSRAM instead of main and they would be dynamic based on the number of stocks requested.  I did not want to start that until I knew the displays at least looked like something.  Will be adding scrolling so that rather than flashing from one to the next stock as the current one rolls off to the right, the next one slides in from the left.  I have a time based bit of code to handle that with offsets in the works locally.
- I do not know what the behavior of HTTPClient.begin() is so I need to test it.
- I saw the JsonDocument sizes go down in the last batch of updates.  I will review and see what I can reduce them to going forward.
- You are right, having no key should be 'fatal'  though that behavior is not what I saw in the weather or subscriber effects.  I used those to model the basic framework.

I look forward to working with you too.

rbergen, 2023-10-10:
@mggates39 Very nice to see you're still working on this, and making progress. I saw you mentioning the need for a Mesmerizer to test - I just wanted to check if you've been in touch with Dave via email about this?

mggates39, 2023-10-10:
As for my M5CStickPlus I have [one](https://www.amazon.com/gp/product/B08VGST8LJ)  that I am using to drive small LED Strips.  I will try to snag another that I can pop open to get to all the I/O Pins.  I also have an [AdaFruit](https://www.adafruit.com/product/4745) Matrix Portal that has the wiring to drive the power to the board.  I have not figured a way to port the code to run on that M0 though.

mggates39, 2023-10-10:
@rbergen Yes, I have requested a board twice now.  Once back at the first of June and again at the end of July.  Received an e-mail back from Dave then saying that the next batch of boards had been DHL'd from the manufacturer.

rbergen, 2023-10-10:
@mggates39 He won't be offended if you reach out to him again to let him know yours hasn't arrived yet. He may have some means to track the package and see if it got held up somewhere.

mggates39, 2023-10-10:
@rbergen I did reach out again tonight.  Thank you.

mggates39, 2023-11-16:
I got this running on my 4MB board like I did for the weather icon feature and I found several bugs that I am cleaning up.  Then I need to rethink how I am doing the scrolling.  The simple thing I did throws a lot of error logs for pixel out of bounds.  Maybe after I get a matrix wired to it I can see how it looks and just live with the error logs until I have a better idea.

rbergen, 2023-11-16:
I'm glad to hear you're still working on this, but we will to have to fix the errors you mention before we add this effect. Uncaught out-of-bound errors are known to cause crashes in certain situations, sometimes even intermittently. 

`GFXBase` (a pointer to which is readily available in any `LEDStripEffect` subclass via `g()`) has a function called `isValidPixel(x, y)` that'll tell you if the coordinates you pass to it are valid. Preventing the out-of-bound errors can be just as simple to not drawing a pixel if `isValidPixel()` says its coordinates are `false`.

rbergen, 2023-11-16:
The main thing is that I don't want code at higher levels in the stack to rely on lower levels doing the checking - it just might not happen there, or not until eternity. **If** we ever were to run into (general) performance problems that are sufficiently significant to fix, that sort of lower-level check could be one of the things to go. Being protective and supportive to our effect writers is a priority, but getting the pixels out there is a higher one.

I agree that in principle, the basic algorithm used should ensure that only pixels that actually exist are sent colors to, so the whole issue doesn't have to be checked for anywhere. That said, for me it's also ok if the effect writer chooses to use boundary check facilities provided by the framework, if they are willing to take any performance hit that comes with. In the case of a casual scroll on a ticker tape effect, that may just work out well enough.

mggates39, 2023-11-16:
Yeah, what I have up there now, just cycles between the three hard coded symbols, Apple, IBM, and Microsoft.  If anyone wants to load it up and send me pictures of the matrix I would appreciate it.

I just have a few things outstanding:

- Process the actual list of symbols.
- Remove bad symbols from the list.
- Determine a reasonable maximum number of symbols or better yet make the ticket object list dynamic.
- Design and build a better scrolling method.

mggates39, 2023-12-01:
Now that Weather is stable, I am going to turn my attention back to this one.

mggates39, 2023-12-08:
I have a new milestone on my [development branch](https://github.com/mggates39/NightDriverStrip/tree/dev/mgg_Multi_Ticker). 

- Now uses std::vector to hold the stock ticker objects
- Removed bad scroll code
- Update the layout of the ticker data
- Added a toggle between high/low and open/close prices during the data display

Needs the following:

- Long term Iterator to go between the Stock Tickers in place of the pointer and manual double linked list
- Add actual parsing of symbol list in system options and creating correct stock ticker objects

I have built and run it on my development board, but I still do not have my hardware set up to drive my matrix display.  I need to clear out some workspace to do the hardware work.  If someone could pull the branch down and run the code, I would appreciate a couple of images of the display..

rbergen, 2024-05-21:
Resolved by #626.

closed 2024-05-21 by rbergen