PlummersSoftwareLLC/NightDriverStrip #348: BUG: AudioGraph bars all at the same level

davepl, 2023-07-07:
## Bug report
After about 30 minutes of running (might require less), the audiograph effect has the same level for all the bars.  

Changing away and back doesn't always fix it.

Changing to the Spectrum effect and then coming back to the Audiograph effect "fixes" it.

davepl, 2023-07-29:
What's weird is that stepping forward to the Spectrum effect and then back to Audiograph fixes it. So I wonder if AudioGraph needs to set peakDecayRate1 and 2 like the SpectrumAnalyzer effect does in Start()?

davepl, 2023-07-29:
I think I've fixed it.  The SpectrumBarEffect needed to set the global audio decay rates, which the spectrum effect did, but the spectrumbar effect did not.

rbergen, 2023-07-29:
Nice! Can I contribute to verifying this has indeed been fixed? If so, what branch should I check out and flash?

rbergen, 2023-10-21:
Fixed by #386.

closed 2023-10-21 by rbergen