PlummersSoftwareLLC/NightDriverStrip #349: WORKITEM: InitializeHardware, PrepareFrame, PresentFrame

davepl, 2023-07-07:
I think BaseGFX should likely grow by three virtual funcctions:

InitializeHardware, PrepareFrame, PresentFrame

They would be overridden as follows:

LEDStripGFX::InitializeHardware would do the AddLeds calls
LEDMatrixGFX::InitializeHardware would do the matrix creation

LEDStripGFX::PrepareFrame needs not exist, as there's no prep
LEDMatrixGFX::PrepareFrame would do the matrix prep code currently in drawing.cpp

PresentFrame would call ShowLEDs() on the strip and do whatever the drawing code does for matrices today in the LEDMatrixGFX::PresentFrame call

rbergen, 2023-07-08:
I think this work item encapsulates #343, which means we can close that one in favor of this.

@davepl Would you agree?

rbergen, 2023-07-14:
@davepl TL;DR I'm making the InitializeHardware function a static function for each GFXBase subclass that needs one, instead of a virtual function in GFXBase. 

The reason is that I don't think we can make InitializeHardware a virtual function. This because the initialization code is executed once for the full set of channels for LEDStripGFX and LEDMatrixGFX at least. For the latter, there already is a static StartMatrix function. For the former, the code needs to perform the addLeds call for each of the channels. 

For LEDStripGFX we _could_ move the addLeds call to an instance member function, but then we'd have to move the ugly loop unroll to another place in setup(), like where the LEDStripGFX shared_ptrs are assigned to the g_aptrDevices array. This is because of the pin number being a template parameter to the addLeds function.

I'll put something together so we can both look at it. I'll do that before picking up the PrepareFrame and PresentFrame part of this issue.

rbergen, 2023-07-16:
I've completed the implementation of InitializeHardware in the issue-349 branch in my fork. In the process, I also made some other changes in an attempt to make the codebase a little cleaner and more consistent. I'll open a PR once #362 is merged and I've resolved the merge conflicts that will create. 

Just for clarity: the upcoming PR will not close this issue, as the PrepareFrame and PresentFrame functions still have to be added. I just think it's better to implement these changes one step at a time.

rbergen, 2023-07-22:
I've completed the implementation in #370. Merging that will close this.

I did decide to rename "PresentFrame" to "PostProcessFrame". I found I got confused by which of the two meanings of "present" it concerned (temporal adjective meaning "current" vs verb meaning "display").

PR #370 (2023-07-24): Add PrepareFrame() and PostProcessFrame() to GFXBase and subclasses
## Description

This fixes #349, to the extent #365 hadn't already. It does this by:

- Adding PrepareFrame() and PostProcessFrame() virtual functions to `GFXBase`
- Overloading both in `LEDMatrixGFX`
- Overloading PostProcessFrame() in `LEDStripGFX`

These changes have been tested on a Mesmerizer board running Mesmerizer, and an M5StickCPlus running Spectrum.

Also, this:

- Declares globally defined variables that are actually only meant to be used by one compilation unit (.cpp file) `static` and gives them an `l_` prefix (i [... cut]
closed 2023-07-24 by davepl