PlummersSoftwareLLC/NightDriverStrip #355: FEATURE: Video clip playback

davepl, 2023-07-12:
This is a juicy feature for anyone up to the task!  I have a client side program called videoserver.py that's checked into the samples folder, and the matrix looks really cool playing certain videos, like a fireplace loop.

It woud be nice if we had an effect that played a YouTube video.  There are a number of code samplers on Github for playing video, but not sure if anything exists for YouTube specifically.

Short of that goal, being able to play a simple mpeg video file that we embed would also be good.

rbergen, 2023-11-28:
I'd not say it's hidden, but rather that it's a dependency. And yes, those can have defines of their own. Interestingly enough, the version of RemoteDebug we use is a PSL fork, which means we _could_ update it to source its port number from another place.

rbergen, 2023-11-28:
I just noticed this:

> See associated PR #543

I think that's actually #549?

rbergen, 2023-11-29:
Was the feature implemented in #543? This issue talks about YouTube videos or embedded MPEGs. #543 provides an implementation for animated GIFs.