PlummersSoftwareLLC/NightDriverStrip #356: FEATURE: Websocket implementation of ColorDataServer

davepl, 2023-07-12:
We currently have a tested and working socket server in ledviewer.h that can serve up the contents of the matrix to a client so they can render a preview and so on.

The problem is that to connect to it from a web page via js, it needs to be a websocket.  So the feature here is to expose the colordata on a websocket, and ideally, produce a small demo page that works with it.

rbergen, 2023-07-12:
For whoever chooses to pick this up, there is a pretty comprehensive Random Nerd Tutorial on implementing a WebSocket server here: <https://randomnerdtutorials.com/esp32-websocket-server-arduino/>. It uses the WebSocket capability of ESPAsyncWebServer, which is the webserver used to serve the on-device website. As the tutorial shows, ESPAsyncWebServer takes care of (almost) all of the "Web" in WebSocket, and brings the implementation work down to handling the data.

In case this does not end up working (for instance, because performance is too poor) and a raw implementation is deemed necessary, then the "authoritative guide" can be found on MDN: <https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API/Writing_WebSocket_servers>.

KeiranHines, 2023-11-26:
@rbergen I have proof of concept implementation of a websocket [here](https://github.com/KeiranHines/NightDriverStrip/tree/feature/websocket). If you pulled that branch and upload it, the Web UI now has a console log that mirrors the `/effects` endpoint, pushed whenever the effect is changed via the IR remote. I have a few questions I am sure you can help with. 

For ws design: 
1. Was there a preference for websocket features. I would like to see at least effect update pushes, I am not sure what else people might want. 
2. Was there a preference for playload standardization? I would prefer JSON with `{topic: <str>, payload: <Object>}`
3. Depending on 1 and 2, can we work to define the topics similar to the REST_API.md doc, just so we all agree. 
  * personally I would prefer a few smaller topic/payloads rather than one big topic/payload e.g. the current payload I send is the `/effects` endpoint. I really don't need to be sending all the effects every time. We could make a `currentEffect` topic and a `effectUpdated` endpoint with the former just pushing the current effect and interval info when the current effect changes. The later pushing just the effect that changed when an effects settings change. Any discussion around how granular to be, and how much to just keep it simple stupid would be welcome.   

For Implementation:
As for backend implementation, I have just hooked into the IR Remote at the moment because I ran into an issue getting the `g_ptrSystem` in the effectmanager.  
1. What are the overall design goals around having parts of the backend be able to fire messages on the socket vs having another "websocketmanager" style class the manages when and how to send messages. 
2. If we go with the global access to publish a message, where would be the better place to hook into that isn't the remote.

Having said all that, there is no hurry to reply to this any time soon. I will be AFK most of December so other than replying to conversation threads, nothing substantial will be done until the new year. Plenty of time to have a think and decide on a direction. 

Cheers, 
Keiran.

rbergen, 2023-11-26:
@KeiranHines Nice to see you want to get going with this!

> 1. Was there a preference for websocket features. I would like to see at least effect update pushes, I am not sure what else people might want.

The only preference from a project perspective is the one described in the issue you commented on: a WebSocket implementation of ColorDataServer, so the web UI can show what an effect would look like, without having to hook up actual LEDs/panels to the board.

As I've mentioned before myself, using them to "push" effect switches that take place on the board to the UI (because of IR remote interactions or otherwise) makes sense to me too. 

> 2. Was there a preference for playload standardization? I would prefer JSON with `{topic: <str>, payload: <Object>}`

I think the payload format _should_ be largely decided by the front-end, because that's what the WebSocket(s) is/are for. I can imagine that we actually implement more than one WebSocket (maybe one for the ColorDataServer, and one for everything the UI currently supports) with different formats; with color data the size of the payload becomes a factor as well.

> personally I would prefer a few smaller topic/payloads rather than one big topic/payload

That makes sense. It's not uncommon that push and pull scenarios use different chunk sizes.

> Any discussion around how granular to be, and how much to just keep it simple stupid would be welcome.

The suggestions you made for the two scenarios you described make sense to me. I think we just need to take this on a case-by-case basis - and certainly on something "new" like this, be willing to review earlier decisions at a later time.

> For Implementation: As for backend implementation, I have just hooked into the IR Remote at the moment because I ran into an issue getting the `g_ptrSystem` in the effectmanager.

You can use `g_ptrSystem` in `EffectManager`, but not in effectmanager.h - that creates an unsolvable circular dependency. If you want to define member functions that use `g_ptrSystem` you have to put them in effectmanager.cpp, and only declare them in effectmanager.h. There already are examples of such member functions in effectmanager.cpp at the moment.

> 1. What are the overall design goals around having parts of the backend be able to fire messages on the socket vs having another "websocketmanager" style class the manages when and how to send [...]

I think it makes sense to:

- Centralize the actual WebSocket-specific logic in one place. As long as we're using ESPAsyncWebServer as the foundation for anything "web" I'd keep it very close to the existing `CWebServer` class - although we could still separate the WebSocket part in separate source files. My suggestion would be to split if and when things become unmanageable in the source files we have.
- Create some sort of pub/sub structure, where the WebSocket handler (class) can register a listener interface (virtual class in C++) with classes that have interesting stuff to report. When something worth sending out happens in such a class, it can call one of the "On..." functions on the interface instance(s) registered with it to let the listener(s) know. Off the top of my head, my first candidates for accepting listeners and calling them would be `EffectManager` and `ColorDataServer`, for obvious reasons. Hooking up the WebSocket logic to the _sources_ of changes (like the IR remote control class) is something I wouldn't do.  
  As with everything, I would develop the listener interfaces gradually. That is: add listener functions to the interfaces in line with actual WebSocket/front-end logic to do something with the knowledge about events taking place.
 
> Having said all that, there is no hurry to reply to this any time soon. I will be AFK most of December so other than replying to conversation threads, nothing substantial will be done until the new year. Plenty of time to have a think and decide on a direction.

I hope what I mentioned above is a start. In the interest of transparency: I have thought about picking up the back-end part of this myself, but concluded that's pointless unless we have a front-end to do anything with it - and indeed have an agreement about content and format of content sent (and possibly, received).

KeiranHines, 2023-11-26:
That all sounds good to me. If you want to collaborate on this if you'd like. 
What are your thoughts on having the following sockets. (naming can be changed). Each one could easily be its own feature added over time. 
1. `/effects` pushes effect related updates similar to the `/effects` api endpoint, only push data that has changed in JSON. The frontend can unpack that data and merge it to the current state. 
2. `/colorData`: two way data to send and receive colorData, similar to the current TCP/IP socket but also with the abiliy to preview the current frame in the frontend
3. `/stats` (optional) pushes stats updates similar to the stats api endpoint. this one probably provides the least value in terms of improving existing functionality. 
4. `/debug` or /notification (optional) a wrapper around the DebugX macros to send the backend log output to the UI for debugging. Possibly also with the option to send notifications to the frontend if that was needed for some reason other than debugging.

davepl, 2023-11-28:
> "web" I'd
>
> Is that immutable? There's a perfectly lovely (actively maintained)
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html
>
> Then the AsyncSockets layer could become plain ole sockets
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html
>
>
> I know those changes aren't trivial, but if we're going to be an ESP
> project, I'd rather lean into that and get rid of anything with "Arduino"
> in the name that's largely just middleware with additional layering.

We’re married to Ardiuno, and I have several children with it already.

Why would we try to be an ESP-IDF project instead?  We’re dependent on a LOT of libs, which I assume are all in turn dependent on Arduino anyway.

Just curious! Sure, it’s lighter weight and closer to the hardware, perhaps, but what tangible benefit would a change deliver?

- Dave

rbergen, 2023-11-28:
Disclaimer: I haven't read the whole exchange - I hope to catch up after work today. I'm just quickly responding to one question that I have an opinion about.

> Is that immutable? 

Not per se, but there has to be a darned good reason to migrate. From a development perspective, I actually find ESPAsyncWebServer very pleasant to work with, in part exactly _because_ it integrates well with other Arduino projects; ArduinoJson very much being one of them.  
Looking at the examples in the esp_http_server documentation the code in our webserver implementation would have to explode to provide the same functionality we now do.

rbergen, 2023-11-28:
@KeiranHines 

> /effects pushes effect related updates similar to the /effects api endpoint, only push data that has changed in JSON. The frontend can unpack that data and merge it to the current state.

Makes sense at this level of discussion. We'd have to very clearly define (to the JSON object level) what data we do and don't send at any one point.

> /colorData: two way data to send and receive colorData, similar to the current TCP/IP socket but also with the abiliy to preview the current frame in the frontend

I think that last thing is the main reason to make a WS implementation of ColorDataServer in the first place.

> /stats (optional) pushes stats updates similar to the stats api endpoint. this one probably provides the least value in terms of improving existing functionality.

You're kind of saying it yourself already, but I don't see what the added value is of pushing this over pulling it. Unless we start raising "alerts" for certain situations, but that's a thing we have nothing in place for yet at any part of our infrastructure/codebase.

> /debug or /notification (optional) a wrapper around the DebugX macros to send the backend log output to the UI for debugging. Possibly also with the option to send notifications to the frontend if that was needed for some reason other than debugging.

I'm not sure I really see this one working yet, either. The problem is that for the Web Sockets to work a _lot_ of stuff already has to function well, and I think common types of malfunction we commonly see could get in the way of the debugging that would explain the malfunction actually reaching the web UI user.

About collaborating: I'd love to. I think the first thing to do is agree on the (data) interface between back-end and front-end, so we can then work on the implementations in parallel.

KeiranHines, 2023-11-28:
/debug would more be for debugging effect level issues. For example if you wanted to to fix a logic error in an maths heavy effect you could debug out your calculations and use the colour data to reconstruct the effect frame by frame. 

For /effect I'd start by using same json keys as the API endpoint. I'd say current effect and the interval keys should update every time the effect changes. The intervals should be sent any time the interval setting is changed. The effects list should only push changes when an effects setting that is in that object is changed. If effect indexes change I don't know if it would be best to send all effects again or attempt just to send those that change and the effectName can be used as a key to reconcile the list.

rbergen, 2023-11-28:
Hm. Then we would need to distinguish between debug logging we'd like to end up in the web UI, and the rest. I'm still leaning to finding it reasonable to ask of a developer that they hook up their device to their computer via USB, or telnet into the thing - that's also already possible.

Concerning /effect I think I can get going with that based on what you've said so far. With regards to the effect indexes, I think I'll actually only send a message with an indication which indexes changed. I think it's a scenario where it's not excessive for the web app to pull the whole effect list in response - maybe unless it finds the message concerns an index change it just triggered itself, for which the indexes should be enough. Using display names as keys is something I really don't want to do.

KeiranHines, 2023-11-28:
Instead of sending what indexes change is it better to just send an effectsDirty flag or similar. If I said flag in the front-end I will just fit the API endpoint again and refresh everything.

rbergen, 2023-11-28:
I was thinking that pulling the effect list from the API by a particular browser (tab) is a bit overkill if the effect order change was initiated by that same browser (tab). That's a scenario the web app could identify by comparing the (moved from and moved to) indexes in a WS message to the most recent effect move performed by the user using the web app, if any. 

If this is not something you'd do and you prefer to pull the effect list anyway, then an "effects list dirty" flag would indeed be enough.

KeiranHines, 2023-11-28:
> I was thinking that pulling the effect list from the API by a particular browser (tab) is a bit overkill if the effect order change was initiated by that same browser (tab). That's a scenario the [...]

Currently by memory I do pull the full list again. Mostly because in the worst csse, moving the last effect to be the first would mean every effect changes index. Granted browser side this wouldn't be a hard update, but I thought it was safer just to sync. 

> If this is not something you'd do and you prefer to pull the effect list anyway, then an "effects list dirty" flag would indeed be enough.

I think I'd prefer the flag, that way at least the code path for updating is the same for all clients and there is less chance of desync.

KeiranHines, 2023-11-29:
@rbergen just a thought, for the colorData push server to browser, I am assuming we will need to supply the color data array and an x,y for width and height of the display so the UI can render out the correct scale. Did you have a preference for this I would just default to json, but that is open for discussion. 

```
{
  data: []
  width: int
  height: int
}
```

rbergen, 2023-11-29:
> I think I'd prefer the flag, that way at least the code path for updating is the same for all clients and there is less chance of desync.

Fair enough. Dirty flag it is.

> Did you have a preference for this I would just default to json, but that is open for discussion.

As I said before, we're adding the Web Socket stuff for consumption by JavaScript, for which JSON is the lingua franca.

rbergen, 2023-12-23:
@KeiranHines So, I've managed to put an initial version of my web socket implementation together. It's in the [colordata-ws branch in my fork](https://github.com/rbergen/NightDriverStrip/tree/colordata-ws); the "web socket" meat of it is in the new websocketserver.h. I've deviated a little from the format you mentioned in one of your earlier comments. Concretely:

- I've created two web sockets. One (`/ws/frames`) for color data, and one (`/ws/effects`) for effect (list) related events. Each of these can be individually enabled or disabled using defines.
- I've added the matrix dimensions to the output of the `/statistics`/`getStatistics` REST endpoints. I've also added two flags to it to indicate if the mentioned web sockets are enabled or not.
- The `/ws/frames` socket will send out a JSON message to registered clients for each frame (provided the device can keep up) that contains an object of the following format:

  ```json
  {"colordata": [<pixel 0>,<pixel 1>,<pixel 2>,...]}
  ```
  
  Where each pixel entry is the 24-bit CRGB value for that pixel as you now receive them for colour and palette settings.
- The `/ws/effects` socket will send out JSON messages for the following situations:
  - When the current effect changes. Composition:
  
    ```json
    {"currentEffectIndex": 1}
    ```
    
  - When an effect is enabled or disabled. Composition (pretty printed here for readability):
    
    ```json
    {
        "effectsEnabledState": [{
            "index": 1,
            "enabled": false
        }]
    }
    ```
    
    Note that the format allows the enabled state to be communicated for multiple effects in one go, although right now it will always only do it for one.
    
  - When the default effect interval changes. Composition:
  
    ```json
    {"interval": 30000}
    ```
    
  - When the overall effect list has become dirty. This happens when an effect is moved, added or deleted. Composition:
  
    ```json
    {"effectListDirty": true}
    ```
    
The overall thinking behind this JSON format is that I'd like to keep the possibility to combine/bundle messages in the future. Of course, if this flies in the face of what works for you, I'm open to alternative suggestions. In any case, I'm going to pause the implementation here until you've been able to do some initial testing with it - and we either have confirmation that it works, or otherwise have an indication how/why it doesn't. 

Until then, happy holidays!

KeiranHines, 2023-12-23:
Thankyou! 
Frame looks like it will plug straight into the UI test setup I have. The effect endpoint looks more than adequate I don't see any issues with it from the details provided. I'll take a look at both in the new year. 

Happy holidays to you as well and anyone else following along.

KeiranHines, 2023-12-30:
@rbergen minor update. 

 I have started to integrate the frames socket. Its available on my [colordata-ws](https://github.com/KeiranHines/NightDriverStrip/tree/colordata-ws) branch. I have noticed periodically I seem to drop connection to the socket. I have not yet implemented a reconnect logic. I have also had three times now where I have got invalid json, normally missing a ']' at a minimum. Finally on every effect I have tested so far I have noticed the mesmerizer will restart periodically. 

I was wondering if you could try and replicate the issues on your side. I have not sure yet if its because of my development environment or something is the socket implementation. 

Any other feedback is welcome on the UI side while you are there. I am noticing some performance and render quality issues on the browser side I aim to work on those as I go but that's next years problem.

rbergen, 2023-12-30:
@KeiranHines Thanks for the update! I'll test with your colordata-ws branch when I have the time - which may also be a "next year problem" for me. Looking at the behaviour you're describing (and particularly the inconsistencies in that behaviour) I think we may be pushing the board beyond breaking point with the JSON serialization of the colour data. Without wanting to abandon that approach already, I'd like to put the question on the table if it would be feasible for you to consume a frame data format that is closer to the "raw bytes" that the regular socket sends out. Maybe an actual raw "binary" packet, or otherwise a simple Base64-encoded string thereof?

KeiranHines, 2023-12-30:
I don't see why I wouldn't be able to process a raw "binary" packet. I have dealt with base64 encoded images before so that could also work. Currently I am using a canvas to render the 'image' of the matrix. Its underlying data structure is simply a uint8 array where every 4 indices are a rgba value for the next pixel. So I am sure I can transform anything you send to that. 

It may also be worth looking to rate limit the frames sent potentially. maybe down to say 10fps just to see if that reduces the load at all.

rbergen, 2023-12-30:
@KeiranHines Yes, the frame rate limit also crossed my mind. I am just exploring options how to decrease the load if that indeed turns out to be the problem. In any case, I need to first see what logging the board spits out when the web socket is connected before we can come to any conclusion on how to move forward.

@robertlipe I'm aware of all that. :) We're trying to cross the low-level system (raw bytes galore)/web browser ("everything" is JSON) boundary here, and we're finding out what works and what doesn't as we go along. Usually this sort of exploratory development is not (or less) visible because it happens on one bench/desk, but because my back-end stuff needs Keiran's front-end stuff to be asked to do anything, it is very visible here.

I actually split off the color data web socket from the effect event web socket from the get-go to be able to change the data format for one independently of the other. As in: I was already thinking we might need to review some interface architecture decisions along the way.

KeiranHines, 2025-03-31:
Hi @rbergen I am just getting back to this tonight. I have updated my branch [colordata-ws](https://github.com/KeiranHines/NightDriverStrip/tree/colordata-ws) with the latest changes from main. I have ran the preview and it looks to at least function. I haven't ran it long enough to see if any of the above issues are sill present on the latest branches. I did notice the frontend is rather slow, the FPS is less than ideal. I will take a look at this next and see what I can do on the frontend to improve the rendering speed.

rbergen, 2025-03-31:
Part of the challenge with the WebSocket implementation is that the data that is sent from the color data WebSocket server to the client is transmitted in JSON format. The JSON is composed quick-and-dirty (manually in a string buffer) to save time, but it's still a per-pixel exercise and a LOT of I/O.

This means on the one hand that the WebSocket server does not benefit from any optimization in the regular color data server generation logic, but on the other that separate cleverness can be implemented to the JSON code path to optimize it.

KeiranHines, 2025-04-05:
I have done a bit more testing on this. I was able to get a stable version of the websocket working with a very crude rate limit on the websocket. Just simply sending everything 6th frame. I haven't yet tested going faster on any other improvements but that at least solved the disconnects which implies that a rate limit, or a performance optimization would be required to get that ws stable and production ready.

rbergen, 2025-04-05:
That's a really good insight! 

Were you able to look at the performance stats on the device while you were testing? I'm mainly wondering if it was the CPU maxing out. If so, I'm not sure there's much we can do. Any optimization on the WebSocket would require some form of compression, which requires computation by definition. On the other hand, if we have CPU cycles left then we could use those for that.

KeiranHines, 2025-04-05:
[image]

After a bit more digging it looks like the socket server will close all connections if the socket queue gets too full. There is a method `availableForWriteAll()` that I have added to the SendColorData method to skip sending if the queue is already full. That looks to have stabilised the socket fully, there is probably a few frames that are dropped here and there, but I cant see them looking at the current UI implementation. You can check the code in my branch if you'd like [colorData-ws](https://github.com/KeiranHines/NightDriverStrip/tree/colordata-ws). 

Also in that branch is a very quick and dirty performance optimization message by adding a simple string encoding to the message. I don't know if it made any real difference, I will leave it there for now, fell free to remove it if you want, just let me know and I will update the frontend to whatever the final solution is for the backend.

KeiranHines, 2025-04-05:
Thanks. I didn't notice I was still getting some periods of higher missed frames. Although that could be due to networking and nothing to do with the software. I did see the changes to main, I hadn't yet updated to them. I will update tomorrow and to the latest iteration of the socket payload schema.

KeiranHines, 2025-04-05:
My two cents is I don't mind what the end mechanism for data transfer is. Json was the fastest to implement to assess the feasibility for the feature. I think the next step would be to run some tests to see when the frame drop cause is. Is it the size of the frames, the frequency of the frames or the overall CPU load. That might determine if it's worth looking at compression to reduce packet size. Or if we need to look at ways to reduce CPU usage.

rbergen, 2025-04-05:
@KeiranHines I've created a colordata-ws-binary branch in my fork with a raw binary implementation of the color data transfer. As you can see, the implementation of that is... shorter. :)

What you will receive is an array of 3-byte RGB values, one per LED. The 3 bytes contain the red, green and blue values for the pixel in question in that exact order.

KeiranHines, 2025-04-06:
I have created a branch [colordata-ws-binary](https://github.com/KeiranHines/NightDriverStrip/tree/colordata-ws-binary) with an implementation of the frontend code to match the new format. I did notice at least on my local network I am getting a frame about every half second, with about a 8s delay total. That could be entirely my network though.

rbergen, 2025-04-06:
How does that compare to the JSON version?

KeiranHines, 2025-04-06:
Compared to the JSON version, I don't really notice any perceptible difference. I haven't setup Wireshark or similar to collect and numbers to compare. I assume some or all of the issue is networking related as its always the message queue that is full. 

I have noticed setting `-DWS_MAX_QUEUED_MESSAGES=1` has helped at least with the delay, which makes sense as the default queue length is 32 so at least with a queue length of one I am always getting the latest frame rendered. Although that is a bandaid fix

rbergen, 2025-04-06:
So then your testing indicates that the exact method of formatting the data doesn't make much of a difference. Alright, I'll checkout both of your colordata-ws branches today and test them as well. 

The non-binary branch has been updated to work with the "no quotes and pipes" JSON format?

KeiranHines, 2025-04-06:
I will update the non-binary branch now. Might take a moment to sort out the merge but shouldn't take too long.

KeiranHines, 2025-04-06:
Both branches are now up to date.

rbergen, 2025-04-06:
Great, thanks! I'll run some tests today and report back here.

KeiranHines, 2025-04-06:
Very quick comparison of json and binary, just packet timings in wireshark.  
json
[image]
binary
[image]

It does look like the binary version fragments where the json version doesn't but timing looks similar.

rbergen, 2025-04-06:
That's interesting. I'm guessing this is decided in the WebSocket implementation within our web server dependency, and it's curious that the objectively smaller binary messages are split up, also in that particular way.

Anyway, I'll let you know later what my findings are.

rbergen, 2025-04-06:
I'm also getting between what I guess is a couple of frames to four frames per second. 

Which is nowhere near the actual framerate obviously and in my opinion the visuals look way too "soft". 

However, I have to say it's still cool to see the effects in the web UI.

davepl, 2025-04-06:
Is this using the ColorData server?  With my old GUI app I could get a full 40fps out of it on the client machine, so it should be capable of it?  But then I just kind of jumped in here and aren’t sure what scenario this is!

= Dave

rbergen, 2025-04-06:
No, this is entirely different. To be able to show it in a browser we _have_ to push the color data through a WebSocket, and that basically changes everything. 

I haven't done a deep dive into the performance statistics yet, but my impression is that the overhead involved with the various layers between raw data and what is effectively a "fake" bidirectional socket between a webserver and a browser is a bit much for the limited computational processing power that these chips actually have.

We _are_ now pushing out the color data through the WebSocket in binary form which is as "lean" as we can get, but there's still 70 pages of WebSocket protocol (as described in [RFC 6455](https://datatracker.ietf.org/doc/html/rfc6455)) that has to be implemented for the whole thing to work. It seems that takes its toll. :)

davepl, 2025-04-06:
Ah, I see.  Ironically web sockets were my first attempt years ago, but they were slow and VERY buggy on the ESP32 IDF.  But they’ve no doubt fixed the bugs since then, but not sure about perf!

- Dave

rbergen, 2025-04-06:
Yeah, the bugginess seems to be okay now, but bottom-line the performance is still pretty terrible. And the code to send the data out is literally this:

```cpp
    void SendColorData(CRGB* leds, size_t count)
    {
        if (!HaveColorDataClients() || leds == nullptr || count == 0 || !_colorDataSocket.availableForWriteAll())
            return;

        _colorDataSocket.binaryAll((uint8_t *)leds, count * sizeof(CRGB));
    }
```

Now, I'm happy to be corrected on this, but I don't think there's a lot of fat on that...

davepl, 2025-04-06:
Yeah, I could run it through ChatGPT for you, but not much there to optimize :-)

I know I’m late to the party, but why web sockets?  They’re great when you’re working across a NAT where you can’t see ports, but are we really envisioning a lot of remote management?  I always assumed mgmt would be done from the local wifi, so raw sockets were ok.  But I don’t know the target scenario!

Cheers,
Dave

rbergen, 2025-04-06:
Haha, because you asked for it? The first post in this thread, as written by yourself in July 2023, reads:

> We currently have a tested and working socket server in ledviewer.h that can serve up the contents of the matrix to a client so they can render a preview and so on.
>
> The problem is that to connect [...]

And that's it. The only way to show the visuals of running effects in a web browser is a web socket.

KeiranHines, 2025-04-06:
@rbergen o reply to your comment that the graphics look soft. I do agree. Currently I am rendering the image to a canvas at the native pixel resolution of the matrix and doing a scale on that to 10x. I haven't made an attempt to do a better pixel art style scaling that keeps the pixels sharp. I can do that in the future if we get to a point where we want to use that style view for something.

rbergen, 2025-04-07:
@KeiranHines To be frank, the implementation now being as straightforward as it is now that we use the binary packages, I think the framerate we see is the framerate we're going to get through a WebSocket. 

How much work would it be to "sharpen" the images?

KeiranHines, 2025-04-07:
After a quick google, one line of code. Before and after below. Both my branches are up to date with the change.  

[image]

[image]

KeiranHines, 2025-04-07:
I did also notice tonight, I don't know if its cause, but it looks like my performance is a lot better, I checked with Wireshark and it doesn't look like my connection was upgraded to a websocket so its doing standard TCP packets. Below is Wireshark, and a gif of the UI. 

[image]

[image]

I don't have much more time tonight to look at it. But it could be worth looking into, there might be a way on the frontend or backend to restrict the websocket upgrade to something that is a lot more performant.

rbergen, 2025-04-07:
Interesting. I think the gif didn't make it, what I see is the Wireshark screenshot, twice.

What do your browser's dev tools say is going on, network-wise?

KeiranHines, 2025-04-07:
edited the comment and have included the response headers from firefox. I re-uploaded to the esp and am getting the same performance again. If you get the slower performance can you share your response header to compare.

KeiranHines, 2025-04-07:
After a few refreshes I got the slow socket again. I can't see any difference in the headers or in Wireshark as to it being a different mode or anything but I am at about 1/8th of the speed I was in the video above.

KeiranHines, 2025-04-07:
I don't know if its just a coincidence, but I am noticing I am more reliably able to get the faster mode when I have the stats tab closed. I am wondering if we are just getting close to a limit somewhere and that's causing the ESP to bottleneck. If I get time tomorrow I might pay around with disabling the other REST endpoints in the UI and see if I can get it to be more reliable if we aren't periodically querying effects and stats.

rbergen, 2025-04-07:
This is all very interesting stuff. I did notice yesterday that the stats endpoint does get called a lot as well, while I believe the stats section is not even actually updating while the effect view is open.

I'll test with the latest image myself after work (it's now 1:30PM here) and comment here with my findings.

KeiranHines, 2025-04-07:
The stats endpoint and effects endpoint are on a timer. I don't recall the exact period. So it is likely both are called more often than they need to be. 

Take your time looking into it. It's currently 9:30pm here.

rbergen, 2025-04-07:
I can concur that collapsing the stats panel makes a **massive** difference in effect visualisation framerate. It looks to me like the stats endpoint gets called once every 3-ish seconds when the stats panel is expanded.

I think the conclusion we are heading towards is that you can have the stats panel or the effect view panel open, but not both at the same time.

The effects endpoint is called far less often, and could be called even less still, once the effects part of the UI is hooked up to the effects WebSocket; that one _should_ also work. In principle, the actual effects endpoint would then only have to be called when the web UI is first opened and the "effect list dirty" event is raised. Separate events exist for changes in the active effect, effect enabled states and interval duration, and those event messages also include the new values that apply.

It did occur to me just now that the "can we write to all" check has not been implemented in the effect event sending functions within `WebSocketServer`, but that is obviously easy to add.

rbergen, 2025-04-07:
I've just added the "availableForWriteAll" check to each function that sends an effect event message in the `colordata-ws-binary` branch in my fork. I will abandon the non-binary branch from now on, as it's clear the binary version transmits far less data.

rbergen, 2025-04-07:
@KeiranHines I've opened [a small PR](https://github.com/KeiranHines/NightDriverStrip/pull/1) on your fork to apply the change mentioned in the previous comment to your `colordata-ws-binary` branch as well.

KeiranHines, 2025-04-08:
I have merged the above PR. However currently none of the other WS endpoints are used. the UI currently polls the REST API. I checked the code, the `/effects` updates every 30s. `/statistics` is every 3s by default configurable by pressing the settings cog in the top left of the stats panel. Both could be updates to support the ws push rather than polling, I don't know if that will help or hinder the performance.

rbergen, 2025-04-08:
@KeiranHines I think turning the stats into a WS push scenario doesn't make much sense, because that is effectively something that does need to be updated every few seconds to be meaningful for some of the stats. A plain REST poll is a lighter implementation for that scenario.

We could split up the stats endpoint in two: one with the static information (about half the statistics don't change after startup) that only needs to be called once, and another with the dynamic stats that does indeed get polled every handful of seconds. That should make the dynamic stats endpoint lighter in terms of resource use. 

I think it would make sense to convert the effects poll to a setup that indeed uses the effects WS endpoint that is already implemented. For one it will only cause _any_ I/O when there is something to report, and the I/O in question will be far smaller in size as well - pulling the whole effects list is quite an expensive operation. 

Of course, I'm interested in your perspective on this as well.

KeiranHines, 2025-04-18:
I have updated my colordata-ws-binary branch to use the new endpoints. I am traveling at the moment so I can only test the connection over my VPN so I cant comment if its made any noticeable difference to the performance of the web socket connection.

[19 more replies]

PR #724 (2025-06-18): Add effect view capability to web UI
## Description

This PR adds a feature to the web UI with which one can see what the device is showing on the LEDs it is driving. In terms of practical applications, it's mainly suitable for setups where the LEDs are in some sort of matrix configuration (`MATRIX_WIDTH` and `MATRIX_WIDTH` are both >1 in a meaningful way).

The following applies:

- The "regular" Color Data Server must be enabled (`COLORDATA_SERVER_ENABLED` set to 1)
- There is a delay of some seconds between what's shown on the LEDs and what the web UI shows. This is due to to the processing [... cut]
closed 2025-06-18 by rbergen