PlummersSoftwareLLC/NightDriverStrip #368: BUG: tearing in MRI/PatternMandala effect

rbergen, 2023-07-20:
## Bug report

**Problem**

The MRI effect (implemented by `PatternMandala`) shows tearing in the context of the Mesmerizer project. The best way I can think of describing the visual appearance of the tearing is an old-fashioned CRT television experiencing distortion because of the antenna not being aligned properly or a scooter driving past the house.

**Steps**

1. Boot a Mesmerizer board with the Mesmerizer project
2. Run the MRI effect and watch it

davepl, 2023-07-20:
I debugged this a while and am fairly confident this might be a bug or artifact in the FillNoise function.  It does not tear if you comment out the noise function.

- Dave

rbergen, 2023-07-25:
Closing, as this was fixed with commit b123adba128625692ab22977adb787a3ca70ea1c.

closed 2023-07-25 by rbergen