PlummersSoftwareLLC/NightDriverStrip #425: Can't set Wifi SSID/password

Rx7man, 2023-09-13:
Followed all the instructions in flashing the ESP32-S3 dev kit (Panlee is the only project available for it it seems) and I'm not able to set my wifi SSID/password, no dialog comes up, I've reset the device, reloaded the browser, etc  (Chrome) and no dice

Here's the console output 
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
E (268) esp_core_dump_flash: No core dump partition found!
E (268) esp_core_dump_flash: No core dump partition found!
Replacing Idle Tasks with TaskManager...
(I) (PrintOutputHeader)(C1) NightDriverStrip
(I) 
(I) (PrintOutputHeader)(C1) ------------------------------------------------------------------------------------------------------------
(I) (PrintOutputHeader)(C1) M5STICKC: 0, USE_M5DISPLAY: 0, USE_OLED: 0, USE_TFTSPI: 1, USE_LCD: 0, USE_AUDIO: 0, ENABLE_REMOTE: 0
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: OK
(I) (PrintOutputHeader)(C1) Version 40: Wifi SSID: "" - ESP32 Free Memory: 287632, PSRAM:2095055, PSRAM Free: 2094659
(I) (PrintOutputHeader)(C1) ESP32 Clock Freq : 240 MHz
(I) (setup)(C1) Startup!
(I) (setup)(C1) Starting DebugLoopTaskEntry
>> Launching Debug Thread.  Mem: 287632, LargestBlk: 262132, PSRAM Free: 2094659/2095055, >> Launching JSON Writer Thread.  Mem: 279064, LargestBlk: 262132, PSRAM Free: 2094623/2095039, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (setup)(C1) Starting ImprovSerial
(W) (ReadWiFiConfig)(C1) Retrieved SSID and Password from NVS: , ********
(W) (setup)(C1) Creating TFT Screen
[  1057][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (793) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1068][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (805) gpio: gpio_set_level(226): GPIO output gpio_num error
E (816) gpio: gpio_set_level(226): GPIO output gpio_num error
E (837) gpio: gpio_set_level(226): GPIO output gpio_num error
(W) (InitializeHardware)(C1) Allocating LEDStripGFX for channel 0
(I) (AddLEDsToFastLED)(C1) Adding LEDs to FastLED...
(I) (AddLEDsToFastLED)(C1) Adding 1152 LEDs to pin 5 from channel 0 on FastLED.
(W) (SetupBufferManagers)(C1) Reserving 312 LED buffers for a total of 1088256 bytes...
(W) (InitEffectsManager)(C1) InitEffectsManager...
(I) (InitEffectsManager)(C1) Creating EffectManager using default effects
>> Launching Drawing Thread.  Mem: 265900, LargestBlk: 249844, PSRAM Free: 986607/2084863, (W) (DrawLoopTaskEntry)(C1) >> DrawLoopTaskEntry
(W) 
(W) (DrawLoopTaskEntry)(C1) Entering main draw loop!
>> Launching Screen Thread.  Mem: 261428, LargestBlk: 245748, PSRAM Free: 986607/2084863, (I) (setup)(C1) Calling ConnectToWifi()
(I) 
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...WL_NO_SHIELD
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...
```

On a side note, if you have the text size increased in your browser you can't see the whole page and there are no scroll bars.. I got that figured out pretty quick though 
[image]

rbergen, 2023-09-13:
@Rx7man Thanks for flagging this. Lacking the device in question, I unfortunately can't replicate or debug the WiFi credential setting issue at my end. 

What I can say is that the logging looks "healthy" to me, at least to the point that setting SSID and password _should_ be possible - specifically the line stating that ImprovSerial is starting is relevant in this context.

The mention of the text sizing/scrollbar issue is very useful. I will look into that.

smiliea, 2023-09-13:
I would start by deleting the cached .json files in the .pio directory. I
believe they get stored (encrypted?) in one of these files the first time
the pio project is compiled, and if you change the SSID or password, the
old SSID gets used.

I was having similar issues for a few weeks, however my solution was to do
a factory reset on my wireless router. Hopefully you won’t need to do
this!  I had done all I could think of and that was my last resort.

Rx7man, 2023-09-13:
> I would start by deleting the cached .json files in the .pio directory. I believe they get stored (encrypted?) in one of these files the first time the pio project is compiled, and if you change the [...]

Where do I find this file?  It's the first time I do anything with web install

This is the specific device I got, https://www.amazon.ca/dp/B0B6HT7V7P?psc=1&ref=ppx_yo2ov_dt_b_product_details

I tried installing it on a different model (ESP32-wroom) and I get the same thing, no place to ever enter the SSID
[image]
[image]

That device doesn't reconnect to the terminal automatically when I reset it so I wasn't getting the boot messages, connected with Putty and got them.. 

Above my pay grade to implement, but would it be possible to enter the SSID/Password over serial?
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (585) esp_core_dump_flash: No core dum▒▒▒ѥѥ▒▒▒found!
E (585) esp_core_dump_flash: No core dump partition found!
Replacing Idle Tasks with TaskManager...
                                        (I) (PrintOutputHeader)(C1) NightDriverStrip
(I)
(I) (PrintOutputHeader)(C1) ------------------------------------------------------------------------------------------------------------
(I) (PrintOutputHeader)(C1) M5STICKC: 0, USE_M5DISPLAY: 0, USE_OLED: 0, USE_TFTSPI: 0, USE_LCD: 0, USE_AUDIO: 0, ENABLE_REMOTE: 0
(I) (PrintOutputHeader)(C1) Version 40: Wifi SSID: "" - ESP32 Free Memory: 215336, PSRAM:0, PSRAM Free: 0
(I) (PrintOutputHeader)(C1) ESP32 Clock Freq : 240 MHz
(I) (setup)(C1) Startup!
(I) (setup)(C1) Starting DebugLoopTaskEntry
>> Launching Debug Thread.  Mem: 215336, LargestBlk: 110580, PSRAM Free: 0/0, >> Launching JSON Writer Thread.  Mem: 206732, LargestBlk: 110580, PSRAM Free: 0/0, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (setup)(C1) Starting ImprovSerial
(W) (ReadWiFiConfig)(C1) Retrieved SSID and Password from NVS: , ********
(W) (InitializeHardware)(C1) Allocating LEDStripGFX for channel 0
(I) (AddLEDsToFastLED)(C1) Adding LEDs to FastLED...
(I) (AddLEDsToFastLED)(C1) Adding 1152 LEDs to pin 5 from channel 0 on FastLED.
(W) (SetupBufferManagers)(C1) Reserving 12 LED buffers for a total of 41856 bytes...
(W) (InitEffectsManager)(C1) InitEffectsManager...
(I) (InitEffectsManager)(C1) Creating EffectManager using default effects
>> Launching Drawing Thread.  Mem: 151212, LargestBlk: 106484, PSRAM Free: 0/0, (W) (DrawLoopTaskEntry)(C1) >> DrawLoopTaskEntry
(W)
(W) (DrawLoopTaskEntry)(C1) Entering main draw loop!
(I) (setup)(C1) Calling ConnectToWifi()
(I)
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...WL_NO_SHIELD
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set...

```

smiliea, 2023-09-14:
I don’t know why this went over my head, but I didn’t realize that your
original post was regarding the web installer.  I have used the web install
interface only a few times, and I am pretty sure I ran into this same issue
with the SSID and Wi-Fi password, so I just moved to using VSCode and
PlatformIO and didn’t spend too much time on it.  I think you will find you
have much more control setting up, building and deploying going this route.
However, this definitely needs to be addressed. The web install is a great
tool and is going to get people started with this project quick.

I will see if I can replicate this issue with one of my “HiLetgo ESP-32
Wi-Fi+Bluetooth” dev boards and see if I can help out.

rbergen, 2023-09-14:
A fundamental challenge with the Web Installer is that the web application is entirely dependent on the [ESP Web Tools](https://esphome.github.io/esp-web-tools/) project for the actual flashing of images and the subsequent configuration of WiFi details. That is to say that "at the browser side", everything beyond the logic to select a device and project is effectively a black box. 

The configuration of WiFi details requires the implementation of the "improv_serial" protocol in the NightDriverStrip image running on the board. The implementation that's in place (in the file [improvserial.h](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/include/improvserial.h) to be precise), is based on [the improv_serial implementation in ESPHome](https://github.com/esphome/esphome/tree/dev/esphome/components/improv_serial). I'm not sure exactly what version of the ESPHome implementation ours is currently based on, but it's possible that ours has become misaligned with the current state of ESP Web Tools over time.

I think assessing this would require a "line-by-line" comparison of our improv_serial code to the one in ESPHome. That's a job I'd personally describe as "not a lot of fun when you're at it, but **very** rewarding if it solves the problem."

The other approach - which one could apply first - is adding some low-level debug logging to our improvserial.h as it stands. That should bring to the surface any situations where we expect a different set of bytes to come in than we're getting. That might lead us on a path to fixing that, maybe by increasing our fault tolerance and/or improving our recovery logic.

Rx7man, 2023-09-14:
I'm not in a rush to play with this, so it's a back burner project.. part of the reason I figured trying the web installer would be a neat thing to try.  I haven't played much with VS Code, only used Visual studio with Visual Micro for Arduino programming.

rbergen, 2023-09-14:
I'm happy to pick this up myself to the extend that I can make the changes blind. The "problem" is that I can't test the scenarios that are now failing. That is to say that all the devices I have do allow their WiFi credentials to be set via the Web Installer. Without someone willing and able to test the changes I (or anyone else) would make, making them is pretty pointless.

Rx7man, 2023-09-14:
I'd test them.. just seems strange that neither of my devices work seeing they're the most common ESP32's

KeiranHines, 2023-09-14:
I noticed when I flashed my mesmerizer I had to connect it to the web installer. Close the dialog and reopen it a second time for the wifi credentials option to show, sometimes it only showed the open logs button. 

You could possibly try connecting to the web installer, leaving the device connected for a minute or two, then trying again.

Rx7man, 2023-09-16:
I have both the ESP32-wroom and the ESP32-S3-wroom-1 (a couple of each) and neither work.. 
I never mentioned it but I am on the latest version of Chrome and Windows 11

I don't know how to do a memory dump but I'll gladly do that if it helps and you walk me through it

rbergen, 2023-09-17:
This may just be an oddness in my personal wiring, but I think that for me it will be most helpful right now if I can see exactly what comes into the board from the browser at the (improv_)serial level, and what it tries to send out. There is debug logging code in the relevant files at the moment, but that's currently not switched on. 

When I have the time, I'll create a branch with low-level improv_serial debug logging switched on in a way that it outputs to the same channel as the rest of the logging we've been looking at so far (i.e. the console). Once we can see what the firmware receives and doesn't from the browser at a character/byte level, we can hopefully figure out what's misaligned with what at the moment.

rbergen, 2023-09-20:
I've spent a few more brain cycles on this and it dawned on me that just enabling debug logging is not going to work, because when Improv is supposedly trying to do its thing concerning setting up WiFi etc., no logging will actually come through. So, what I'll do instead is have the Improv logging persisted to SPIFFS (when told to do so), and update the on-board web server to allow that log to be retrieved. As this is something we may very well need again in the future, I want to implement it "properly". That does mean it will take me a little longer to finish this.

I'm currently finishing off some work to add a new set of effects to the Mesmerizer project, and I'll pick this up after.

rbergen, 2023-09-20:
I just merged #426 which fixes the "missing scrollbar" sidenote issue. It'll show in the Web Installer when CI finishes - it includes the deployment to GitHub Pages.

rbergen, 2023-09-22:
@Rx7man I've created an "add-improv-logging" branch in my fork (<https://github.com/rbergen/NightDriverStrip/tree/add-improv-logging>) that includes the ability to write Improv debug logging to SPIFFS, and retrieve it via the webserver. 

The capability can be enabled by defining ENABLE_IMPROV_LOGGING towards the top of improvserial.h, or passing -DENABLE_IMPROV_LOGGING=1 as a build flag.  
Once the log has been generated, it can be retrieved from the device using http://&lt;device IP&gt;/improv.log

Disclaimer: it's untested - there's not much point in me doing so, because with my gear Improv basically works as it should.

Maybe you can give it a run with ENABLE_IMPROV_LOGGING set to 1 when you have a few minutes, try to config WiFi with the Web Installer, and let me know whatever logging that comes out of that?

Rx7man, 2023-09-22:
I am a total noob at this, but I tried installing again and now the logs only say "Terminal disconnected: Break received"
I'm not sure how to set flags, etc on the web installer
This is what I got.  I unplugged the device and did a hard reboot, same thing

[image]

Rx7man, 2023-09-22:
Tried it on the ESP32-S3 and got the "can't connect"

[esp-web-tools-logs (1).txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/12704300/esp-web-tools-logs.1.txt)

rbergen, 2023-09-23:
My apologies, I think we need to take a step back here. 

The version of NightDriverStrip that I created can't be installed using the Web Installer. It needs to be built and uploaded via PlatformIO. That means that:

- The source code needs to be downloaded from the "add-improv-logging" branch in my fork. You can downloaded it in a ZIP file from the URL in my previous comment: https://github.com/rbergen/NightDriverStrip/tree/add-improv-logging
- After unpacking the ZIP file, the 0 in line 46 of file include/improvserial.h (`#define ENABLE_IMPROV_LOGGING   0`) needs to be changed to a 1. 
- PlatformIO has to be installed and used to build your downloaded and now modified version of NightDriverStrip. You can find information about this in the "Build pointers" section in the README in this project. Note that your WiFi network details need to be specified before the build, using the instructions in the "Wifi setup" section in the same README.
- The firmware image that the build creates needs to be uploaded to your device over USB using PlatformIO as well.

When this is done, your device should boot with the right version of NightDriverStrip, and connect to WiFi on startup. With all that in place and it (still) connected to USB, you can now access the Web Installer again, and try to (re)configure WiFi on your device using it. While you do this, a log file should be created on your device that documents exactly what data is sent back and forth between your web browser and the device. When you're done testing, that log file can be retrieved using the IP address your device was issued by your WiFi access point, via the following URL: http://<device IP address>/improv.log. That's the log file I would need to hopefully be able to analyze what's going on.

Rx7man, 2023-09-23:
I'm really not good with command line stuff, but I just can't get it to find PIO.exe, followed all the instructions to add it to the path and it's just not working.. I have some things to do today and I'll get back to this later.. I did make a copy of secrets.h and put my wifi info into it, as well as modified the improvserial.h to debug= 1

[image]

rbergen, 2023-09-23:
If you install the PlatformIO IDE (i.e. the VSCode plugin) then you don't need to separately make the command-line tools available. They'll be part of the package and available if you open a PlatformIO Core CLI terminal through the plugin in VSCode. 

Even better: with the VSCode/PlatformIO combo, you can do the whole thing of building (including environment/project selection) and uploading (including port selection) from within VSCode.

Rx7man, 2023-09-23:
I did install platformIO through VScode.. 
I'm also getting the red squiggly line under the improv.h include.. I see it there in the directory with everything else
[image]

Rx7man, 2023-09-23:
It sees Hexdump.h but not improv.h 
[image]

rbergen, 2023-09-23:
I've learned that red squigglies (or the absence of them) don't mean too much until you've actually performed a build. So, my recommendation would be to use VSCode/PlatformIO to create a build for the environment/project of your choosing, and see how far you get with that.

Rx7man, 2023-09-23:
I'm not sure where to go from here and what to do.. perhaps this is a little over my pay grade

dosipod, 2023-09-23:
We are working a lot with open source firmware and can confirm that the site  https://plummerssoftwarellc.github.io/NightDriverStrip/ does not work with the same issue the OP has posted , this is the issue faced with multiple esp32.
To compare ( or for devs to borrow the correct code for web installer ) try the following open  source web installer 

WLED https://install.wled.me/v2.htm or https://wled-install.github.io/   , source is  (https://github.com/Aircoookie/WLED and web installer source is  https://github.com/Aircoookie/WLED-WebInstaller )

Tasmota 
https://tasmota.github.io/install/

rbergen, 2023-09-24:
>I'm not sure where to go from here and what to do.. perhaps this is a little over my pay grade

@Rx7man That's ok. I'll just have to wait for someone else with a non-functioning chip to take note of this issue and generate the logging I'm asking for.

Rx7man, 2023-09-24:
So just a little about me, I'm a farmer that's worked with PLCs in industrial environments, done a little bit of VB programming all my life, make gizmos for myself with arduino-ish stuff (VGT turbo controller, etc), but when it comes to "real" programming I am out of my depths
I have both the ESP32-wroom and ESP32-S3 and neither of them even give me a prompt to enter the wifi SSID/password (Win11/chrome).. My wifi is terribly insecure as far as passoword strength.. because there's just no one around except me.. I could try this on my Windows 7 computer too I guess

I'm in Canada if that helps..  I think I'm behind 2 NAT's on my internet so I don't think I could set something up like remote desktop
I'm just in the process of moving so my hands are pretty full, but I'd love to help where I can..

Rx7man, 2023-09-24:
I haven't tested the others, but it's kinda strange that two different model board I have don't work (and I did upload some other arduino sketches to test the wifi and they work just fine)
Yes, I am a spinning doritos guy 
I'll go through your entire message tomorrow when I'm a bit more awake

rbergen, 2023-09-24:
I'm pretty sure there's nothing wrong with the boards. If memory serves me, there have been other reports of S3 owners running into trouble when trying to use the Web Installer. The report on the WROOM _is_ new, IIRC.

I'm basing this on nothing but intuition (which has certainly been wrong before), but I'm guessing there is some "static on the line" between the browser/ESP Web Tools and the Improv protocol implementation in NDS. Maybe some characters get lost, maybe we're not processing a message type we should, maybe we're being confronted with an Improv protocol version we don't know and ignore, etc. etc.  
That's what the new logging should tell us.

> (W) (setup)(C1) Starting ImprovSerial
> (W) (ReadWiFiConfig)(C1) Error (ESP_ERR_NVS_NOT_FOUND) opening NVS handle!
> (W)
> (W) (setup)(C1) Could not read WiFI Credentials
> (W) (WriteWiFiConfig)(C1) Stored SSID and Password to NVS: , *******

The NVS partition isn't there because the Web Installer fully erased the board's storage before flashing it. Upon determining this (i.e. immediately after reporting the error), the NVS partition is created and formatted on the fly. Which is why the storing of the default empty SSID (and password) to NVS _does_ work "4 lines later".

On subsequent boots the (empty) SSID _is_ loaded from NVS, but obviously still equally useless.

In all cases, the fact that the SSID is empty then puts the networking logic in "waiting mode" for WiFi credentials to be provided via the Web Installer interface/Improv. And that's where things get stuck at the moment.

rbergen, 2023-09-24:
> Rutger, does "WL_NO_SHIELD" seem alarming?  That might make some sense in
an Arduino environment where you have to add a board (and they can't call a
board a board, it's a "shield" in their programs, err, sketches.) but the
absence of finding networking hardware on an ESP32 is pretty alarming since
that's literally their claim to fame - including WiFi on the chip.

No, not in the particular spot it shows up. This is what I get on my perfectly functional Mesmerizer and M5StickC Plus as well. I think the code may be trying to set the hostname too soon for the WiFi library's liking, which is something I may look into further at some point. 

> ```cpp
>       debugW("Retrieved SSID and Password from NVS: %s, %s", WiFi_ssid.c_str(), "********");
> ```
>
> but the info printed a blank for the first. There's no SSID!?!?

Correct. The Web Installer flashes images with an empty SSID, because it can't know what SSID to use. With the Web Installer, the WiFi creds are supplied _after the firmware flash_. In fact, the absence of an SSID puts the board in a "waiting mode" for Web-Installer-supplied WiFi credentials.

> Rutger, is there any code in this path that should show nearby APs and
signal levels just to confirm it's able to see the network radio space at
all?

As part of the Improv implementation that _should_ be handling the setting of WiFi creds via the Web Installer, there is indeed logic to identify nearby APs (but I don't think signal levels; I'd have to look at improvserial.h to check). A list of nearby APs is actually sent back to the Web Installer so the user can choose which one to connect to, if everything works as it should.

(I noticed you mentioning me in your earlier comment more or less by accident. Tagging me with an "at handle" will work better - as in by using @rbergen instead of my first name. GitHub will then send me a specific "you were mentioned" notification.)

rbergen, 2023-09-25:
I've gone over the Improv code for the umpteenth time just now, and there is a (small) chance I may have found something that can explain the erroneous behavior of the Web Installer with S3-based boards. 

Long story short: within the context of ESP Web Tools, the ESP-S3 is a different chip family than the "plain" ESP32. At the moment, the Improv implementation in NDS always identifies itself as an ESP32, due to a fixed string being passed to the `ImprovSerial::setup()` function in main.cpp (line 325, to be precise). It's possible that after flashing, ESP Web Tools is looking for a device identifying as an ESP32-S3, finds nothing, and concludes there's nothing it can do.

I'll make some updates to my "add-improv-logging" branch so that S3 projects identify themselves as ESP32-S3 via Improv. I'll then open a PR to merge that branch. When that's done, @Rx7man can test if that allows him to configure WiFi after a reflash of his S3 with the Web Installer - this is a test that does not require a build or firmware flash to take place via PlatformIO.

To be continued!

Rx7man, 2023-09-25:
I'll gladly test it!

I'm just wondering if there's something in my computer setup that's preventing the select SSID dialog box from popping up, I've disabled uBlock for the site though

rbergen, 2023-09-27:
@Rx7man I appreciate you considering the possibility that the problem is at your end, but I'm pretty sure that's not where the issue lies. If the flash dialog does show up, the SSID dialog should also.

The new images have been pushed to the Web Installer, so you could give it another try. Before you do, please consider clearing your browser cache for the Web Installer domain (plummerssoftwarellc.github.io) at least, to ensure you're using the new binaries.

Rx7man, 2023-09-27:
Nope, still not working 

`Logs
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
E (268) esp_core_dump_flash: No core dump partition found!
E (268) esp_core_dump_flash: No core dump partition found!
Replacing Idle Tasks with TaskManager...
(I) (PrintOutputHeader)(C1) NightDriverStrip
(I) 
(I) (PrintOutputHeader)(C1) ------------------------------------------------------------------------------------------------------------
(I) (PrintOutputHeader)(C1) M5STICKC: 0, USE_M5DISPLAY: 0, USE_OLED: 0, USE_TFTSPI: 1, USE_LCD: 0, USE_AUDIO: 0, ENABLE_REMOTE: 0
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: OK
(I) (PrintOutputHeader)(C1) Version 40: Wifi SSID: "" - ESP32 Free Memory: 287632, PSRAM:2095055, PSRAM Free: 2094659
(I) (PrintOutputHeader)(C1) ESP32 Clock Freq : 240 MHz
(I) (setup)(C1) Startup!
(I) (setup)(C1) Starting DebugLoopTaskEntry
>> Launching Debug Thread.  Mem: 287632, LargestBlk: 262132, PSRAM Free: 2094659/2095055, >> Launching JSON Writer Thread.  Mem: 279064, LargestBlk: 262132, PSRAM Free: 2094623/2095039, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (setup)(C1) Starting ImprovSerial for ESP32-S3
[  1259][E][vfs_api.cpp:182] remove(): /improv.log does not exists or is directory
(W) (ReadWiFiConfig)(C1) Retrieved SSID and Password from NVS: "", "********"
(W) (setup)(C1) Creating TFT Screen
[  1268][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1004) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1279][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1016) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1028) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1049) gpio: gpio_set_level(226): GPIO output gpio_num error
(W) (InitializeHardware)(C1) Allocating LEDStripGFX for channel 0
(I) (AddLEDsToFastLED)(C1) Adding LEDs to FastLED...
(I) (AddLEDsToFastLED)(C1) Adding 1152 LEDs to pin 5 from channel 0 on FastLED.
(W) (SetupBufferManagers)(C1) Reserving 312 LED buffers for a total of 1088256 bytes...
(W) (InitEffectsManager)(C1) InitEffectsManager...
(I) (InitEffectsManager)(C1) Creating EffectManager using default effects
>> Launching Drawing Thread.  Mem: 265896, LargestBlk: 249844, PSRAM Free: 986607/2084863, (W) (DrawLoopTaskEntry)(C1) >> DrawLoopTaskEntry
(W) 
(W) (DrawLoopTaskEntry)(C1) Entering main draw loop!
>> Launching Screen Thread.  Mem: 261424, LargestBlk: 245748, PSRAM Free: 986607/2084863, (I) (setup)(C1) Calling ConnectToWifi()
(I) 
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...WL_NO_SHIELD
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 1 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.
(W) (ConnectToWiFi)(C1) Waiting for WiFi credentials to be set. Pass 2 of 99...
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(W) (ConnectToWiFi)(C1) WiFi credentials not set, cannot connect.`

rbergen, 2023-09-28:
This is where someone like me says "ah well, it was worth a shot".

I think we're back to waiting someone with an S3 who can generate the logging that has now also been enabled. Regardless, thanks very much for trying!

Rx7man, 2023-09-28:
as a contribution to the project I'd send you one if that would help

rbergen, 2023-09-28:
That's very kind of you! If you're sure (and please make sure that you are) then I think it's best to reach out to me via email. My address is my GitHub handle followed by "at" and xs4all.nl.

Rx7man, 2023-09-28:
mine is my handle at gmail

Rx7man, 2023-09-29:
I don't know if anything has changed in the ESP32-wroom side of things, but I got 2 other, very similar wroom boards in the mail yesterday, and both of those, and the one I was using before are getting "Terminal disconnected - Break received" messages on the logs, and then something is hanging the serial port, after some combination of resetting the device and reloading the page or pressing the BOOT button I'll finally be able to get it back.. 
Of course now that I've typed that it actually worked.. once 
[image]
[image]

Here's the 3 boards, on the left is the original one I was using, then two variants
[image]

rbergen, 2023-10-23:
Ok, I think I found the problem, and it's an interesting one. In short:

- Panlee is a PlatformIO project (environment) that defines `LEDSTRIP` to be 1. 
- The respective block in globals.h in turn defines `WAIT_FOR_WIFI` to be 1.
- When `WAIT_FOR_WIFI` is 1, the functions that tries to connect to WiFi during setup (`ConnectToWiFi()` in network.cpp) blocks and loops (invoking `delay()`), explicitly waiting for credentials until WiFi is connected.
- This means that the main `setup()` function never finishes, and the `loop()` function is never invoked.
- That means Improv commands will never come in as they are polled for in the main `loop()` function, and those commands are the _only_ means to set WiFi credentials on a device that doesn't have any configured already.

So, when `WAIT_FOR_WIFI` is 1, the device gets caught in a catch 22 that it can never get out of.

In my eyes, there are three ways to solve this:

- Remove all projects that lead to `WAIT_FOR_WIFI` being 1 from the web-flashable projects. That does mean that such projects are dead in the water if no (valid) WiFI credentials are configured through secrets.h.
- Add a separate task (thread) for the Improv loop, and kicking it off before the WiFi connect in `setup()` is started.
- Stop blocking `setup()` if WiFi can't connect, potentially effectively removing the `WAIT_FOR_WIFI` feature. 

My preference would be the latter as I think it's the cleanest, but there are very specific reasons @davepl implemented things this way so I'll have to align with him.

All of this also means that devices will become (mostly) unresponsive to Improv while the (first) WiFi connection is attempted (which may well be why the "show log and back out" approach often works), or if the WiFi credentials that were entered are wrong. Fixing this would again require a separate Improv thread or an overhaul of the WiFi connection logic. I have some unfinished thoughts on the latter, which I'll develop further when I have the time.

To be continued.

rbergen, 2023-10-24:
The problem with Improv is not that the response packets don't get through. I know for sure that they do, because when I open the Logs & Console window in the web installer after the logging on SPIFFS confirms that the response packets have been sent, they somehow do show up in the logging shown. Why the actual Improv logic in the web browser doesn't pick them up before I push that button I don't know.  
I can't help thinking I must be overlooking something that's glaringly obvious, which is quite frustrating.

> Are you getting command.ssid as Strings, asking for the c_str() const char* string_view, and then promoting that back to a String class?

Yes, and I thought you might take me up on it. 🙂 This is where I chose to stick to the "we store strings in `Strings`" de facto project approach, except where I had a specific reason not to (that being the `ConnectToWifi()` arguments, where it was useful to be able to give `nullptr` meaning). I note that I may still change my mind on this one, as this is still WIP until it actually does what it should.

rbergen, 2023-10-24:
Well, talking about things **I** don't like: I just decided to reboot my laptop, after which Improv seems to work like a charm - &lt;insert Windows joke here&gt;. 

I've pushed the latest state of polish-webprojects to my staging branch, from where it will be pushed to my fork's web installer in an hour and a half or so. I'll test from scratch using that, and hope to confirm I just ran into a glitch in the system yesterday - which may well have been caused by the Improv implementation on the ESP32 misbehaving until I fixed it.

**If** that confirms things actually work, I'll reconsider the `String` ➡️ `const char *` ➡️ `String` deal, and see if I can find a less cringeworthy solution for that one. (Conceptually `std::optional` comes to mind, but storing references in that is not possible and solving that with `std::reference_wrapper` seems overly complex. Maybe I'll just use pointers to `String`s. Anyway, as one can tell the jury is still out.)

rbergen, 2023-10-24:
Alright. I've done a reflash-with-erase of Panlee on the ESP32 S3 using the staging web installer, and I've been able to set up WiFi on it after the flash. It did require a reboot of the device and a browser restart, but that's not out of the ordinary. 

I'll confirm Improv also works on the M5StickC Plus and Mesmerizer using this particular image. If that works I'll do any code clean-up I should as a good citizen and open a PR to merge this afterwards.

rbergen, 2023-10-24:
Improv confirmed working on the M5StickC Plus and Mesmerizer as well.

dosipod, 2023-10-26:
Not sure why all the MCUs ( all normal esp32  ) i have do not work with https://plummerssoftwarellc.github.io/NightDriverStrip/ but works fine with web installer of Tasmota , wled ,openmqttgateway and ESPEasy . So this is general issue to web flashing . I was able to run the project on the same MCUs only by compiling and hard coding the SSID and pass  .

And not sure of the change of the post subject as it seems the demo and ledstrip project also do not flash correctly from site 

[image]

rbergen, 2023-10-26:
@dosipod I think @robertlipe has done a great job explaining the reason for the title change with context, and as the guy making the change I can confirm that:

- The change reflects that the report concerns a specific issue (inability to configure WiFi credentials) on a specific project (Panlee), on a specific device (ESP32-S3). The latter part we get for free and don't have to mention because that's the device Panlee is configured for. The issue was confirmed locally by me, and then confirmed fix by me as well after applying the fix that is the primary content of PR #469.
- Builds that do not have WiFi enabled will not offer the facility to configure WiFi credentials via the web UI; the Improv protocol loop is not executed on those builds (as per [this section in the main program loop](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/5a78b33ae34c59d36f085b71336acc77cc5e714a/src/main.cpp#L530-L535)) because there is no point in doing so.
- If you or anyone else experience other issues than the one this one is now addressing, then a new bug report Issue is the way to let us know. And yes, details of exactly what does and does not happen, including captures of logging via serial if available will then be extremely helpful.

dosipod, 2023-10-26:
Forget the fact that we can  install on the same MCUs from other sites . I can also open a new issue as the aim is to help improve things and not just point out the site is not working ( which is unfortunately the case )  . The important thing is if you do not see issues in flashing via site then that is more then enough for me . I do also have s3 but only one and not sure it would be best to report an issue if I can not replicate on multiple MCUs . If there is anything you would need from normal esp32 flashing logs then that i could do

rbergen, 2023-10-26:
@dosipod At the moment I would actually wait a little while until #469 is merged - at which time this issue will be closed. The fix in that PR should address the WiFi configuration issue on all builds that are based on the Ledstrip configuration - including Panlee on the S3, and Ledstrip on a variety of other devices.  
As mentioned, the Demo project won't allow WiFi configuration on the ESP32, the M5StickC or the M5StickC Plus, in any case. On the other devices with a Demo project WiFi does seem to be enabled for it, so credential configuration should work.

dosipod, 2023-10-26:
@rbergen Thank you yes it is clear now about the demo project  , i could wait no issue as i am really just exploring leds projects and will only try Ledstrip after the merge . I do have a lot of fixtures and MCUs which I hope i could demo for this project as i did with other projects  . Appreciate the quick response

PR #469 (2023-10-28): Update WiFi connection logic and improve Improv logging
## Description

This:

- Changes the WiFi connection logic to not block the main `setup()` function.
- Makes the stored WiFI credentials (which we consider sensitive) local to the only function that really uses them, instead of global variables that are available "everywhere".
- Maintains the possibility to demand functioning WiFi connectivity at the penalty of a reboot if it's down for too long.
- Makes the hostname a device config setting, and makes the hostname consistent across use cases (WiFi, OTA, RemoteDebug).
- Significantly impro [... cut]
closed 2023-10-28 by rbergen