PlummersSoftwareLLC/NightDriverStrip #457: Problem with storing / recalling effects?

revision29, 2023-10-08:
## Bug report
I have noticed that when I erase the flash on a device and re-upload a fresh compile which loads the rainbow effect it stretches a rainbow over the whole LED strip (depending on the speed one sets the effect to). The rainbow shifts across the length of the strip and wraps around and pulses as it moves. In other words, most pixels are different colors and you can see the rainbow shift linearly.  If I power cycle the device, the effect is not the same. It draws all pixels in the strip as the same color. As it fades to the next color in the spectrum, all pixels are the same color.

This is not the desired behavior.

I don't have the knowledge / familiarity with the codebase to figure this out. It seems there is either a problem in how the effect is stored as json blobs or after loaded from json on boot it is not being triggered properly. I suspect this is happening to more than just the rainbow effect. The rainbow effect is more readily noticeable. Since the effect is drawn as expected when it's loaded directly from code, the problem seems to be in the json processing code. 

**Problem**

The rainbow effect breaks after a reboot.

**Steps**

1.  Install a fresh compile of the demo environment on a freshly erased device.
2.  Watch the beautiful rainbow glide across the strip.
3.  Reboot the device.
4. Watch in horror as the rainbow no longer shifts. All pixels show the same color as they all shift from hue to hue. 
5. Spend hours testing different boards, checking wiring, double checking code.

revision29, 2023-10-08:
I just tested this with a fresh branch from the main repo and it breaks the same way. It actually fails to load the effect properly after the second reboot. If I comment out line 498 in effects.cpp (return jsonObject;) it will draw as expected. I realize that this prevents the effects stored through the json code from loading into the effect manager. This confirms there is something wrong how json handles the effects.

rbergen, 2023-10-08:
@revision29 

> This confirms there is something wrong how json handles the effects.

This may sound pedantic, but it confirms there is something wrong with how this effect serializes  and/or deserializes itself. It's up to each individual effect to save and load enough of its state to and from JSON to be able to restore itself after a reboot. Serialization happens in the effect's `SerializeToJSON()` function, deserialization in the effect's constructor that accepts a `JsonObjectConst` as its only parameter. Apparently the effect in question (you didn't mention a class name, so I'd have to make an educated guess about what it is) does not save and/or load everything it needs to "reincarnate".

Is there anywhere (like a published branch on GitHub) I can see _exactly_ what code you are running when this happens?

rbergen, 2023-10-08:
Also, are WiFi and the web sever enabled on your device? If so, it would be very helpful to see the contents of http://&lt;device IP&gt;/effectsConfig before and after the reboot that breaks things.

revision29, 2023-10-08:
The effect is the RainbowFillEffect in misceffects.h around line 151. It’s that way on the main branch. I made a fresh branch off the main code base just to confirm I did not break it.Don’t worry about being pedantic. You are very familiar with the code base and I’m not. I had no intention of learning more about effects and how they work but here I am.I’ll look at the effectsConfig later this afternoon. I was wondering how I would look at the stored version of the effect. You might have given me enough information to attempt to fix it. It’s a minor issue that I’m surprised did not get noticed until now.

revision29, 2023-10-08:
Here is what I have found: the json blob is missing a variable for delta hue. That results in all of the LEDs being the same color. I can manually edit the code to use a set integer instead of the missing variable from json. This causes the effect to draw as it should. If I set that integer to 0 (simulating json missing the variable), it draws incorrectly. I'll show my work below.

JSON defines as: {"en":16,"fn":"RainobwFill Rainbow","es":1,"mi":1,"ce":1,"sdd":6}
effects.cpp initializes the effect as : ADD_EFFECT(EFFECT_STRIP_RAINBOW_FILL, RainbowFillEffect, 6, 2);
It is missing a variable in the JSON blob. It is missing PTY_DELTAHUE which translates to dth.

The json blob should read: {"en":16,"fn":"RainobwFill Rainbow","es":1,"mi":1,"ce":1,"sdd":6, "dth":2}

rbergen, 2023-10-08:
In that case I think the serialization logic runs out of buffer space while putting the JSON object together. Could you maybe test with the code in the fix-serialization branch in this repo and let me know if that works?

PR #460 (2023-10-10): Bump smallest effect JSON buffer size to 192 bytes
## Description

This increases the smallest JSON buffer size used for effect serialization from 128 to 192 bytes. It seems that the introduction of the "mi" property in #454 makes that some effects that serialize additional properties (like RainbowFillEffect) now run out of buffer space, meaning properties get lost while serializing. 

As such, this PR fixes #457. 

Thanks to @revision29 for reporting this.

## Contributing requirements

* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDr [... cut]
closed 2023-10-10 by rbergen