PlummersSoftwareLLC/NightDriverStrip #464: Babel/traverse version used in web UI contains critical vulnerability

rbergen, 2023-10-20:
Dependabot reports that the version of babel/traverse used in the web UI contains the vulnerability documented in [CVE-2023-45133](https://nvd.nist.gov/vuln/detail/CVE-2023-45133), which is marked as critical by GitHub. 

Although we are technically not actually exposed to the vulnerability because we only compile trusted code, I think we need to upgrade to a babel/traverse version that does not contain this vulnerability, as soon as possible.

@KeiranHines Is this something you can look into?

KeiranHines, 2023-10-20:
@rbergen I'll take a look over the next few days I should be a simple update to the version numbers in npm.

rbergen, 2023-10-20:
@KeiranHines Thanks!

PR #465 (2023-10-21): Dependency update
## Description
This fixes #464 and this fixes #447 

## Contributing requirements
* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein is subjected to the license terms in [COPYING.txt](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/COPYING.txt).
commit d4108c32 (2023-10-21): Fixed #464: Udpated babel to version 7.23.x
closed 2023-10-21 by rbergen