PlummersSoftwareLLC/NightDriverStrip #468: Large Embedded Webserver Files are not Being Sent

kriztioan, 2023-10-24:
## Bug report

**Problem**

_What happens_: The 'larger' embedded web resource files `index.js`, `favicon.ico`, and `timezones.json` are not sent/received, but `index.html` is, as is the `JSON` from the 'simple' web endpoints like `effects`, etc.

_Expect to happen_: All files being sent/received.

**Steps**

1. Compile the `demo` target in VSCODE with webserver/wifi defines set to 1 in globals.h.
2. Upload to ESP32.
3. On an other device on network use `curl` to access web-resources.

**Example**

```curl -vvv 192.168.20.47/index.js | gunzip```

**Notes**

My device is an `esp32doit-devkit-v1`. The serial output does show 'GET for index.js', but subsequently no data is sent/received for the 'larger' files. I attempted some trouble shooting in the `ServeEmbeddedFile`-method of the `CWebServer`-class by using the `beginChunkedResponse`-method, but with no success. Similarly, I modified the code to read the files from an uploaded `SPIFFS`-image with the data, which had some success with the Chunked method, but the chunk-sizes were forced to only 256 bytes. I tried increasing `RESERVE_MEMORY` in `globals.h` with no success. When using a lambda for the `AwsResponseFiller` parameter in `beginResponse` that prints a message to the serial shows, again, the expected results for `index.html`, but not for the 'larger' files.

Result for `effects` end-point:

```
curl -vvv 192.168.20.47/effects      ✔  17:33:08 
*   Trying 192.168.20.47:80...
* Connected to 192.168.20.47 (192.168.20.47) port 80 (#0)
> GET /effects HTTP/1.1
> Host: 192.168.20.47
> User-Agent: curl/8.1.2
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 166
< Content-Type: application/json
< Server: NightDriverStrip
< Access-Control-Allow-Origin: *
< Connection: close
< Accept-Ranges: none
<
* Closing connection 0
{"currentEffect":0,"millisecondsRemaining":13931,"eternalInterval":false,"effectInterval":30000,"Effects":[{"name":"RainbowFill Rainbow","enabled":true,"core":true}]}%
```

Result for `index.html`:

```
curl -vvv 192.168.20.47/index.html | gunzip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 192.168.20.47:80...
* Connected to 192.168.20.47 (192.168.20.47) port 80 (#0)
> GET /index.html HTTP/1.1
> Host: 192.168.20.47
> User-Agent: curl/8.1.2
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 456
< Content-Type: text/html
< Content-Encoding: gzip
< Connection: close
< Accept-Ranges: none
<
{ [456 bytes data]
100   456  100   456    0     0   5514      0 --:--:-- --:--:-- --:--:--  6000
* Closing connection 0
<!doctype html><html lang="en"><head><title>NightDriverStrip</title><script crossorigin src="https://unpkg.com/react@18/umd/react.production.min.js"></script><script crossorigin src="https://unpkg.com/react-dom@18.2.0/umd/react-dom.production.min.js"></script><script crossorigin src="https://unpkg.com/@emotion/react@11.11.1/dist/emotion-react.umd.min.js"></script><script crossorigin src="https://unpkg.com/@emotion/styled@11.11.0/dist/emotion-styled.umd.min.js"></script><script crossorigin src="https://unpkg.com/@mui/material@5.14.9/umd/material-ui.development.js"></script><script crossorigin src="https://unpkg.com/html-react-parser@4.2.2/dist/html-react-parser.min.js"></script><script src="https://unpkg.com/prop-types/prop-types.min.js"></script><script crossorigin src="https://cdnjs.cloudflare.com/ajax/libs/recharts/2.3.2/Recharts.min.js" referrerpolicy="no-referrer"></script><link rel="shortcut icon" type="image/x-icon" href="favicon.ico"><link crossorigin rel="stylesheet" href="https://fonts.googleapis.com/icon?family=Material+Icons"><script defer="defer" src="index.js"></script></head><body><div id="root"></div><script src="index.js.gz"></script></body></html>%
```
[... 65 lines cut ...]
(W) 
(W) (ConnectToWiFi)(C1) Received IP: 192.168.20.47
(I) (ConnectToWiFi)(C1) Setting Clock...
[  6717][E][WiFiUdp.cpp:221] parsePacket(): could not receive data: 9
(W) (UpdateClockFromWeb)(C1) NTP clock: Raw values sec=1995184889, usec=464540
(I) (UpdateClockFromWeb)(C1) Old Time: 6.924527, New Time: 1698108210.464540, Delta: 1698108203.540013
(I) (UpdateClockFromWeb)(C1) Adjusting time by 1698108203.540013 to 1698108210.464540
(I) (UpdateClockFromWeb)(C1) NTP clock: response received, updated time to: 1698108210.464540, DELTA: 1698108203.540013
(I) 
(I) (ConnectToWiFi)(C1) Starting Web Server...
(I) (begin)(C1) Connecting Web Endpoints
(I) (begin)(C1) Embedded html file size: 456
(I) (begin)(C1) Embedded jsx file size: 9645
(I) (begin)(C1) Embedded ico file size: 6434
(I) (begin)(C1) Embedded timezones file size: 17834
(I) (begin)(C1) HTTP server started
(I) (ConnectToWiFi)(C1) Web Server begin called!
GET for: /index.html
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.20.47, Mem: 50312, LargestBlk: 38900, PSRAM Free: 0/0, LED FPS: 30 LED Bright:  22%, LED Watts: 11, CPU: 004%, 001%, FreeDraw: 0.019
GET for: /index.js
```

**NB** The current html-file asks for `index.js.gz`, should this not be `index.js`?

**NB2** My understanding is that the ESP webserver can only handle a single request at a time, however, modern browsers will do simultaneous request---can this cause an issue?

rbergen, 2023-10-24:
@kriztioan I think this issue's main point addresses something that isn't a thing, but your first NB does.

Bullet-point:

- favicon.ico is retrieved by actual browsers, to get the icon to put in the page's tab. You can refer to <https://en.wikipedia.org/wiki/Favicon> for more information about how this works.
- timezones.js is offered for retrieval in case a future iteration of the web UI wants to offer a drop down (or a similar control) to select the time zone for the respective device setting. The web UI doesn't use it at the moment, but as it is used within NightDriverStrip anyway, serving it via the webserver is basically free.
- index.js _is_ retrieved, but initially indeed using the wrong filename (index.js.gz instead of the index.js that the webserver uses to serve it, as per [line 157 in webserver.cpp](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/5a78b33ae34c59d36f085b71336acc77cc5e714a/src/webserver.cpp#L157)). @KeiranHines Maybe this is something you can take a look at, when you can find the time? I note that my browser does end up fetching the file using the correct URI after all, which it has to as the UI simply wouldn't work otherwise.

Concerning your second NB: I've been running the web UI on multiple types of devices for quite a while now, and never experienced this to be a problem.

KeiranHines, 2023-10-24:
I will try to get time on the weekend to look at it.

kriztioan, 2023-10-25:
As eluded to, I am able to get `index.js` somewhat reliably served using `beginChunkedResponse` and setting `max_bytes` as low as 32 bytes(!) using a modified version of ` ServeEmbeddedFile` (see below). However, it takes a while to render the website (Vivaldi on MacOS) and the bell-icon lists an increasing number of errors.

```C++
    void ServeEmbeddedFile(const char strUri[], EmbeddedWebFile &file) 
    {
        _server.on(strUri, HTTP_GET, [strUri, file](AsyncWebServerRequest *request) {
            Serial.printf("GET for: %s\n", strUri); 
            AsyncWebServerResponse *response = request->beginChunkedResponse(file.type,
            [file](uint8_t *buf, size_t max_bytes, size_t sent_bytes) -> size_t {
                    max_bytes = 32;
                    uint32_t send_bytes = file.length - sent_bytes;
                    if(send_bytes == 0) return 0;
                    if(send_bytes >= max_bytes) send_bytes = max_bytes;
                    memcpy_P(buf, file.contents + sent_bytes, send_bytes);
                    return send_bytes;
                });
            response->addHeader("Content-Encoding", file.encoding);
            request->send(response);
        });
    }
```

As for the **NB**'s, yes I had to update `index.js.gz` to `index.js` in `index.html`-perhaps browser caching is preventing you from having any issue? In addition, using `wireshark` to follow the network traffic shows as lot of `TCP_ACK_RETRY`s, leading me to postulate that the ESP32 is struggling with handling the multitude of connections. Another thing I'm wondering is whether it could be my specific ESP32 version: [DEVKIT V1 VROOM 32](https://www.amazon.com/gp/product/B084KWNMM4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)?

@rbergen Bedankt voor alle hulp!

rbergen, 2023-10-25:
For me the problem here is that I just can't relate any of the issues that @kriztioan describes to what I'm seeing, except for NB1. I've used Chrome (and Edge, but that's Chrome at its core, of course) on a number of computers (all Windows 11) to access the web UI on a collection of different (types of) ESP32 devices, and the current webserver code in this repo's main branch just works.

Caching is not a factor either. I often test the web UI with the Chrome dev tools running, and cache is disabled every time I do. I can literally see the browser fetch index.js after the GET for index.js.gz fails.

This being the case, I can't do anything with or about this issue other than addressing the NB1 thing, which @KeiranHines already indicated he'll look into.

> Bedankt voor alle hulp!

Graag gedaan! :)

kriztioan, 2023-10-25:
@robertlipe `index.html` _does_ successfully load, it is the 'larger' `index.js` that is not being served. It is difficult to say whether my network configuration is 'special'; I do have `dd-wrt` flashed to my router, but am not doing anything 'weird' with it and I haven't had issues with it before. I tried the Web Installer with Edge under Windows 11 with selecting the `ESP32` device and `Demo` project. The project is successfully flashed, but no option to 'CONNECT TO WI-FI' button appears, even after 'LOGS & CONSOLE' shows output. Doing a bisection on the codebase seems a bit excessive at this point as using a chunked response works adequately for me.

@rbergen Since my issue appears not to be reproducible and my modified `ServeEmbeddedFile`-method works sufficiently for my needs, I'm closing the issue. Thank you for your consideration and engagement.

closed 2023-10-25 by kriztioan

rbergen, 2023-11-11:
@kriztioan As described in a [comment](https://github.com/PlummersSoftwareLLC/NightDriverStrip/discussions/498#discussioncomment-7540230) on discussion #498, I have now also run into this. It seems to happen when there is just enough RAM available to start the web server, but not enough to serve the bigger files. I have prepared an update to the build configuration in question (Spectrum for the M5StickC Plus) to fix it there; your low-byte-count chunked response may be another way to (just) dodge the issue on whatever device/configuration combination you are/were building for.

kriztioan, 2023-11-11:
@rbergen I was able to track the issue down to a lack of RAM as well. Following the entire stack down through `AnyTCP` and `lwip` shows an `err_t` of `ERR_MEM` being return by a call to`_tcp_write_api`, which is not being explicitly handled by the `ESP Async WebServer` and no such errors are propagated further up the calling chain.

Some research into the different flavors of ESP32s revealed my [ESP32-WROOM-32](https://www.amazon.com/gp/product/B084KWNMM4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) is not equipped with `PSRAM`. More consciousnessly reading through the different documentation/issues/discussions lead me to a statement by @davepl that having `PSRAM` is a must. To that end, I went and acquired myself three [ESP32-S3-DevKitC-1-N16R8](https://www.amazon.com/gp/product/B0CGYXJB6Y/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)s. After adding its [board configuration](https://github.com/platformio/platform-espressif32/blob/deb6aa342245ad7bcb3d9ed69050f203779833ab/boards/esp32-s3-devkitc-1-n16r8v.json) file to `PlatformIO`, using `default_16MB.csv` for the partition table, and configuring a specific `Device section` for it it in `plattformio.ini` has everything purring like a kitten.

Admittedly, my chunked approach did often require a number of page reloads for it to work and should likely not be considered for 'production'.  My hypothesis there is that while the chunks themselves require a smaller memory footprint the number of pending requests balloons, reducing the available RAM there.  One possible strategy is to require certain 'specs' to have certain features be enabled.