PlummersSoftwareLLC/NightDriverStrip #472: Web UI doesn't save empty string values for settings

rbergen, 2023-10-25:
## Bug report

**Problem**

The web UI doesn't save empty string values for string settings. More or less coincidentally this made sense for all available settings until now, but with the upcoming merge of #469 a string setting (hostname) will be introduced that can be empty.

**Steps**

1. Load the web UI on a device with firmware that includes the hostname device setting. 
2. Open the device settings dialog.
3. Enter and apply a hostname if one is not already set. Then reopen the device settings dialog.
4. Remove the hostname and apply again.
5. Reopen the device settings dialog, and observe that the hostname that was configured before step 3 is still set.

**Notes**

To support the web UI knowing what settings are optional (i.e. may be empty), an optional "optional" property has been added to the output of the /settings/specs endpoint. That will also be merged with #469.

KeiranHines, 2023-10-27:
@rbergen I have implemented a fix [here](https://github.com/KeiranHines/NightDriverStrip/tree/optionalSettings) however there is some changes I would like to discuss before considering this feature complete. 
1. The use of the word 'optional' I feel like that may be error prone in the future as my initial thought would be that an 'optional' setting is not required to be sent. This implementation is the opposite, the property is more of a "allows null/empty" values. 
2. Should we consider adding a 'defaultValue' property. This could address both the ambiguity of 'optional' and allow for greater flexibility in the future. With the way my change currently is. If the user input is empty and the setting is 'optional' it send and empty string to the endpoint. However, if the input was say a numeric input and optional was true, I would still send an empty string. If we had a 'defaultValue' property I could sent that instead.

rbergen, 2023-10-28:
@KeiranHines Thanks! In response to your points:

1. I think you're right about "optional" being an unfortunately chosen name for the property I added. I chose "optional" as an antonym for "mandatory" (as in, _doesn't have to have a value_ as an opposite for _must have a value_), but that is actually only ever true for strings as you say. Numeric settings/properties always have to have "a" value, which may even have to respect minimum and/or maximum value boundaries.  
  Maybe we should rename "optional" to "emptyAllowed", and only look at that property if the property type is String.
2. In the context of the previous point, I think adding a default value for numeric settings/properties to be able to deal with empty strings being sent for numeric values is wagging the dog.

I'm happy to make the name change from "optional" to "emptyAllowed" at the back-end side in the bugFixes branch you used to open #475, if you're ok with me doing so. You can then update the front-end to look for that property (and then only for Strings) accordingly.

Let me know if this seems a sensible approach to you.

rbergen, 2023-10-28:
@KeiranHines I'll also open another issue to request the implementation of minimumValue and maximumValue boundary checking at the front-end, for settings that come with them.

KeiranHines, 2023-10-28:
Sounds good. Feel free to update that branch or #469 as I am currently branched off that for this fix as it relates to that feature.

rbergen, 2023-10-28:
Good point. I'll update #469 and report back here when done.

rbergen, 2023-10-28:
#469 has been updated, and CI completed on it. Which means you can "safely" pull in my latest commits to my polish-webprojects branch into yours.

Also, I opened #476 for the boundary checking thing.

I realize this is a bit of a sidestep to this discussion, but I was wondering if you were considering picking up other Web UI issues as well? Particularly #300 and #301 would be nice to have in, IMHO.

KeiranHines, 2023-10-28:
#300 is something I considered but I haven't put much thought into the technical aspects. #301 I have not really considered. 

I will add the min max as part of this branch. My next priority after that was to go back and look at the global effect timer and fix those issues. Once I am done with that I may look at 300 as that code would be reusable for 301.

rbergen, 2023-10-28:
@KeiranHines All of that sounds great! I genuinely don't intend to apply any pressure here, but I think the whole web UI will reach a new level of maturity if/when all of that is done - particularly if #301 is in fact part of "all of that".

Again, not seeking to apply any pressure, just voicing enthusiasm about potential future progress on this part of the codebase. :)

PR #478 (2023-11-06): Config dialog improvements
## Description
Fixed #472 Added support for `emptyAllowed` configurations to be saved.
Fixed #476 Added support for configuration to limit between min and max values for numeric input types. If no min/max is set the dataTypes min/max is used as a proxy.  

## Contributing requirements
* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein  [... cut]
commit 2010f205 (2023-11-06): Updated Fix #472: changed optional variable to emptyAllowed
commit 84af44c6 (2023-11-06): Fixed #472 Added support for optional config settings
closed 2023-11-06 by rbergen