PlummersSoftwareLLC/NightDriverStrip #473: Programmatically changing the effect interval causes an infinite recursive json buffer increase

revision29, 2023-10-26:
## Bug report

**Problem**

When I set the effect interval from a key on my remote control, the interval is set. After it writes the value to json storage it then starts a loop where the json buffer size continues to increase until the board is power cycled. This does not happen when other events trigger json to store

What I see in the console:

(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 6144 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 8192 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 10240 bytes

**Steps**

1. Have the debugger console running in VS Code.
2. Somewhere in code trigger: effectManager.SetInterval(30000);   
3.  Watch the json buffer size increase by 2048 bytes (as defined in jsonbase.h).
4. Watch the effects being drawn on the leds lag more and more.
5. Power cycle the board and the effect interval now reflects the value triggered by the remote. The buffer increase has stopped.
6. Press the button that changes the interval. Watch the console fill up with SerializeWithBufferSize notices.

**Example**

effectManager.SetInterval(30000);   

**Notes**

Until today, I was running on an outdated codebase (last changes from main branch merged into the branch I'm working on are from October 9). I merged the changes to the main branch from the pull requests dating from October 17th onward. I don't think the error showed up until I did that.

I suspect there are other instances where this error is popping up, but I am having a hard time isolating when / where. I just pushed a fresh build after erasing a device and it is giving the out of memory notice as well. But that appears to be related to a couple of audio effects I added to effects.cpp. If I disable those and disable audio, the out of memory warning goes away.

I will verify on a different board later and see if the error is present there as well.

rbergen, 2023-10-26:
This happens when a `SerializeToJson()` function called by `SerializeWithBufferSize()` returns false - which it should only do if it actually does run out of memory while serializing. Can you provide a link to the branch you're working on, and tell me what PlatformIO env you are building?

revision29, 2023-10-26:
Update: only is present when audio is enabled. Maybe there is something missing from my configuration that is causing this to break. Will investigate further.

rbergen, 2023-10-26:
I just tested this with the current version of the codebase on Mesmerizer and Spectrum (the latter running on an M5StickC Plus) by changing the effect interval using the web UI (which calls the same `g_ptrSystem->EffectManager().SetInterval()` function in the end) and the problem does not occur on either.

What I find interesting in your logging is the double `NotifyJSONWriterThread` trigger - only one of those should really be originating from the `SetInterval()` invocation.

revision29, 2023-10-26:
@rbergen https://github.com/revision29/NightDriverStrip/tree/remoterework 
The environment is espcross. it's based off of the basic demo environment. 

I had to put the code back in a broken state. Having you look at the branch is like inviting a stranger over to a messy house. So, don't just judge the mess. 😃 

Anyway, I don't know if the problem is caused by me not having everything properly setup for audio compatibility, if there is something amiss with the web server, or maybe a combo of both.

revision29, 2023-10-26:
@rbergen 
I clicked the wrong button when responding and accidentally closed the issue. So I reopened it.

NotifyJSONWriterThread is being triggered twice. On the remote I am using the AUTO button to either set the interval to 30 or 0. If it is setting to 30, I go ahead and trigger the next effect to give visual feedback that the button press worked. So, it writes the setting for current effect and then writes a second time for the change in interval.

Now that I know what methods are involved, I can dig a little deeper.

I changed 4 things and then discovered the problem:
Enabled wifi and web server on the device.
Enabled audio.
Added two audio based effects to effects.cpp.
Merged the latest changes from the main branch.

I will see which combo of those changes result in the error and work from there. 

I do know that disabling the web server, disabling audio, and removing the 2 audio effects makes the error go away. The problem with debugging is that this device is mounted on the wall in my office a few miles from home. I have the same type of board at home, but it is brand new with nothing soldered to it. I have another board of a different brand that I use for testing my code at home. That just introduces more potential differences / points of failure. I will do more testing and report back.

rbergen, 2023-10-26:
I think the problem may be in the `SerializeToJson()` function in the `PaletteFlameEffect` class that starts at line 258 on include/effects/strip/fireeffect.h. Could you change the value 512 in line 260 of that file (`AllocatedJsonDocument jsonDoc(512);`) to 1024 and see what happens then?

rbergen, 2023-10-26:
Actually, I now think that whole `SerializeToJSON()` function in the `PaletteFlameEffect` class is borked, and always has been. It becomes visible now because we switched to a newer version of ArduinoJson in #466, which is better in flagging buffer overflow conditions - which in turn relates to the comment @robertlipe just made. It seems I missed some effect serialization functions while implementing the changes he mentions, included in said PR.

@revision29 Could you replace lines 258 up to and including 267 in your /include/effects/strip/fireffect.h file with the following, and see what happens then with audio and the audio effects enabled?

```cpp
    bool SerializeToJSON(JsonObject& jsonObject) override
    {
        AllocatedJsonDocument jsonDoc(1024);

        JsonObject root = jsonDoc.to<JsonObject>();
        FireEffect::SerializeToJSON(root);

        jsonDoc[PTY_PALETTE] = _palette;

        assert(!jsonDoc.overflowed());

        return jsonObject.set(jsonDoc.as<JsonObjectConst>());
    }
```

Note that this will not be the final, final version of this function. I will open a PR with an update to all `SerializeToJSON()` functions I missed before that will include the actual update to this specific function, but I'd first like to know if the problem you are seeing is indeed coming from it.

revision29, 2023-10-27:
@rbergen I will try that in about 12 hours when I am back in my office. 

Here's another piece of the puzzle. I got home and started to figure this out while using a different board. At home my primary testing board is a heltec wifi kit v3. When I set ENABLE_AUDIO to 1 it fails to build no matter what I try. Documentation says audio is limited to the m5sticks, but the code says it should work. The problem is the library / driver for processing audio is not being included at compile time and the code complains that the required functions are absent.

When I set ENABLE_AUDIO to 1 on my cheaper esp32 board, the same one defined in the basic demo config, it compiles just fine. But, I don't think it is including the library / driver for processing the audio either. I think VS Code is being inconsistent with its error checking and it should fail to build. What I think might be happening is that I am triggering effects that utilize audio, but the audio processing system is no where to be found. So the effects are returning errors, null, or undefined and is causing the json buffer problems when the effect's serialize to json method is being called.

The compile errors I am getting when compiling for the heltec board include: "'i2s_adc_enable' was not declared in this scope," "'i2s_adc_disable' was not declared in this scope," and "'I2S_MODE_ADC_BUILT_IN' was not declared in this scope"

I don't know if these issues are related or if this enable_audio failing to compile with some configurations is a distinct issue. Thought I would add some data points in case it's relevant.

rbergen, 2023-10-27:
@revision29 I've actually opened #474 with the (proposed) final solution. If nothing else, it should unequivocally tell us which serialization process is driving things off the rails.

rbergen, 2023-10-27:
>  If I turn on audio in DEMO for the heltecv3 builds, bad things happen.

This is precisely why I stopped debugging anything that's specific to devices I don't own. There are too many variables to be able to do the debugging at a mental level only. In this case, when it concerns devices that do not have microphones built in, then the assumption should be that no effort has been made by past or current contributors to make them work with audio.

And that's where my involvement with this particular exploration/discussion ends.

revision29, 2023-10-27:
@robertlipe I disabled audio and it breaks just the same as without it. That eliminates one possible point of failure.
You mention excessive copy and paste. Yes. I have 2 devices in real world deployments. I have two different boards that I use to test various random things as I learn and experiment. That is on top of the already large number of devices defined in globals. 

Regarding audio support. That is now a rabbit hole I will not go down. I thought it would be as "simple" as getting some non-serial mics from Amazon and wiring them to particular input pin. The mic would output a standard analog waveform. It looked like the audio code in NightDriver handles serial or analog input. But it looks like I am wrong. It will be much easier to source m5sticks, even though they are not available on amazon at the moment. My wild guess about audio support causing problems turns out to be, wrong.

@rbergen the code you pasted in a comment above did not fix the problem. I disabled the web server and the problem goes away. The project I have running the code does not need a webUI, so it's no big loss. At least now I can do a little more digging to see if the problem is memory or something else. What's crazy is I have a button that just triggers NextEffect and it does not trigger the problem. Only SetInterval does.

rbergen, 2023-10-27:
@revision29 The fact that `NextEffect` doesn't trigger the problem and `SetInterval` does makes sense. The latter triggers a JSON serialization of the effect list in `EffectManager` (and as such, all effects in it), the former only updates the "current active effect" index file.

On the other hand, disabling the webserver should not make a difference whatsoever. The IR remote code doesn't/shouldn't interact with it in any way, so why disabling that circumvents anything I can't fathom.

You could sync with the main branch again after #474 is merged, and then see what logging that generates when you retry.

rbergen, 2023-10-27:
@revision29 In the meantime, you could add some additional logging to see which effect fails to serialize. In effectmanager.h, within the function `SerializeToJSON()`, you could replace the following lines (line numbers 288 and 289 in the main branch):

```cpp
            if (!(effect->SerializeToJSON(effectObject)))
                return false;
```

with these:

```cpp
            if (!(effect->SerializeToJSON(effectObject)))
            {
                debugW("JSON serialization of effect \"%s\" with effect number %d failed!", effect->FriendlyName().c_str(), effect->EffectNumber());
                return false;
            }
```

and then rebuild, reflash and rerun. That should log name and effect number of the effect that causes the buffer increase loop - and indeed if it is the same effect every time.

revision29, 2023-10-27:
I will be glad to merge and test.

It’s weird behavior for sure. If it were a memory issue the serializer should  also break with NextEffect. 

I just started throwing debug code all over the serializer. Set interval sets the variable and the notifies the json thread to write the config. As it builds the json document it successfully creates the interval object. The it iterates through the effects to build those objects. It fails to serialize the first object. That effect is EFFECT_STRIP_RAINBOW_FILL. I disabled that one. Now the first effect I  My listn is EFFECT_STRIP_PALLETTE. That one fails too So disabled all of the palette effects. The next one is EFFECT_STRIP_COLOR_FILL but does not break. I only had two effects, the other being strip fire. I could not get it to break. Two effects?Maybe I was running too many. So, I added 6 or 7 more color fill effects. Did not break. Then I set strip fire as the first effect. It did not break. 

I then added back a strip palette effect. It broke again. Then I disabled that and added back the rainbow effect. It did not break. So, the problem has been narrowed to EFFECT_STRIP_PALLETTE. 
I suspect the problem is when the serializer is returning the result of jsonObject.set(…). It is successful on initial effect build. But when I am triggering SetInterval jsonObject.set is failing for  EFFECT_STRIP_PALLETTE. Then the problem cascades and it starts to fail on the fire effect. Then it loops back through the effects and tries it all again, fails and the loop continues.

revision29, 2023-10-27:
@rbergen I was off testing when you typed your debug code suggestion. I went a different route but got the same result. The effect that fails consistently is EFFECT_STRIP_PALLETTE (number 21). Sometimes I can get the fire effect (number 8) to fail.

I have been having trouble connecting to my device's web server. I don't know if that is a problem with the board running out of memory or if there is something with the network preventing me from accessing it. But, that is all the time for debugging I have for today.

prschguy1, 2023-10-28:
I have been working with using audio for led strip builds for some time now. Recently, I went back in time to a build before the code base changed. I put Atom light on a M5stick plus, and copying in strip effects from other projects, many of the effects were audio reactive, many more than what the current code base does. In further experimentation I tried some new things I have never tried before on the current code base. In trying to use M5 stack for led strip projects I noticed that if: -DSHOW_VU_METER=1 is added to build env it will output a vu_meter  on 1 spoke. Not sure how it works, but this works on Heltec v1, M5 stack, and M5 stick. While many of the audio reactive effects draw blanks, this behavior is pretty consistent. That effect is sound reactive, and seems pretty accurate. That code is in drawing .ccp. Additionally, and this really blew my mind, I added #define NUM_INFO_PAGES 2 to the globals for Atom light. Low and behold it displays spectrum analyzer and bar graph to the oled display. That behavior appears to be in sync with the VU_meter effect. Unfortunately, it does not display current effect which is something I really wanted. While I was seeing VU values on serial out, wasn't sure if it was real or not. Apparently, it is, and audio is there, it is being processed, but for whatever reason it just isn't being pushed out onto the strips. Since I have a Heltec V1 on breadboard with mic, remote, I'll take that back into the old code base, and maybe TTGO, and see what works there.

rbergen, 2023-10-28:
@revision29 I'm reading back in this thread now, and I now notice this in the original bug report:

5. Power cycle the board **and the effect interval now reflects the value triggered by the remote**. The buffer increase has stopped.

I don't know how I missed this before, but that means JSON serialization of `EffectManager` _with all effects_ **must** have completed successfully _before_ the power cycle, otherwise the new interval won't be persisted - it's integrated in the `EffectManager` JSON blob that also includes the whole effect list. That just won't happen if the serialization process runs out of buffer memory.  
That makes me wonder if the buffer increase loop is indeed infinite.  

Can you provide a longer excerpt of your console logging that shows the JSON buffer size continues to climb (way) beyond the 10K your original bug report shows?

revision29, 2023-10-30:
@rbergen I am deep into shoving debug statements all over the serializer codebase to find the issue. I suspect I am running into some hardware constraint of the board. If I add enough debug statements in some parts of the code it adds enough delay in the processing that it will not fail. It's weird. But, I have enough information to track down when and where it is failing. My goal is to see if I can provide a fix or optimization.

Some brief observations. When the serialization succeeds, the size of the whole json object is using about 2134 bytes. The failure happens when the json for a specific effect is set into the effects json object. Each effect has a limit of 192 bytes. The buffer size for the json object is not being overrun. When it fails, the loop of buffer increase is infinite. Out of humor, I was going to paste in about 10 seconds of buffer increase statements, but the VS Code monitor buffer could not hold all of the data. At some point drawing the LED pixels gets very, very laggy.

I'll keep poking around and eventually propose a fix. If I need more help, I will reach out. I now have a very good understanding of the whole json serializer system. 

This is about 5 seconds of buffer increases:
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 6144 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 8192 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 10240 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 12288 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 14336 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 16384 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 18432 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 20480 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 22528 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 24576 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 26624 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 28672 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 30720 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 32768 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 34816 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 36864 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 38912 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 40960 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 43008 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 45056 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 47104 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 49152 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 51200 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 53248 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 55296 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 57344 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 59392 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 61440 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 63488 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 65536 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 67584 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 69632 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 71680 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 73728 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 75776 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 77824 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 79872 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 81920 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 83968 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 86016 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 88064 bytes
[... cut ...]
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 301056 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 303104 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 305152 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 307200 bytes
(W) (SerializeWithBufferSize)(C0) Out of memory serializing object - increasing buffer to 309248 bytes

rbergen, 2023-10-30:
@revision29 Yep, that looks like an infinite loop to me.

It's a weird error state indeed. ArduinoJson's `JsonObject::set()` should really only return `false` if the containing `JsonDocument`'s buffer is too small - or that's how it worked till very recently at least.

revision29, 2023-11-11:
I have submitted a fix for the infinite loop: [pull request 499](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/499). It was a fun adventure figuring out the complexity of the serialization process. The root cause was not in the effects manager or the serializers for each individual effect. It was a memory allocation problem. The code did not handle that failure point. If there was not enough memory for the JSON document, it would fail to create the variable in memory. Any attempts to add data to it, such as effects definitions, it could not. The code assumed the failure was too small of a buffer so it would increase it. On the next loop it would fail again. If there was not enough memory the first time, increasing the buffer by 1024 (or whatever it is) is NOT THE SOLUTION. "You don't have 4kb available? Okay, give me 6kb." Doomed to fail every time. Despite the dozens of hours spent looking for the problem, I learned a lot.

closed 2023-11-15 by rbergen

revision29, 2023-11-18:
@robertlipe I don't remember what I have said in which context, so I will summarize here. It turns out that no effect was causing any issue. The root cause was a failure to allocate memory for the jsondoc that holds all of the serialization data that will be written to the effects file. That failure happens before any effect is serialized. It then passes the failure to each effect as it serializes. The last PR I submitted mitigated the failure. So, the infinite loop has been patched. But, it still does not deal with the root cause of a lack of memory. This is a problem I want to address later. I have another set of changes I am working on that is taking my focus. After that, I will need to get debugging working correctly in VS Code to see if I can determine what is using memory. That part is a little over my head for now.