PlummersSoftwareLLC/NightDriverStrip #479: Web UI: Global Brightness and Color1/Color2 selection

davepl, 2023-10-29:
I believe we need a global brightness slider and a way to set the global color/palette.  So something like a slider and two color panels with color pickers attached?

I'd imagine it to be either at the top of the main page/effects list, perhaps even in a non-scrolling region below which the list scrolls.

You might imagine it differently, and that's fine too, as long as you imagine it better :-)

rbergen, 2023-10-29:
Hahaha, that last comment is rather cheeky, Mr. Plummer. :D

We already have a device setting for the global brightness that can be set via the web UI, but the slider is currently not yet in place. I'll think of a way to "suggest" to the UI to use a slider instead of a numeric input field.

I think the color picker control is something that still needs to be developed as well. Once that's in, we can easily apply that to the global color/palette once it has been "promoted" to a device setting itself.

Specifically picking the brightness and color/palette as settings that are lifted out of the device settings dialog (and then exactly where they are put on the main screen) is a matter of personal preference. I don't have a strong opinion about it, so I'll go with whatever those who do think is best.

rbergen, 2023-11-02:
So I played with some ideas and then decided to apply the KISS principle: I think it's best to just add a "Slider" data type (next to "String", "Integer", "PositiveBigInteger", "Boolean", etc.). It will translate to an `int32_t` at the C++ level, and it will be mandatory to provide both a minimum and a maximum value when the corresponding `SettingSpec` is created.

This does mean that I can't apply this change to the existing global brightness setting until the slider control has actually been implemented in the frontend.

KeiranHines, 2023-11-05:
@rbergen is color1/color2 exposed via the API similar to the global settings or effectSettings? If its exposed so I have something to work with, I am happy to build out a colorPicker or similar Config.

rbergen, 2023-11-05:
@KeiranHines Not yet. For one, I didn't want to expose colours or palettes until there was an editor to pick/compose them - but the JSON data type encodings for both can be found in /effectsConfig.

Concerning color1/color2 I'm not sure I fully understand those properties themselves. I do know we have a global color (which may well be color1) and a global palette, but a palette is usually a set of _16_ colours, not 2. 

@davepl Could you maybe help out by clarifying why we're talking about 2 colors in the context of a discussion about 16 color palettes?

KeiranHines, 2023-11-05:
Fair enough with respect to exposing the pallette. I will take a look at the backend and have a go at getting the UI working at some point soon.

rbergen, 2023-11-05:
Colours are basically encoded by sticking the 0-255 value for the R, G and B in a 24-bit integer, where R is in the most significant byte, and B in the least significant byte. The exact line in the back-end code taking care of this is here: https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/9ac393c4bf04411bab9024e48362923bed2c9b9b/include/jsonserializer.h#L79

A palette is a JSON array of 16 of such values. Its encoding logic is a few lines down in the same source file.

closed 2023-11-15 by rbergen