PlummersSoftwareLLC/NightDriverStrip #482: Web UI: allow user to reset the device config, reset the effect list and/or reboot the device

rbergen, 2023-10-30:
The on-board API's `/reset` endpoint allows three types of "reset" to be triggered:

- A reboot of the device (`board=true`)
- A reset-to-defaults of the device configuration (`deviceConfig=true`), which should always be combined with a device reboot
- A reset-to-default of the effect list (`effectsConfig=true`), which should always be combined with a device reboot

So far, the UI does not offer a facility to trigger the latter two. A board reset can be triggered by opening the global settings dialog and clicking "Apply and reboot device" without changing any settings, but I think that's not very intuitive.

I think the web UI should allow a user to trigger each of the three resets, as an independent action. Because of the impact of the latter two, a confirmation flow should probably be triggered before actually applying the resets in question. When it comes to location, the global settings dialog does seem to be a logical place to me. Exactly how to embed it into that dialog's design is something I'm not exactly sure of, which also means I don't have a strong opinion about it, up-front.

KeiranHines, 2023-11-05:
One other idea would be to add an icon to the sidebar for, in that menu/dialog have reset controls. 
[image]

rbergen, 2023-11-05:
You know what, that's also fine by me! :)

PR #518 (2023-11-18): Feature/device control
## Description
Fixes #482 added a sidebar button for reboot, reset device and reset effect options. 
[image]

## Contributing requirements

* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein is subjected to the license terms in [COPYING.txt](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/COPYING.txt).
closed 2023-11-18 by rbergen