PlummersSoftwareLLC/NightDriverStrip #486: Smoke Broken!

davepl, 2023-11-02:
Looks like a recent change has broken the noise effect of Smoke.  It's now all sparkly on the leading edges, and looks broken compared to what it used to, where it was a smooth smoke edge.

rbergen, 2023-11-02:
@davepl It's either what you say or PR #450. If it's the latter that means the change was deliberate based on the opinion of @robertlipe and myself that this is actually a better look for the effect than the previous one. 

@robertlipe Could you maybe check if the look of SMSmoke as the current code produces it, is indeed what I merged on September 30th?

rbergen, 2023-11-07:
@robertlipe I've had some out-of-band communication with Dave about this, and he has a strong preference to revert to a smooth operation of SMSmoke, so without the sparkles. If we can do that while keeping the rolling black bar out of the equation, then I think we're still in a better place than we were before #450.

Is this something you can look into?

PR #525 (2023-11-18): Remove tear/pepper in fire from SMSmoke
## Description

This updates the SMSmoke effect to remove the tear/"pepper in fire" that was added in #450. As such, fixes #486.

## Contributing requirements

* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein is subjected to the license terms in [COPYING.txt](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/ma [... cut]
closed 2023-11-18 by rbergen