PlummersSoftwareLLC/NightDriverStrip #504: Web UI: improve rendering of setting editors

rbergen, 2023-11-13:
While working on #503 I noticed a couple of things that I think we need to improve about the way setting editors are rendered in the global and effect setting dialogs. Specifically:

1. The name/label of a setting now seems to be rendered using the exact same font, including size and color, as the setting description. Particularly with combinations of setting editors that are not textual, this can make it difficult to distinguish one setting from the next. I think the setting names/labels should be made to stand out a little more. Whether this should be done using a bigger font, brighter color, text decorations or something else altogether, I'll leave to someone with actual UI design skills.
2. For Boolean/checkbox settings, the setting description is not shown and as such completely inaccessible. I think we should somehow make the description for these settings visible or accessible also. As with the other point, I'm not sure how to best do this.

KeiranHines, 2023-11-14:
Thanks for the feedback. Checkbox settings is easy enough
[image]
As for names/labels. Hows about this? 
[image]
[image]
I have changed the names/labels to be more consistent and slightly bigger. I have also re-positioned description for the checkbox although I am less sure about that idea.

KeiranHines, 2023-11-14:
Alternatively with a less padding around the inputs
[image]
And with a background to the inputs
[image]

rbergen, 2023-11-14:
@KeiranHines Thanks for the designs!

In my opinion:

- Adding checkbox setting descriptions underneath the checkbox control + label works best (i.e., as shown in your first screen snippet). 
- I think the change in label format helps, but maybe we should also add a little more whitespace between the individual settings (as in "above" the labels). 
- I don't think adding a background colour to the input fields adds much. It's actually mainly with the non-input field controls (sliders, colour pickers, etc.) that things get a bit "cramped" at the moment.

PR #517 (2023-11-16): Bug fix/config dialog
## Description
Fixes #504. Added descriptions for checkboxes, added padding before input names/titles. Increased font size of titles and made titles consistently ontop of the input.  
[image]
[image]

## Contributing requirements
* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein is subjected to the license terms in [COPYING.txt](http://git [... cut]
commit e58980db (2023-11-16): Fixed #504 added descriptions to checkboxes, restyled some inputs and added some padding before input title
closed 2023-11-16 by rbergen