PlummersSoftwareLLC/NightDriverStrip #534: Smoke effect  not working correctly on 64x64 Hub75 matrix.

mikejohnau, 2023-11-21:
I have a 64x64 Hub75 matrix hardwired to an ESP32-Wrover dev board. Most effects seem to work fine, Audiograph, Metaballs, Fireplace, SuperNova and Pnog Clock all work. The some effect appears to run but the frame is fractured and large diagonal lines appear accross the frame.

Sample video is at https://user-images.githubusercontent.com/26905608/284036081-3890110a-ef98-4420-9a42-e43d738cc517.mov

## Bug report
64x64 Hub75 matrix not displaying Smoke effect correctly

**Problem**
64x64 Hub75 matrix not displaying Smoke effect correctly. I have tried different Matrix_Tye settings none have had any effect.

**Steps**

1.  Connect 64x64 Hub matrix
2. Load Smoke effect.
3.  
4.

mikejohnau, 2023-11-22:
Going back to the previous build
[9e1bf8f](https://github.com/PlummersSoftwareLLC/NightDriverStrip/commit/9e1bf8fa428a7945858fe48f60d3aeef0cc5357d)

Appears better. The panel now seems divided into quarters that float up and down and generally move around the screen. The wipers are also less slightly visible. I did have to remove the g()->Clear() line after the EVERY_N_MILLISECONDS line. With that in place I could only see the wipers.

latest video..

https://github.com/PlummersSoftwareLLC/NightDriverStrip/assets/26905608/a13e237e-4a7f-4b1b-afb0-f8ea2c5bf065

Just looking at the code and your comments above. Are we actually touching the smartmatrix include with this effect? It looks to my untrained eyes that this is all GFXbase.

Last part of code for reference.

    EVERY_N_MILLISECONDS(100)
    {
     // g()->Clear();
      g()->SetNoise(1000, 1000, 1000, 4000, 4000);
      // (orig) "speed of movement through the noise array"
      // Calling SetNoise() in here will index past what was
      // FillGetNoised, which returns slowly scrolling bars
      // of black along X and Y axes.
      g()->FillGetNoise();
    }

    // Lower number for thicker, more static fog. Higher for more wisp.
    g()->MoveFractionalNoiseX(3);
    g()->MoveFractionalNoiseY(3);
   
    // Without this, we get tornadoes where the diagonals cross as there's
    // an excess of set pixels there.
    g()->BlurFrame(20);
   
  }

davepl, 2026-01-03:
Cool!  I was just watching it on my 48x16 LED screen and wishing it fit better! :-)