PlummersSoftwareLLC/NightDriverStrip #547: The mesmerizer reboots when clicking some effects

heidepiek, 2023-11-27:
Just downloaded the code again. Saw that there were new effects. Compiling went well, upload to esp too. But when I click the following effects in the web ui then the mesmerizer restarts: 
colorball
Atomic
Three rings
Pacman

Also when I scroll through the effects with the ir it crashes when I get to these effects, and restarts! It then goes back to an effect that works.

heidepiek, 2023-11-27:
This is in my serial:

- Terminal on COM9 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (loop)(C1) Sending Improv packets to indicate WiFi is connected. Ignore any IMPROV lines that follow this one.
IMPROV�
IMPROVhttp://192.168.1.248
(W) (ColorDataTaskEntry)(C1) Started color data server!
(W) (ConnectToWiFi)(C1) Connected to AP with BSSID: "A0:2D:13:50:FA:94", received IP: 192.168.1.248
(I) (ConnectToWiFi)(C1) Starting/restarting Socket Server...
(W) (ProcessIncomingConnectionsLoop)(C1) Error accepting data!
(W) (SocketServerTaskEntrocket server started.
sed.  Retrying...
(I)
I)
tToWiFi)(C1) Socket server started.
(I) (ConnectToWiFi)(C1) Publishing OTA...
(I) (ConnectToWiFi)(C1) Setting Clock...
[  4963][E][WiFiUdp.cpp:221] parsePacket(): could not receive data: 9
(W) (UpdateClockFromWeb)(C1) NTP clock: Raw values sec=1174682160, usec=273502
(I) (UpdateClockFromWeb)(C1) Old Time: 5.068206, New Time: 1701117312.273502, Delta: 1701117307.205296
(I) (UpdateClockFromWeb)(C1) Adjusting time by 1701117307.205296 to 1701117312.273502
(I) (UpdateClockFromWeb)(C1) NTP clock: response received, updated time to: 1701117312.273502, DELTA: 1701117307.205296
(I)
(I) (ConnectToWiFi)(C1) Starting Web Server...
(I) (begin)(C1) Embedded html file size: 502
(I) (begin)(C1) Embedded jsx file size: 15259
(I) (begin)(C1) Embedded ico file size: 6434
(I) (begin)(C1) Embedded timezones file size: 17834
(I) (begin)(C1) Connecting Web Endpoints
(I) (begin)(C1) HTTP server started
(I) (ConnectToWiFi)(C1) Web Server begin called!
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39628, LargestBlk: 31732, PSRAM Free: 385919/4173083, LED FPS: 20 Refresh: 60 Hz, Power: 1942 mW, Brite: 100%, Audio FPS: 17, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 081%, 092%, FreeDraw: 0.020
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(I) (SaveToJSONFile)(C0) Number of bytes written to JSON file /effects.cfg: 4110
[  8084][E][WiFiClient.cpp:517] flush(): fail on fd 54, errno: 11, "No more processes"
(I) (UpdateSubscribers)(C1) Got YouTube subscriber count for channel Daves Garage (GUID 9558daa1-eae8-482f-8066-17fa787bc0e4)
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 40196, LargestBlk: 31732, PSRAM Free: 395907/4173083, LED FPS: 36 Refresh: 60 Hz, Power: 1710 mW, Brite: 100%, Audio FPS: 47, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 075%, 055%, FreeDraw: 0.013
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 40196, LargestBlk: 31732, PSRAM Free: 395379/4173067, LED FPS: 46 Refresh: 60 Hz, Power: 1710 mW, Brite: 100%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 061%, 041%, FreeDraw: 0.007
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 40196, LargestBlk: 31732, PSRAM Free: 395379/4173067, LED FPS: 47 Refresh: 60 Hz, Power: 1710 mW, Brite: 100%, Audio FPS: 43, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 061%, 037%, FreeDraw: 0.011

I did not tinkered with the code robert, i just downloaded it! And uploaded it. .

heidepiek, 2023-11-27:
Little bit shaky but this is what he does: 
https://drive.google.com/file/d/1fgDO394v6c-qoGPHwzUGK_QSz7QE0NIe/view?usp=sharing

rbergen, 2023-11-27:
Those are all animated GIF effects. It could be memory, but it could also be something in the new animated GIF dependency ([GifDecoder](https://github.com/pixelmatix/GifDecoder)) that disagrees with whatever specific hardware you are using.

heidepiek, 2023-11-27:
I`m using a lolin d32 pro v2 with 16mb flash and 8mb psram indeed.  So all the effects that are new are gif? colorball atomic three rings and pacman are the effects my lolin does not like! I get this in the serial when i click on them:

--- Terminal on COM9 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39836, LargestBlk: 31732, PSRAM Free: 395907/4173083, LED FPS: 30 Refresh: 60 Hz, Power: 3066 mW, Brite: 100%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 051%, 045%, FreeDraw: 0.014
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39836, LargestBlk: 31732, PSRAM Free: 395379/4173067, LED FPS: 30 Refresh: 60 Hz, Power: 3052 mW, Brite: 100%, Audio FPS: 47, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 053%, 048%, FreeDraw: 0.014
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39836, LargestBlk: 31732, PSRAM Free: 395379/4173067, LED FPS: 30 Refresh: 60 Hz, Power: 3031 mW, Brite: 100%, Audio FPS: 43, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 051%, 044%, FreeDraw: 0.021
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39836, LargestBlk: 31732, PSRAM Free: 395379/4173067, LED FPS: 30 Refresh: 60 Hz, Power: 2864 mW, Brite: 100%, Audio FPS: 45, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 051%, 046%, FreeDraw: 0.017
(W) (NotifyJSONWriterThread)(C0) >> Notifying JSON Writer Thread
Successfully opened GIF; Canvas size = 64 x 12
(W) (NotifyJSONWriterThread)(C0) >> Notifying JSON Writer Thread
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4011a6a6  PS      : 0x00060130  A0      : 0x8011b020  A1      : 0x3ffe2bf0  
A2      : 0x3f8019f8  A3      : 0x00000002  A4      : 0x3f802ba8  A5      : 0x3f801a68  
A6      : 0x3f802ba6  A7      : 0x3f8076f8  A8      : 0xaab8a8a2  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0xaab8a8a2  A13     : 0x00000009  
A14     : 0x00000000  A15     : 0x00000040  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xaab8a8a2  LBEG    : 0x4008c45c  LEND    : 0x4008c467  LCOUNT  : 0x00000000  

Backtrace: 0x4011a6a3:0x3ffe2bf0 0x4011b01d:0x3ffe2c50 0x400df747:0x3ffe2cb0 0x400df820:0x3ffe2ce0 0x400d9209:0x3ffe2d00 0x400d9309:0x3ffe2d20 0x40081c4f:0x3ffe2d50

  #0  0x4011a6a3:0x3ffe2bf0 in GIFMakePels(gif_image_tag*, unsigned int) at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:752
      (inlined by) GIFMakePels at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:881
  #1  0x4011b01d:0x3ffe2c50 in AnimatedGIF::playFrame(bool, int*, void*) at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:985
      (inlined by) AnimatedGIF::playFrame(bool, int*, void*) at .pio/libdeps/mesmerizer/AnimatedGIF/src/AnimatedGIF.cpp:201
  #2  0x400df747:0x3ffe2cb0 in GifDecoder<64, 32, 12, false>::decodeFrame(bool) at .pio/libdeps/mesmerizer/GifDecoder/src/GifDecoder_Impl.h:402
  #3  0x400df820:0x3ffe2ce0 in PatternAnimatedGIF::Draw() at include/effects/matrix/PatternAnimatedGIF.h:311
  #4  0x400d9209:0x3ffe2d00 in EffectManager::Update() at include/effectmanager.h:649
  #5  0x400d9309:0x3ffe2d20 in LocalDraw() at src/drawing.cpp:120
  #6  0x40081c4f:0x3ffe2d50 in DrawLoopTaskEntry(void*) at src/drawing.cpp:280

ELF file SHA256: 894b99bb89c0a3fd

E (3674) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (1220) esp_core_dump_flash: No core dump partition found!
E (1220) esp_core_dump_flash�r��core dump partition found!
Replacing Idle Tasks with TaskManager...
(I) (PrintOutputHeader)(C1) NightDriverStrip
(I)
(I) (PrintOutputHeader)(C1) ------------------------------------------------------------------------------------------------------------
(I) (PrintOutputHeader)(C1) M5STICKC: 0, USE_M5DISPLAY: 0, USE_OLED: 0, USE_TFTSPI: 0, USE_LCD: 0, USE_AUDIO: 1, ENABLE_REMOTE: 1
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: OK
(I) (PrintOutputHeader)(C1) Version 40: Wifi SSID: "heidepiek" - ESP32 Free Memory: 200632, PSRAM:4191643, PSRAM Free: 4161999
(I) (PrintOutputHeader)(C1) ESP32 Clock Freq : 240 MHz
(I) (setup)(C1) Startup!
(I) (setup)(C1) Starting DebugLoopTaskEntry
>> Launching Debug Thread.  Mem: 200632, LargestBlk: 110580, PSRAM Free: 4161999/4191643, (W) (ReadWiFiConfig)(C1) Retrieved SSID and Password from NVS: "heidepiek", "********"
(W) (setup)(C1) Starting ImprovSerial for ESP32
[  1173][E][vfs_api.cpp:182] remove(): /improv.log does not exists or is directory
>> Launching JSON Writer Thread.  Mem: 192064, LargestBlk: 110580, PSRAM Free: 4161963/4191627, (I) (LoadJSONFile)(C1) Attempting to read JSON file /device.cfg
(I) (DeviceConfig)(C1) Loading DeviceConfig from JSON

Starting SmartMatrix Mallocs
Heap/32-bit Memory Available: 186668 bytes total, 110580 bytes largest free block
[... cut ...]
[  8045][E][WiFiClient.cpp:517] flush(): fail on fd 54, errno: 11, "No more processes"
(I) (UpdateSubscribers)(C1) Got YouTube subscriber count for channel Daves Garage (GUID 9558daa1-eae8-482f-8066-17fa787bc0e4)
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 2
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 39968, LargestBlk: 31732, PSRAM Free: 395099/4173035, LED FPS: 30 Refresh: 60 Hz, Power: 2847 mW, Brite: 100%, Audio FPS: 41, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 068%, 060%, FreeDraw: 0.023

heidepiek, 2023-11-27:
The specs of my board are: wrover e.  16mb flash and 8mb psram. I don't think that' qualifies as s3. 
I hope the software guys figure out why my lolin keeps crashing.  And find a solution.

heidepiek, 2023-11-27:
All i know is that when i tried to upload rhe mesmerizer on to my boards that i had. It failed!
Dave's mesmerizer uses if i recall correctly a wrover e with 8mb flash and 8mb psram. So i bought  the lolin d32 pro with 16mb flash and 8mb psram. And for me that did the trick. Till today i had no problems. I don't think i have something else in my configuration. The pinout  to the matrix is not tampered with. The mic is on the same pinout, and also the ir sensor. So why it crashes now beats  me. You can tel me where it goes wrong Just by looking at the code. So the only thing that i can think of that is unique is that it is not the original board dave has, but a lolin.

rbergen, 2023-11-29:
While doing work on #550, I also ran into this issue on an original Mesmerizer board. The board now consistently crashes when it starts an instance of the animated GIF effect. 

The logging around the time of the crash looks as follows:

```log
Successfully opened GIF; Canvas size = 32 x 32
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4011a89e  PS      : 0x00060e30  A0      : 0x8011b218  A1      : 0x3ffbc3c0  
A2      : 0x3f801a28  A3      : 0x00000005  A4      : 0x3f802bd8  A5      : 0x3f801a98  
A6      : 0x3f802bd3  A7      : 0x3f807728  A8      : 0xeafaa4ab  A9      : 0x0000000e  
A10     : 0x00000029  A11     : 0x00000000  A12     : 0xeafaa4b9  A13     : 0x0000000f  
A14     : 0x000000c0  A15     : 0x00000020  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xeafaa4b9  LBEG    : 0x4008c468  LEND    : 0x4008c473  LCOUNT  : 0x00000000  

Backtrace: 0x4011a89b:0x3ffbc3c0 0x4011b215:0x3ffbc420 0x400df7bf:0x3ffbc480 0x400e77f2:0x3ffbc4b0 0x400d920d:0x3ffbc4e0 0x400d930d:0x3ffbc500 0x40081cb7:0x3ffbc530
```

When I disable the animated GIF effects, the board runs normally.

rbergen, 2023-11-29:
Yeah, in my rush to capture this between things, I used the wrong serial monitor. This is the stack trace:

```log
#0  0x4011a817:0x3ffbc3d0 in GIFMakePels(gif_image_tag*, unsigned int) at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:752
      (inlined by) GIFMakePels at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:881
  #1  0x4011b191:0x3ffbc430 in AnimatedGIF::playFrame(bool, int*, void*) at .pio/libdeps/mesmerizer/AnimatedGIF/src/gif.inl:985
      (inlined by) AnimatedGIF::playFrame(bool, int*, void*) at .pio/libdeps/mesmerizer/AnimatedGIF/src/AnimatedGIF.cpp:201
  #2  0x400df823:0x3ffbc490 in GifDecoder<64, 32, 12, false>::decodeFrame(bool) at .pio/libdeps/mesmerizer/GifDecoder/src/GifDecoder_Impl.h:402
  #3  0x400df8fc:0x3ffbc4c0 in PatternAnimatedGIF::Draw() at include/effects/matrix/PatternAnimatedGIF.h:311
  #4  0x400d9205:0x3ffbc4e0 in EffectManager::Update() at include/effectmanager.h:649
  #5  0x400d9305:0x3ffbc500 in LocalDraw() at src/drawing.cpp:120
  #6  0x40081cb7:0x3ffbc530 in DrawLoopTaskEntry(void*) at src/drawing.cpp:280
```

davepl, 2023-11-29:
I’ve seen it die in GIFMakePels before, but the a rebuild and reupload seemed to fix it, and never saw it again.

When it happened, I backed out one line of code at a time until it worked, then added it back, and it still worked.  Any help in debugging a repro would be appreciated!

-Dave

heidepiek, 2023-11-30:
it is 6.21, I have to haul my butt to work. But wanted to test the code again. Dave closed the animated gif pull. So tried it just a minute ago. But no luck!!!! I have now 5 effects that crash my mesmerizer??!

Harrie

rbergen, 2023-11-30:
The timing of this is unfortunate, because it's likely I won't be able to retest with the current state of main, or the proposed changes in #553, until tomorrow EOD. However, if the problems are still there when I can test (as @heidepiek's most recent tests are now suggesting) then I'm considering pushing a commit with a forced effects settings wipe that disables the Animated GIF effects by default.

heidepiek, 2023-11-30:
Accidentally posted something in a pull request that is already closed. Emotion😞  Just copy it here again. Maybe rutger can remove the other one. Or that I can do it myself.

Well dave, that saying of yours doesn't apply to me! It doesn't work in my office! And why? No idea! If I got it right your board uses an ESP32-WROVER-E-N8R8. What is different about that as the chip my wemos lolin D32 pro v2.0 uses? ESP32 chip with 8MB PSRAM and 16RAM (240Mhz dual core processor). With esptool:

((esptool.py v4.5.1
Serial port COM9
Connecting...................
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz

Using the same pinout as your board. Just upload the code, and go. But without the gifs!

Just downloaded another fresh code from the github page nightdriver. And again the same thing, a crashing mesmerizer. Frustrating to the bone! I have hope that rutger, if he has time will have the same problem with his mesmerizer. Then I am not completely alone. Where are the responses from the other contributors who have an original mesmerizer board???? I don't hear anyone! Am I really the only tinkerer who has been pretty active with the mesmerizer lately?
Sorry for my response, but that had to come out. Want so much for my setup to work well too. But you are so convinced it's all good! Had my mesmerizer running so nicely until you started the gif files. Do know that it runs beautifully without them too. But it's just frustrating that then again the gifs don't work with my mesmerizer! Must be patient, rutger has yet to test. But patience is not my virtue😞

rbergen, 2023-11-30:
@heidepiek I just merged #555, which updates main to still add the animated GIF effects, but in a disabled state. That means the effects won't run by default, which means the crashes also don't happen - and this I have confirmed on the same Mesmerizer board that indeed crashes with the other code (previous version of main and the code in #553).

I understand your frustration, but I will have to ask you for some patience and also a slightly calmer tone of voice. It's at least as frustrating for the people who work on this project most as it is for you that this is happening, and in the end nobody is obliged to help out if they don't want to - or can't. That's one of the characteristics of an open source and totally free project, where everyone who contributes is volunteering.

We're working on fixing it, and in the mean time the current state of main will should stop crashing your board, unless you enable the animated GIF effects yourself.

rbergen, 2023-11-30:
#553 doesn't fix the issue for me, and I did test with that. Do note that my Mesmerizer **worked with the #550 firmware** _(i.e. the original "animated GIF" PR)_ **for a while**, and then suddenly started crashing after it had been turned off for a while.

rbergen, 2023-11-30:
Actually, after I quickly test #548, I will erase my board and then do a clean flash of #553, just to make sure I'm not tripping over remnants of earlier bits that got stuck at my end. I'll report back here when I've done that.

rbergen, 2023-11-30:
Stating first that I know this is a suggestion that makes no sense whatsoever from a perspective of logic, but should we maybe see what happens if we don't allocate the GIF decoder in PSRAM, but use regular RAM instead?

rbergen, 2023-11-30:
Yes, and I should have known you did. :)

rbergen, 2023-11-30:
Ok, it's 12:23AM in my neck of the woods, and I have one more early and full work day ahead of me tomorrow. Which means I'm going to have to sign off for today, right about now. Best of luck with continuing chasing this one down while I'm getting some z's!

rbergen, 2023-12-01:
@heidepiek I've just merged #553, which consistently fixes the animated GIF effects on my Mesmerizer. Could you test with the current state of main, and let us know what happens on your setup?

davepl, 2023-12-01:
I just synced and built a new enlistment against main, and it works!

heidepiek, 2023-12-01:
Sorry guys for the late reply. I had car trouble. Just at home, and have to hurry! I have to bring my son to chess  practice. But i have managed to quickly download a New copy of the nightdriver and uploaden it. I could only test it bare. Without the screen. But it doesn't crash if i use the gif files. If i Come home tonight, i wil test it with the matrix. But i think you guys solved it.

heidepiek, 2023-12-01:
Sorry guys for reacting so emotionally!😳  Maybe a tone too high, but was frustrated. Thought for a moment that my lolin board was pointed out as the culprit when rutger changed the title of my issue. When dave also said, it works in my office! That's when I started to get all twitchy.  But luckily you guys found the problem, and the mesmerizer is now running fine on my lolin👌 . 
Thanks guys!💪  

https://github.com/PlummersSoftwareLLC/NightDriverStrip/assets/104028471/2931cc41-bb2d-4ae5-8ef2-bbed59e63afe

@robertlipe You asked me if i had broken pixel. Yep, in two of my matrices. The movie i send you has 2 dead pixels in the upper left corner. And the one in this comment has 1 dead pixel above somewhere in the middle🙈. Bummer! Have to order new ones.

Harrie

rbergen, 2023-12-01:
@heidepiek Thanks for confirming that things work for you as well now - I will close this issue after this comment. 

About the points you mentioned:

- I changed the title of this issue twice. First to point out that it was reported on a Lolin - at the time, the effects worked fine on my official Mesmerizer. Then again, to express that the issue was a broader issue (also occurring on "official" Mesmerizer boards), and specifically related to the animated GIF effects. There is no accusation in any of that, but it really helps when the name of an issue expresses specifically what the identified problem is. First it was only seen on a (your) Lolin board, then it was seen on Mesmerizer boards as well - I made the issue name follow what was seen.

- The "it works in my office" comment was a general programmer joke. Another version of it is "it works on my machine" - it's wat many developers say when they're confronted with bugs that happen outside of the very controlled environment that is the computer they do their development on.  
You can trust me when I say that Dave didn't really mean he didn't care about the problem; what he was actually saying is that it's hard to analyze a problem he wasn't experiencing himself at the time.

In any case, all's well that ends well, and I'm comfortable concluding that we can now indeed stop worrying about this very hard-to-grasp bug. Thanks (again) to @robertlipe for pointing us in the right direction of the root cause, and to @davepl for figuring out the fix!