PlummersSoftwareLLC/NightDriverStrip #579: SPIFFS Failing on a 16MB ESP32-S3-DevKitC1

GameTec-live, 2023-12-27:
## Bug report
ESP32-S3-DevKitC-1 8M PRAM 8MB Flash

**Problem**
On a ESP32-S3-DevKitC-1 8M PRAM 8MB Flash i can compile and flash the Chip but SPIFFS Seems completly broken. It cant mount, format or similar and therefore it doesnt work and constantly breaks.

Probalby not relevant, but im trying to drive a LED Matrix...

**Steps**

1.  Expand the Demo script to use the esp32-s3...
[image]

3.  Enable Wifi / Anything that needs IPFSS
4.  See it fail

Yes, i have tried to add in other parametrs like using the 8MB table:
[image]

**Example**

**Notes**

GameTec-live, 2023-12-27:
> I use the 16mb versions of those boards. They're fine with spiffs.  Have
> you perhaps reused the gpio lines that are repurposed for spi flash/ram?
> 
> Does the board. Pass the psram test earlier?
> 
> 
> > Bug report
> >
> > ESP32-S3-DevKitC-1 8M PRAM 8MB Flash
> >
> > *Problem*
> > On a ESP32-S3-DevKitC-1 8M PRAM 8MB Flash i can compile and flash the Chip
> > but SPIFFS Seems completly broken. It cant mount, format or similar and
> > therefore it doesnt work and constantly breaks.
> >
> > Probalby not relevant, but im trying to drive a LED Matrix...
> >
> > *Steps*
> >
> >    1.
> >
> >    Expand the Demo script to use the esp32-s3...
> >    image.png (view on web)
> >    2.
> >
> >    Enable Wifi / Anything that needs IPFSS
> >    3.
> >
> >    See it fail
> >
> > Yes, i have tried to add in other parametrs like using the 8MB table:
> > image.png (view on web)
> >
> > *Example*
> >
> > *Notes*
> >
> > <https://github.com/PlummersSoftwareLLC/NightDriverStrip/issues/579>, or
> > unsubscribe

There's one with 16MB flash? Oh wow...
And what psram tests are we talking about? I don't see any in the pio menu?

GameTec-live, 2023-12-27:
the plot thickens... the box has 2 stickers on top of each other, top one stating its a n8r8 with 8 flash and 8 PSRAM, the one below saying its a n32r8v which would mean it has 32 flash and 8 psram. The chip on the board is a wroom2, so i got the one with more flash as no wroom2 with 8mb exists?

GameTec-live, 2023-12-27:
Okay, i do have the ESP32-S3-DevKitC-1-N32R8V apperently... 32MB Flash and 8MB PSRAM. Redid the entire setup so that i can document the exact changes i made and the result.
PlatformIO under the `[env:demo]`:
- change extends to `dev_32-s3`
- change build flags to `${dev_esp32-s3.build_flags}`
- add `${psram_flags.build_flags}`
- add `board_build.partitions = config/partitions_custom_8M.csv`
- add `board_upload.flash_size = 32MB` (i also tried leaving it at 8MB, samilar issue)
- add `board_build.flash_mode = dio` (it looks right from what i understand from the docs, but i have no clue about what exactly...)
- add `board_build.arduino.memory_type = opi_opi` (i also tried with the flash_size at 8M like in dev_mesmerizer, doesnt work either)

Log from clicking on "monitor":
 https://hastebin.skyra.pw/pupecadaje.yaml

My globals.h:
[globals.h.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/13781464/globals.h.txt)

GameTec-live, 2023-12-27:
it looks like the watchdog is timing out? Setting the led matrix to 2x2 for example works fine, theory of someone i talked to is that its trying to adress all of the LEDs which arent attached rn (testing on the bench) and therefore they all time out after one another, this results in a watchdog timeout?

GameTec-live, 2023-12-27:
this pio config now seems to work... ill just reopen this issue if any more issues arrise:
```
[env:demo]
extends         = dev_esp32-s3
build_flags     = -DDEMO=1
                  ${dev_esp32-s3.build_flags}
                  ${psram_flags.build_flags}
board_build.partitions = config/partitions_custom_8M.csv
board_upload.flash_size = 32MB
board_build.flash_mode = qio
board_build.arduino.memory_type = opi_opi
```

GameTec-live, 2023-12-28:
@robertlipe thanks for that information, but...
Another issue cropped up... i can configure it to drive a 30x30 matrix and fun stuff like that, but driving a 50x30 or 50x50 matrix is impossible without it panicing? (StoreProhibited)

Monitor Log:
https://hastebin.skyra.pw/wikevijele.yaml

rbergen, 2023-12-28:
This seems to be a completely different problem indeed, maybe related to #534 (but, maybe not). To avoid confusion about what Issue discusses which scenario, please open a new Issue for the problem you're now running into. I'm closing this one, again.

closed 2023-12-28 by rbergen

GameTec-live, 2023-12-28:
but sure, ig ill just open a new issue then

rbergen, 2023-12-28:
Thanks. This issue discusses a problem with SPIFFS on the device in question. You're now running into a problem with driving larger matrices, which really is a different issue.

As a general rule, we like to keep issues as focused as possible, as they represent potential work items. Discussions in the equally named section of this repo are known to (and allowed to) meander a bit - or sometimes a lot - more.

GameTec-live, 2023-12-28:
1500 LEDs... (50x30 matrix) The generic normal esp32 i got there runs the exact same config flawless (globals.h config), just had to disable the webserver, and some other stuff id like to have, to get a decent performance over the network, so i thought id get a more powerful chip to replace it which then should be able to provide the web ui, etc...

GameTec-live, 2023-12-28:
Created a new issue as requested... #580
Have a good night robertlipe...

GameTec-live, 2024-09-16:
@robertlipe should be done