PlummersSoftwareLLC/NightDriverStrip #580: Panic when driving larger matrixes (one long strip) on 32MB 8MB ESP32-S3-DevKitC1

GameTec-live, 2023-12-28:
## Bug report
32MB FLASH 8MB PRAM ESP32-S3-DevKitC1(ESP32-S3-DevKitC-1-N32R8V)

**Problem**

**Steps**

1.  Modify env:demo to compile properly for the chip (#579)
  ```
  [env:demo]
  extends         = dev_esp32-s3
  build_flags     = -DDEMO=1
                    ${dev_esp32-s3.build_flags}
                    ${psram_flags.build_flags}
  board_build.partitions = config/partitions_custom_8M.csv
  board_upload.flash_size = 32MB
  board_build.flash_mode = qio
  board_build.arduino.memory_type = opi_opi
  ```
2. In global.h define a matrix width and height larger than 36 (so 37+ results in the panic)
3. See core 1 panic and the system reboot

**Example**

**Notes**
In this case a "Matrix" is just a bunch of daisychained LED strips going back and forth.

My globals.h
[globals.h.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/13785889/globals.h.txt)

Same exact config file works fine on a generic, less powerful, esp32.
And reportedly running 1500 leds on one controller isnt optimal, but it works fine and i get a decent frame rate on my underpowered esp32. It shouldnt crash anyways ;)

Monitor Log:
https://hastebin.skyra.pw/wikevijele.yaml

rbergen, 2024-01-04:
@GameTec-live The hastebin link at the end of your description leads to an empty page with a blinking cursor in my browser.

GameTec-live, 2024-01-04:
> @GameTec-live The hastebin link at the end of your description leads to an empty page with a blinking cursor in my browser.

oh, weird... ill reupload later...

GameTec-live, 2024-01-04:
[crashlog.log](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/13834734/crashlog.log)
Sorry for the late reply, but here you go...

GameTec-live, 2024-01-04:
changed `build_type` under `base` from `release` to `debug` and set `monitor_filters = esp32_exception_decoder` under `[env:demo]`

heres the new 
[debug-log.log](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/13836222/debug-log.log)
log:

GameTec-live, 2024-01-05:
probably the stupidiest thing youve heard in a while, but.... i do have a pico debug probe (SWD) and cant figure out where im supposed to hook it up...

GameTec-live, 2024-01-05:
Okay, thanks for the help, i managed to get a debugger running (it was a driver issue...)... I don't know what the value of mCur is supposed to be, etc... I'll poke around a bit more and report back...

GameTec-live, 2024-01-05:
cant really see anything unusal? (i mean; i also dont know what those values are supposed to be lol)
mPixelData seems to be large or even "infinite" as i can request a lot more than 1500 array entries with the debugger... mCur also goes past 1500...

rbergen, 2024-01-07:
@GameTec-live Glad to see you got the debugger working, and I appreciate the earlier upload of the debug log. As @robertlipe already indicated, it does show that the actual problem (which is a form of illegal memory access) takes place several levels below "our" code. In fact, the backtrace doesn't even include references to any code that's part of NightDriverStrip. 

Concerning your last comment, we're obviously not looking over your shoulder, so we can't see what you are looking at. Also, even if we could then figuring out what the cause of the invalid behavior is would effectively require us to debug the dependency libraries involved. Which isn't entirely impossible, but very difficult if we can't ourselves debug trace through the code just before the problem occurs. 

Without having the hardware and software setup that triggers these crashes available, I therefore think we won't be able to solve this. You could (still) consider raising a bug report in the dependent libraries (Espressif ESP-IDF and/or FastLED) and see if they are able to provide pointers to what's actually behind this.

I'll leave this issue open in case someone else runs into the same problem, and may be able to provide additional information that can help get to the root of this.

GameTec-live, 2024-01-07:
Yeah, kinda hard to replicate a issue without hardware... Ill open a issue over at FastLED... Thanks for the help though...

prschguy1, 2024-01-09:
hey [GameTec-live](https://github.com/GameTec-live)
While i am not one of the bigger brains on this repository, perhaps I can provide some help to the problems you are experiencing.
here. My understanding is that demo is intended for strip effects only. I was a bit surprised that you tried putting spectrum into a demo build. have never tried that, but assumed it wouldn't work as newer spectrum builds use dma. .Instead of using demo as your build, might suggest using spectrum-elecrow, I have that working properly on the chip you specify. it looks good with pdm mic, remote, display, and all the effects. when I run it with strip effects, I run out of memory, but runs well  on spectrum. might give it a try. Memory usage looks pretty good here. just my 2 cents. 
[image] 

https://github.com/PlummersSoftwareLLC/NightDriverStrip/assets/69419979/5ded102a-77ff-4674-a675-4e96acdfb413

GameTec-live, 2024-01-09:
> hey [GameTec-live](https://github.com/GameTec-live)
> While i am not one of the bigger brains on this repository, perhaps I can provide some help to the problems you are experiencing.
> here. My [...]

thx for the info, ig ill try that... My matrix is just one long strip though...

GameTec-live, 2024-01-11:
altough using the spectrum-elecrow project seems to work, it doesnt help me much as spectrum drives hub whatever its called matrixes and i just have one long strip snaking back and forth...

prschguy1, 2024-01-11:
All of the spectrum builds use a series of strips that zig-zag back and forth as you describe. They can be identified as there is only 1 pin used for output led_pino. The hub75 is currently only used for the Mesmerizer builds that require around 14 output pins depending on what you are doing. Dave does a great job of describing all of this here:

https://www.youtube.com/watch?v=COJnlehBcKw&t=224s

The video I posted is using a standard zig-zag strip as you describe at 16 pixels high by 48 pixels wide on spectrum build, using the chip you specified.

GameTec-live, 2024-01-12:
ah, ok... so ig I will fumble a bit more with spectrum elecrow and try to use that... thanks...

GameTec-live, 2024-01-12:
nvm, setting the matrix to the required site crashes elecrow too...

prschguy1, 2024-01-12:
Perhaps if you could articulate exactly what you are trying to accomplish, we might be able to help. As far as I am seeing, this chip is working without any crashes for the spectrum build. While there aren't a lot of spectrum effects here, intend to test it more thoroughly.

GameTec-live, 2024-01-13:
> Perhaps if you could articulate exactly what you are trying to accomplish, we might be able to help. As far as I am seeing, this chip is working without any crashes for the spectrum build. While there aren't a lot of spectrum effects here, intend to test it more thoroughly. 

drive my 50x30 matrix (being one long strip) with the new, more powerful ESP32-S3... It works fine with my current, not as powerful ESP32 (afaik its even singlecore?),  had to disable nice to haves like the webserver though for it to run a stream from the computer at a decent framerate which im hoping to fix with this a lot more powerful variant...

GameTec-live, 2024-02-12:
Stupid question, but can some of the devs or someone more competent than me try and compile demo with a 50x30 matrix?
I tried to compile elecrow, 50x30, similar error. Tried to compile demo for a seeed studio esp32 c3 (ik, not officially supported) and it's still the same error (from what it looks like)
Havnt tried a nodemcu (clone) yet as thats currently driving the matrix and id rather not break it until my replacement mcu works... :/

GameTec-live, 2024-02-12:
Ok, that makes sense then, was just the one I had laying around... Id still be interested if someone else can replicate this issue or if its just me or maybe even a defect MCU... And having a reproducable thing might help speed things up here (or over at fastLED)

prschguy1, 2024-02-12:
Hello [GameTec-live], have looked at your situation a good bit over the last couple of weeks, and do get similar results. While I can approach your matrix size, cannot quite get there. Have a similar problem when trying to use this chip with 4 channel strip effects. Have tried  different board define files as well as different memory tables, but still have not solved this problem. While an alternate led program makes both of our build problems work, I would like to get it working here. What I have learned is programs can make use of 16 mb, and the 32mb that these chips support is only useful for storage. Have tried both of our builds with unexpected maker s3 pro, wemos s3, and generic and official builds of esp32-s3 in various memories. Have a wemos d32 pro, and m5stack I'll try our build on next. M5 stack used to work on my build, but suspect that no longer works. Will let you know what I find.

GameTec-live, 2024-02-13:
Ah, so it isn't just me XD
Thanks for trying to help though.

GameTec-live, 2024-05-15:
@robertlipe while ordering other stuff, i threw in a N16R8, so the 16MB version you apperently use... I still get the same panic...

davepl, 2024-05-15:
> Seems we're getting no traction with FastLED looking into this.
>
> I'm about to be on the road through the end of the month, so I can't commit
> to looking into this in the short term, though I know this has been
> simmering a long time. Looks like I may have to debug FastLED. 😐
>
> Is there a repro case for this here? Does it take a while to bonk or is it
> close to immediate?

The S3 LEDSTRIP project with the onboard LED will fault out reliably.  Only problem is the S3 reboots without showing a call stack or exception chain!
- Dave

rbergen, 2024-05-15:
Should we maybe include the fix for this in #626? 

I believe it is as simple as not touching the on-board LED. We could achieve that by just commenting out the respective `#ifdef ESP32FEATHERTFT` block in globals.h.

davepl, 2024-05-15:
I would… there might be an ONBOARD_LED define or something that we can turn off for that build, as it’s not on in most.
- Dave

rbergen, 2024-05-15:
I think the on-board LED code is only activated if `ONBOARD_PIXEL_POWER` is defined.

GameTec-live, 2024-05-19:
So, it seems like #626 is supposed to contain a fix for this? I pulled down that branch/fork, compiled it, nothing, still panics.
For clarity ive now generated git diffs of the exact changes i made and once again uploaded the logs. (For both of my controllers)
(the N16R8 log is a bit wierd though, as it doesnt spit out the core dump? It did before, with the same illegal cache access, but when capturing the log it just didnt...)
[N16R8-crashlog.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15368822/N16R8-crashlog.txt)
[N32R8V-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15368823/N32R8V-diff.patch)
[N32R8V-crashlog.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15368824/N32R8V-crashlog.txt)
[N16R8-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15368825/N16R8-diff.patch)

rbergen, 2024-05-19:
No, this is a different problem than the one the latest change in #626 is trying to address. That fixes random crashes without any panic logging on the S3 with the Feather project "as standard", after the device has been running for a while.

Your controllers seem to consistently crash as soon as WiFi tries to connect, with PSRAM enabled. I think I remember Dave configuring PSRAM to be off on all devices _except Mesmerizer_ because he was seeing exactly this behavior.

GameTec-live, 2024-05-19:
Ah, ok... Well, i tried leaving the psram off (not adding the build flag) too, but it didnt work either, i can try again later and send some logs for that too...

GameTec-live, 2024-05-19:
Well, observation is usually the first step to figuring stuff out? XD

Anyways, enjoy the other 2 log files and diffs of the 2 PSRAM less builds:

[N32R8V-noPSRAM-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15369242/N32R8V-noPSRAM-diff.patch)
[N32R8V-noPSRAM-crashlog.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15369243/N32R8V-noPSRAM-crashlog.txt)
[N16R8-noPSRAM-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15369244/N16R8-noPSRAM-diff.patch)
[N16R8-noPSRAM-crashlog.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15369245/N16R8-noPSRAM-crashlog.txt)

davepl, 2024-05-19:
> Agreed. It looks like it IS a different problem.
>
> So we have Dave's problem of treating PSRAM pins as LEDs.
> We have Gametec's issue of the crash inside FastLED when "lots of LEDs" are
> used. (Is network traffic a key?)
>
> What's the third? Sorry. My on-board CPU is running slow today.

>  Dave's problem of treating PSRAM pins as LEDs. 

Can you explain?

- Dave

rbergen, 2024-05-20:
> Agreed. It looks like it IS a different problem. So we have Dave's problem of treating PSRAM pins as LEDs. We have Gametec's issue of the crash inside FastLED when "lots of LEDs" are used. (Is network traffic a key?) What's the third? Sorry. My on-board CPU is running slow today.

Problem number 3, as I remember it(!), is S3 boards crashing at WiFi connect when PSRAM is enabled. Which I may have misremembered, in which case there is no problem number 3.

GameTec-live, 2024-05-20:
@rbergen you seem to be misremembering, as atleast for me on my 2 MCUs, with PSRAM enabled, driving a smaller matrix (eg 10x10) boots up perfectly, etc...
For completeness have the logs and git diffs anyways:
[N16R8-smallWorking-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15374655/N16R8-smallWorking-diff.patch)
[N32R8V-smallWorking-diff.patch](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15374657/N32R8V-smallWorking-diff.patch)
[N32R8V-smallWorking-log.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15374658/N32R8V-smallWorking-log.txt)
[N16R8-smallWorking-log.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/15374659/N16R8-smallWorking-log.txt)

rbergen, 2024-05-20:
No, I'm not - or at least not in that way. The point is that there are now 3 problems with (certain) S3 boards we know of. Your problem - that being the one this issue concerns primarily - is the second problem in @robertlipe's summary. There are two others, though. One is the first Robert mentions, the other the one I mentioned in my previous comment.

rbergen, 2024-05-20:
> That means we have only Gametec's still live and in play, right?

Based on what you say in the rest of your comment, that's quite likely true.

> If we track problems we *used to have*, I'm sure we'll all lose our minds even more quickly. 

That may be true for some, many or most, but not me. I need to keep some record of problems we used to have to retain _my_ sanity. If only because some "solved" problems have the unpleasant habit of rearing their heads again sometime later.

But, I can be quiet about it if that's generally preferred. :)

davepl, 2024-05-20:
We should certainly have a bug/issue to track this.  If it never manifests again, we can close or ignore it, but shouldn’t forget about it!

- Dave

GameTec-live, 2024-06-30:
Dont want to be annoying (greatly apprecite the work your doing here), but has there been any progress on this issue?

davepl, 2024-06-30:
Sorry, I’m not up to speed on this one, can someone refresh my memory?

GameTec-live, 2024-07-02:
So progress but also not? *fun*

Thanks for the update though

GameTec-live, 2024-09-05:
As the FastLED issue has been closed unceremoniously last week, I decided to try it again. Cloned the repo again, made the required changes, built, uploaded aaaand... Panic... Yaaay...

rbergen, 2024-09-05:
That's obviously disappointing. "Made the required changes" includes making sure PlatformIO pulls in the right version of FastLED? I don't have the code in front of me, but I half-remember the FastLED version was pinned.

GameTec-live, 2024-09-05:
In the PIO logs it said that it pulled version 3.7.5 (or whatever the latest version is)

rbergen, 2024-09-05:
Thanks for checking. Well, I guess that settles that then. In the sense that it doesn't.

GameTec-live, 2024-09-05:
😂 yep, sadly lol

GameTec-live, 2024-09-05:
Ig, but idk how id reproduce this in plain FastLED?

zackees, 2024-09-06:
> You can reopen the issue. Zack's trying to burn down the list of ancient bugreports that have been open for up to 8 years and apparently just cut too deeply. Be prepared to build up a test case that repros this in "raw" FastLED, though. It's increasingly time to replace FastLED with with Makuna NeoPixel or maybe I2SClocklessLedDriver. Maybe. I'm leaning toward the former. But given the requirement of building and running every combination, I cannot undertake that here. I'll probably eventually do it in a fork and get "advanced" features like RGBWW, changing pin numbers and strip types without recompiling, etc.
> […](#)

I already tried to import the I2SClocklessDriver.

Here's the bad news: Doesn't work in C3,S3,C6, only works on Esp32dev.

The developer said he has a working version for s3 but I didn't see it, maybe I didn't look good enough.

Additionally the readme states that the library works on Arduino/PlatformIO as a drop in for FastLED.

Problem is: The developer forgot to put in a header include for the library.json file. I fixed the issue and issued a pull request. But it's been three days and the developer hasn't merged it in.

I looked at the neopixel bus to see what they did and - yup, same problem, only works on ESP32DEV and all other platforms are compiled out.

Major bummer. Though I like the thought of running 24 channels of I2S WS2812 data, the driver is about only 33% done, realistically for someone that is new and wants to come in and fix it. For the original developer, this is probably 90% done. I've reached out to the developers of this to help them get it in for the library but haven't heard back.

Bummer, because this is a fantastic feature. But the ESP32DEV user base is just too small to justify me wrapping my head on all the crazy espressif headers to get this done. Our user base wants WS2812 RGBW and that's my top priority for the next release.

GameTec-live, 2024-09-06:
Ill try and poke a bit more at nightdriver and fastled... for reference ill also drop the links to the exact 2 boards i have (incase someone is crazy enough and wants to try on the exact boards):
https://www.berrybase.at/espressif-esp32-s3-devkitc-1-n32r8v-esp32-s3-wroom-2-8mb-psram-32mb-flash
https://de.aliexpress.com/item/1005006418608267.html

zackees, 2024-09-06:
> While you have that can open, Zack, please add another W to that request. Another project just went to the effort to add RGBW, then their top request became RGBWW when the additional work from there is mostly mechanical. It's a bummer that atruct Pixel no longer fits in a word which can be a drag on low memory configurations. This problem, though, is likely something that's just not pinned in IRAM. Someone with a debugger just needs to repro it. I've tried to repro it (when filed) and can't. Perhaps with a better repro case, Zach and/or I can repro it in NightdriverLED context. I do agree that there are a ton of packages out there that just haven't kept up in the later silicon, but as existence proof, I have WLED running on S3, son Makuna code must work there, though I have only run it to a few hundred pixels. Oh, one of the things that NightdriverLED had to work around is the antiquated use of 'register'..I haven't looked lately but has that been fixed in FastLED? It's a problem in the post c++2017 world. P. S. Are you Zack or Zach? Over the lunch table I never noticed. On Fri, Sep 6, 2024, 11:04 AM Zachary Vorhies ***@***.***> wrote:
> […](#)

I really don't understand why everyone wants to make the RGBW implementation so freaking difficult.

This is my opinion, but Adafruit did it the completely wrong way. They said: there are 4 pixels being outputted so let's give the user 4 pixels to manipulate.

Everyone that I've talked to about the RGBW issue seems to have the same opinion.

I come from the games industry. We use RGB8 everywhere until HD became a thing, and that was only what, RGB10?

Going beyond RGB8 on the client side seems like absolute madness. Are we going to partition our entire blending / fx code too? Are we going to do it again with RGBWW? Shoot me.

RGB8 It's good enough for games, it's good enough for our users.

So what's my approach?

Convert RGB -> RGBW at the fastled driver level. Boom. Done, easy peezy. All the sketches that work today for the RGB pixel type now magically work on the RGBW output pixel type. Blending code stays the same. If RGBWW becomes the rage, then that's supported too very easily (well for non AVR chipsets unless someone wants to change all that asm code).

What makes this even easier right now is that it appears that the WS2812 makers have so far standardized W as the last component. So 3-component ordering may stay the same. Though I suspect that the entire industry may just assume standardized ordering going forward since it's a major headache for them to support anything but the standard GRB ordering, which becomes the default on the next release.

What's great about the RGBW types is that it seems to reduce power usage. Full blast white for RGB for 30 pixels is 3.8 watts. Full blast white for RGBW (using the white color stealing algorithm) reduces that to 2.6 watts. So it cuts the power down by a 1/3rd.

Experimental support for RGBW will be released on Monday. Fun times.

(btw it's Zach but you can use either spelling - i don't care)

zackees, 2024-09-06:
> Ill try and poke a bit more at nightdriver and fastled... for reference ill also drop the links to the exact 2 boards i have (incase someone is crazy enough and wants to try on the exact boards): [...]

Use our clone-and=compile mode for best results. Just clone our repo, open it up in VSCode (+platformio extension) and hit compile. It's that easy.

Your changes go in [dev/dev.ino](dev/dev.ino). It's already preset to the esp32-s3 using the open source platformio codebase and uses idf 5.1, which is quickly becoming standard.

Once you've repro'd the case, push it to your fork and send me a link and I can see if I can repro it with my XIAO esp32 s3.

GameTec-live, 2024-09-06:
@zackees heres the fun part, I cant rly reproduce it with a simple fastled sketch alone, tried for ~1h now, so ig its the combination of smthng Nightdriver is doing and Fastled thats the problem?

zackees, 2024-09-06:
Can you package it all up into one repo?

GameTec-live, 2024-09-07:
That's like this repo or what do you mean? All you need to do is change the config as ive shown already.I can generate a new patchfile for the current repo versions as the older ones dont apply anymore and youd have to modify the files by hand... I'll send them in a bit...

GameTec-live, 2024-09-07:
[N32R8V-diff.patch](https://github.com/user-attachments/files/16915104/N32R8V-diff.patch)

GameTec-live, 2024-09-07:
@robertlipe appreciate the effort, but then smthng must be different? Ive tried 2 different boards, from different sellers, so I doubt that both just happened to randomly be broken in the same way?

zackees, 2024-09-07:
> That's just not true. I have strips in my collection that use different ordering for the white channels. At least one of them swaps a white with Green. One of them swaps a white with Blue. I can't remember any putting a/either white FIRST, but at least one of the whites always seems to be last.

Okay. This is exactly the real world information that i was looking for but couldn't find. Thank you. Looks like I have to define a total ordering for RGBW then. Luckily there is enough room in the EOrdering enum for one additional component.

Cheers as always.

zackees, 2024-09-12:
Still looking for a repro case. If there isn't an environment to reproduce this issue then I can't begin the fix it process

GameTec-live, 2024-09-13:
@zackees Idk what more you want / I can provide... It's this repo, I published multiple diffs that set up either a working or broken config (also detailed how to do it yourself, diffs just for convenience) and sent links to the exact boards I own (a ESP32-S3-DevKitC1 32MB 8MB (the main subject) and a smaller S3 board too, id be happy if any of those 2 worked...)

Idk what else I can provide? I don't even have to have the matrix connected for the crash to occur, so that isn't a factor... My wifi SSID contains a _, but the crash also happens when putting in my phone hotspot or disabling wifi entirely, so that isn't a factor either...

zackees, 2024-09-13:
I just spent an hour trying to build your project. There are a lot of errors being emitted from the compiler. It could be that you don't have one of your libraries pinned and you are now in a broken state, or something else I can't figure out.

What I need from you is the following:
  * Upgrade to FastLED 3.7.6. Right now you are ^3.4.0 and that is ancient.
  * I need you to create a minimal reproducing branch
    * Just have the demo env and anything else that you need to compile the minimal build.
    * Modify the include/globals.h

Once you have the branch, reply with this link so that I can `git pull` and try again.

By the way, you mentioned that one large alloc of memory works, but that it's not optimal because you can only allocate it to one pin. However, are you aware that FastLED does support segmenting one large LED block into multiple blocks by using offsets?

This might solve your problem or at least kick the can down the road.

zackees, 2024-09-13:
Here are some of the errors emitted:

```bash

✓ built in 407ms

✨ [vite-plugin-compression]:algorithm=gzip - compressed file successfully:
dist/C:/Users/niteris/dev/NightDriverStrip/site/dist/index.html.gz    1.28kb / gzip: 0.49kb
dist/C:/Users/niteris/dev/NightDriverStrip/site/dist/favicon.ico.gz   15.04kb / gzip: 6.32kb
dist/C:/Users/niteris/dev/NightDriverStrip/site/dist/index.js.gz      50.21kb / gzip: 14.96kb

Build completed, html: 501 B, jsx: 15320 B, ico: 6471 B, total: 21 KB
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, 
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IRremoteESP8266 @ 2.8.6
|-- FastLED @ 3.7.6
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit GFX Library @ 1.11.10
|-- U8g2 @ 2.35.27
|-- arduinoFFT @ 1.6.2
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3+sha.a6cd518
|-- ArduinoJson @ 6.21.5
|-- Bounce2 @ 2.72.0
|-- RemoteDebug @ 3.0.5+sha.706f093
|-- QRCode @ 0.0.1
|-- TJpg_Decoder @ 1.1.0
|-- UrlEncode @ 1.0.1
|-- HTTPClient @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- SPI @ 2.0.0
|-- SD @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\demo\src\deviceconfig.cpp.o
Compiling .pio\build\demo\src\drawing.cpp.o
Compiling .pio\build\demo\src\effectmanager.cpp.o
Compiling .pio\build\demo\src\effects.cpp.o
Compiling .pio\build\demo\src\gfxbase.cpp.o
Compiling .pio\build\demo\src\jsonserializer.cpp.o
Compiling .pio\build\demo\src\ledstripgfx.cpp.o
Compiling .pio\build\demo\src\main.cpp.o
Compiling .pio\build\demo\src\network.cpp.o
Compiling .pio\build\demo\src\ntptimeclient.cpp.o
Compiling .pio\build\demo\src\screen.cpp.o
Compiling .pio\build\demo\src\socketserver.cpp.o
Compiling .pio\build\demo\src\webserver.cpp.o
Compiling .pio\build\demo\libdeb\IRremoteESP8266\ir_Carrier.cpp.o
Compiling .pio\build\demo\libdeb\IRremoteESP8266\ir_ClimaButler.cpp.o
Compiling .pio\build\demo\libdeb\IRremoteESP8266\ir_Coolix.cpp.o
In file included from include/systemcontainer.h:37,
                 from include/effects/matrix/spectrumeffects.h:38,
                 from src/drawing.cpp:35:
[... 306 lines cut ...]
include/deviceconfig.h:43:2: error: #error Copy include/secrets.example.h to include/secrets.h, edit to taste, and retry. Please see README.md.
 #error Copy include/secrets.example.h to include/secrets.h, edit to taste, and retry. Please see README.md.
  ^~~~~
include/deviceconfig.h:49:2: error: #error A definition for cszSSID is missing from secrets.h
 #error A definition for cszSSID is missing from secrets.h
  ^~~~~
include/deviceconfig.h:53:2: error: #error A definition for cszPassword is missing from secrets.h
 #error A definition for cszPassword is missing from secrets.h
  ^~~~~
include/deviceconfig.h:57:2: error: #error A definition for cszHostname is missing from secrets.h
 #error A definition for cszHostname is missing from secrets.h
  ^~~~~
include/deviceconfig.h:61:2: error: #error A definition for cszOpenWeatherAPIKey is missing from secrets.h
 #error A definition for cszOpenWeatherAPIKey is missing from secrets.h
  ^~~~~
include/deviceconfig.h:65:2: error: #error A definition for cszLocation is missing from secrets.h
 #error A definition for cszLocation is missing from secrets.h
  ^~~~~
include/deviceconfig.h:69:2: error: #error A definition for bLocationIsZip is missing from secrets.h
 #error A definition for bLocationIsZip is missing from secrets.h
  ^~~~~
include/deviceconfig.h:73:2: error: #error A definition for cszCountryCode is missing from secrets.h
 #error A definition for cszCountryCode is missing from secrets.h
  ^~~~~
  ```

zackees, 2024-09-13:
Yeah that didn't work. Lots of other compilation issues.

I'm halting my work on this bug until a repo branch can be made for testing this bug that I can just clone and compile.

Compilation errors:

```bash
src/main.cpp:221:5: note: in expansion of macro 'debugI'
     debugI("Version %u: Wifi SSID: \"%s\" - ESP32 Free Memory: %u, PSRAM:%u, PSRAM Free: %u",
     ^~~~~~
In file included from include/webserver.h:52,
                 from src/webserver.cpp:32:
include/deviceconfig.h:114:24: error: 'cszHostname' was not declared in this scope
     String  hostname = cszHostname;
                        ^~~~~~~~~~~
include/deviceconfig.h:114:24: note: suggested alternative: 'GetHostname'
     String  hostname = cszHostname;
                        ^~~~~~~~~~~
                        GetHostname
include/deviceconfig.h:115:24: error: 'cszLocation' was not declared in this scope
     String  location = cszLocation;
                        ^~~~~~~~~~~
include/deviceconfig.h:115:24: note: suggested alternative: 'GetLocation'
     String  location = cszLocation;
                        ^~~~~~~~~~~
                        GetLocation
include/deviceconfig.h:116:29: error: 'bLocationIsZip' was not declared in this scope
     bool    locationIsZip = bLocationIsZip;
                             ^~~~~~~~~~~~~~
include/deviceconfig.h:116:29: note: suggested alternative: 'locationIsZip'
     bool    locationIsZip = bLocationIsZip;
                             ^~~~~~~~~~~~~~
                             locationIsZip
include/deviceconfig.h:117:27: error: 'cszCountryCode' was not declared in this scope
     String  countryCode = cszCountryCode;
                           ^~~~~~~~~~~~~~
include/deviceconfig.h:117:27: note: suggested alternative: 'countryCode'
     String  countryCode = cszCountryCode;
                           ^~~~~~~~~~~~~~
                           countryCode
include/deviceconfig.h:118:24: error: 'cszTimeZone' was not declared in this scope
     String  timeZone = cszTimeZone;
                        ^~~~~~~~~~~
include/deviceconfig.h:118:24: note: suggested alternative: 'GetTimeZone'
     String  timeZone = cszTimeZone;
                        ^~~~~~~~~~~
                        GetTimeZone
include/deviceconfig.h:119:33: error: 'cszOpenWeatherAPIKey' was not declared in this scope
     String  openWeatherApiKey = cszOpenWeatherAPIKey;
                                 ^~~~~~~~~~~~~~~~~~~~
include/deviceconfig.h:119:33: note: suggested alternative: 'GetOpenWeatherAPIKey'
     String  openWeatherApiKey = cszOpenWeatherAPIKey;
                                 ^~~~~~~~~~~~~~~~~~~~
```

[7 more replies]