PlummersSoftwareLLC/NightDriverStrip #582: Update from M5Core2 to M5Unified

davepl, 2023-12-30:
It looks like all of the M5 support has been unified into a single library, and the old individual stuff for the M5 Stick / Stack/ Plus/ Core2/ etc will go away.

Someone should move us to the unified lib!

prschguy1, 2024-02-17:
hmmm, have looked at that a bit. It doesn't seem to be a very easy task. Would like to see that work though as I want to see M5Stack-s3 work here. Haven't tried that chipset yet, but support looks pretty good. Suspect it would work well as all M5 products do. Have played around with it a bit, but m5stack_core2  isn't as easy as m5stick. Get a few blowups getting audio to work with the unified libraries. Will continue to work with it, but might need some help with the bigger brains on the project. Have m5 stick plus, and m5 stack_core2 to work with at the moment for testing. Any help would be appreciated.

rbergen, 2024-04-10:
@prschguy1 I just realized I completely missed this when you posted it. What kind of help would you need? (I don't want to claim ownership on the "bigger brains" moniker, but I think it will help me to understand the nature of the challenge better.)

PR #642 (2024-07-16): Add M5 Unified support (M5 Stick C Plus 2, M5 Stack Core 2)
## Description
Moves all M5-based projects to the unified M5 stack

Fixes #582. Fixes #618.

* [X] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [X] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [X] I selected `main` as the target branch.
* [X] All code herein is subjected to the license terms in [COPYING.txt](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/COPYING.txt).
closed 2024-07-16 by rbergen