PlummersSoftwareLLC/NightDriverStrip #586: Hypnosis is a memory hog

davepl, 2024-01-08:
The Hypnosis effect has a static declaration of two bytes per pixel that go into base RAM.  These should be dynamically allocated using PreferPSRAM or similar.  The only catch is the [x][y] addressing in the Draw() code gets harder, but I think we need to reclaim the memory.

davepl, 2024-01-08:
BTW, if the allocation actually happens after setup() is called, this will already be in PSRAM for Mesmerizer, so no win - but if it's allocated before setup() - like as a global - then it will use base ram.

davepl, 2024-01-08:
> Any mallocs over, I thinf, 16 bytes already goes into psram. (It's a number
> way smaller than a frame buffer.) So it's really all coming from a similar
> pool sooner or later,. Assuming it's inside a class that's getting newed
> anyway.
>
> The *real* solution here is to not allocate effects that aren't being
> displayed. Things not in use shouldn't have a cost. But I know there's
> resistance to that.
>
> I'm a bit out of order right now and have a funeral coming up (not
> mine).but if nobody beats me to it,. I'll take this one.

It’s not currently a malloc.  It’s static decl.

davepl, 2024-01-08:
…but it could be contained inside a dynamically allocated object, and therefore be OK.  But I’m not fully cromulent on all the construction paths and their timing.  If they’re loaded dynamically by effects manager, and are NOT instantiated in the table, then it’d be OK.

If you think there are effects that can contract their memory usage, we could just add an End() function that is called like Start(), and effects could manage their memory that way pretty easily.  But I haven’t looked very hard yet!

- Dave

davepl, 2024-01-08:
OK, I read the code :-). All of these things are constructed as a result of InitEffectsManager and hence they’ve got access to PSRAM.  Non-issue, I think!

davepl, 2024-01-08:
> Sounds right. If it's inside the class, it's (sort of) dynamically
> allocated - as a global ctor, I think. (That's the part I groused about.)
> It'd be different if it was outside the class and a global, where it would
> land in .data instead of on the heap. Or the other heap.
>
> From memory (I'm AFK) I *thought* I cleaned all that kind of thing up from
> its arduino-esque heritage when I modernized them, but I couldn't have
> faked shock if I'd missed one or more. In their original form, all the
> effects lived in separated executables (Atmega328, you know...) so
> lifecyle of memory wasn't exactly a concern.
>
> I do think we have more than a few matrix effects that allocate large-ish
> things. I think life, for example, was one as was one of the fires. As a
> broad measuring stick, grep for '[MAT_WIDTH][MAT_HEIGHT];' or maybe with an
> explicit multiplication in declarations.
>
> If that keeps all/most our allocations out of the arena all the time, jolly
> good. I don't think I knew about start(), so if there are things we can
> move from thectors, we should. If ther's someclever reference counting
> voodoo we can do to not manually manage it, even better.
>
> If there's something we can methodically do better, I'd even go back and
> update "my" branch of effects that didn't make it so that anyone wanting to
> manually bring those forward can benefit.

Construct() and Destruct() could be called on Start() and End(), could they not?  Then it would release most memory when not active. 

I don’t see any large matrix-sized allocations in base memory… if you can find one, let me know!

_ Dave

rbergen, 2024-01-09:
> But I’m not fully cromulent on all the construction paths and their timing.  If they’re loaded dynamically by effects manager, and are NOT instantiated in the table, then it’d be OK.

Effects are constructed during the execution of `setup()`, but pretty much towards the end of its invocation. Also, effects are explicitly allocated using `psram_allocator`.  
The table that's put together in effects.cpp only contains the effect factories; this means that in practice the table is a couple of small lambdas for each effect in the list.

> Any mallocs over, I think, 16 bytes already goes into psram. 

It's actually anything over 128 bytes. But considering the previous that shouldn't matter anyway.

> Construct() and Destruct() could be called on Start() and End(), could they not?

The challenge with the destruct part is that we have no "`End()`". An effect is told to prepare itself for drawing through `Start()`, and then told to draw frames through `Draw()`. It is _not_ told when its time in the spotlight has ended.

davepl, 2024-01-09:
Unless I’m just inferring, you keep implying that there are lots of effects with base memory demands, but having looked, I’m dubious.  I’ve been careful to allocate things dynamically from PSRAM in all my own effects, and have fixed most others.  But like I said, if you KNOW about any, call it out so I can fix it.  But call it out or move on, I’d say :-)

I have no had luck reducing the PSRAM allocation auto limit below 96, as WiFi has code that needs base memory but doesn’t spec it, so it gets PSRAM and chokes if you go to 64.

The practical limit of the number of LEDs you can control is a function of refresh rate, which I explain in the latest video.  So haveing 16x300 LEDs would work, but (a) you’d have fewer buffers since they’re larger, and (b) the refresh rate will be terrible.

I run a 5000 LED strip on the Cabana, but it’s broken into four of a about 8x144 each

The umbrella shows that NDS can handle 8 parallel effects, and the tiki lamp can receive multiple channels over wifi, too.  So that all works.

- Dave

> On Jan 9, 2024, at 9:11 AM, Robert Lipe ***@***.***> wrote:
> 
> We have scores of effects. Especially on Matrix, where we may have large 
> secondary rendering buffers or other per pixel data, this might add up - 
> literally doubly so for the 64*64 cases. Are multiple displays unique (e.g 
> 128*128 canvas) or are they pixel mirrored? That, too, is probably a road 
> little traveled.