PlummersSoftwareLLC/NightDriverStrip #596: Blank LED strip since changes to 'improvserial.h'

kriztioan, 2024-01-22:
After syncing my [fork](https://github.com/kriztioan/NightDriverStrip) with upstream, building and uploading the firmware to my `esp32-s3-devkitc-1-n16r8v`, the 107 LEDs on my strip stay blank. The development board is up-and-running, with `Serial` happily reporting all is well. The same for the web server, which allows me to change effects, but the LEDs remain blank. 

Manually bisecting the issue traces it down to commit [Blindly declare presence](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/573/commits/c92e39ea39fd885dbee2686ad7b0ad383d4696cd) made in pull request #573 to `improvserial.h`. Unfortunately, the commit does not provide a rationale for the change. However, undoing [c92e39ea39fd885dbee2686ad7b0ad383d4696cd](https://github.com/PlummersSoftwareLLC/NightDriverStrip/pull/573/commits/c92e39ea39fd885dbee2686ad7b0ad383d4696cd) fixes the issue.

rbergen, 2024-01-22:
@kriztioan Do you have the exact codebase that now works for you (i.e. based on upstream:main minus only the commit you mention) pushed to a branch somewhere? If not, could you make one and provide a link to it? The reason I ask is that I don't understand how a few _potential_ extra bytes being output to serial (which is the only thing that the commit you mention does, with a real chance that in reality _nothing_ changes) would have that effect on the functioning of the LEDs.

kriztioan, 2024-01-22:
> @kriztioan Do you have the exact codebase that now works for you (i.e. based on upstream:main minus only the commit you mention) pushed to a branch somewhere? If not, could you make one and provide [...]

@rbergen Thank you for taking a look. The _exact_ codebase (minus `secrets.h`) working for me is at [i596](https://github.com/kriztioan/NightDriverStrip/tree/i596). Note that this includes my `boards`-directory, `custom_*`-files, and a tiny adjustment to `ledstripeffect.h` that reverses the fill pattern.

rbergen, 2024-01-23:
@kriztioan To be honest this one intrigues me, because I again can't see how a handful of extra bytes to serial could knock out 107 LEDs.

Of course, what I should have asked immediately is if you could also provide me (us) with captures of serial output from 1) a full boot cycle with the commit in, and 2) a full boot cycle with the commit out. Would that be possible?

kriztioan, 2024-01-23:
@rbergen The intrigue deepens. When hitting `Monitor` in `pio` with the 'broken' build uploaded to the device the LEDs spring to life. I'm able to consistently reproduce this behavior. That is, _1._ turn the device ON via the power supply--LEDS are off; _2._ Plug in the USB cable--LEDs are still off; and _3._ hit `Monitor` in `pio`--LEDs spring to life. Thus far, I've been unable to catch the earlier serial data. My device has two USB-C ports and I will try the other. At the end of this comment is what I was able to catch thus far for the 'working' build.

@robertlipe My understanding of the inner workings of the code base, its knobs, and how and what each of the different libraries exactly does is not sufficient enough to assess whether "It really doesn't make sense". I'm simply providing a data point to consider to help out this amazing project. Perhaps when others encounter the same the signal-to-noise ratio will increase enough to get a clearer picture. I do really enjoy the _Aha!_ moments when these puzzles get solved, though.

Serial output 'working' build:
```verilog
[  1422][E][vfs_api.cpp:182] remove(): /improv.log does not exists or is directory
(I) (setup)(C1) Sending Improv packet to declare we're up. Ignore any IMPROV lines that follow this one.
>> Launching JSON Writer Thread.  Mem: 295808, LargestBlk: 278516, PSRAM Free: 8385939/8386231, (I) (LoadJSONFile)(C1) Attempting to read JSON file /device.cfg
(I) (DeviceConfig)(C1) Loading DeviceConfig from JSON
(W) (InitializeHardware)(C1) Allocating LEDStripGFX for channel 0
(I) (AddLEDsToFastLED)(C1) Adding LEDs to FastLED...
(I) (AddLEDsToFastLED)(C1) Adding 107 LEDs to pin 5 from channel 0 on FastLED.
(I) (SetupBufferManagers)(C1) Could allocate 20922 buffers but limiting it to 500
(I) 
(W) (SetupBufferManagers)(C1) Reserving 500 LED buffers for a total of 176500 bytes...
(W) (InitEffectsManager)(C1) InitEffectsManager...
(I) (LoadJSONFile)(C1) Attempting to read JSON file /effects.cfg
(W) (LoadJSONFile)(C1) Out of memory reading JSON from file /effects.cfg - increasing buffer to 7228 bytes
(W) (LoadJSONFile)(C1) Out of memory reading JSON from file /effects.cfg - increasing buffer to 9276 bytes
(W) (LoadJSONFile)(C1) Out of memory reading JSON from file /effects.cfg - increasing buffer to 11324 bytes
(I) (InitEffectsManager)(C1) Creating EffectManager from JSON config
(I) (ReadCurrentEffectIndex)(C1) Attempting to read file /current.cfg
>> Launching Drawing Thread.  Mem: 282136, LargestBlk: 270324, PSRAM Free: 8175899/8369495, (W) (DrawLoopTaskEntry)(C1) >> DrawLoopTaskEntry
(W) 
(W) (DrawLoopTaskEntry)(C1) Entering main draw loop!
(I) (setup)(C1) Making initial attempt to connect to WiFi.
(I) (ConnectToWiFi)(C1) WiFi credentials passed for SSID "C&C@centralpark"
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...
(W) (ConnectToWiFi)(C1) Connecting to Wifi SSID: "C&C@centralpark" - ESP32 Free Memory: 239544, PSRAM:8369351, PSRAM Free: 8164939
(W) 
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
>> Launching Network Thread.  Mem: 239496, LargestBlk: 229364, PSRAM Free: 8164939/8369351, >> Launching ColorData Thread.  Mem: 225336, LargestBlk: 217076, PSRAM Free: 8164939/8369351, (W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (SaveToJSONFile)(C0) Number of bytes written to JSON file /effects.cfg: 5180
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...
(W) (ConnectToWiFi)(C1) Connecting to Wifi SSID: "C&C@centralpark" - ESP32 Free Memory: 222384, PSRAM:8369351, PSRAM Free: 8164939
(W) 
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (loop)(C1) WiFi: WL_NO_SSID_AVAIL, IP: 0.0.0.0, Mem: 222128, LargestBlk: 212980, PSRAM Free: 8164939/8369351, LED FPS: 28 LED Bright:  24%, LED Watts: 3, CPU: 040%, 003%, FreeDraw: 0.030
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (ConnectToWiFi)(C1) Setting host name to NightDriverStrip...
(W) (ConnectToWiFi)(C1) Connecting to Wifi SSID: "C&C@centralpark" - ESP32 Free Memory: 221952, PSRAM:8369351, PSRAM Free: 8164939
(W) 
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (loop)(C1) WiFi: WL_NO_SSID_AVAIL, IP: 0.0.0.0, Mem: 221840, LargestBlk: 212980, PSRAM Free: 8164939/8369351, LED FPS: 30 LED Bright:  24%, LED Watts: 3, CPU: 000%, 000%, FreeDraw: 0.030
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(W) (ConnectToWiFi)(C1) Not yet connected to WiFi, waiting...
(I) (loop)(C1) Sending Improv packets to indicate WiFi is connected. Ignore any IMPROV lines that follow this one.
IMPROV�
IMPROVhttp://192.168.20.17
(W) (ConnectToWiFi)(C1) Connected to AP with BSSID: "A0:04:60:31:7D:E2", received IP: 192.168.20.17
(I) (ConnectToWiFi)(C1) Setting Clock...
[ 14961][E][WiFiUdp.cpp:221] parsePacket(): could not receive data: 9
(W) (ColorDataTaskEntry)(C1) Started color data server!
(W) (UpdateClockFromWeb)(C1) NTP clock: Raw values sec=828423580, usec=192882
(I) (UpdateClockFromWeb)(C1) Old Time: 1706024124.157608, New Time: 1706024125.192882, Delta: 1.035274
[... cut ...]
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.20.17, Mem: 193828, LargestBlk: 184308, PSRAM Free: 8163691/8369143, LED FPS: 30 LED Bright:  24%, LED Watts: 3, CPU: 000%, 000%, FreeDraw: 0.030
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.20.17, Mem: 193828, LargestBlk: 184308, PSRAM Free: 8163691/8369143, LED FPS: 30 LED Bright:  24%, LED Watts: 3, CPU: 000%, 000%, FreeDraw: 0.030
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.20.17, Mem: 193828, LargestBlk: 184308, PSRAM Free: 8163691/8369143, LED FPS: 30 LED Bright:  24%, LED Watts: 3, CPU: 000%, 000%, FreeDraw: 0.030
```

rbergen, 2024-01-24:
@kriztioan That is interesting. In theory, this could be caused by the IMPROV package write to serial blocking when nobody's listening. The write in the commit you reversed in your fork happens during setup, so _before_ the effects are activated. I'd still call this behavior weird, but I guess it's possible.

I'll think of a way to test this by moving the "offending" write from `ImprovSerial::setup()` to `ImprovSerial::loop()`. Can I open a PR on your fork to apply that change so you can test it? The problem is still not reproduceable for me. If yes, what branch shall I open it on?

kriztioan, 2024-01-24:
@rbergen I have no luck yet with catching the earlier serial output. Please use the [i596](https://github.com/kriztioan/NightDriverStrip/tree/i596) branch for the PR--I'll pull it in and give it a test.

rbergen, 2024-01-24:
@kriztioan Thanks. What's the config I should build to check things at least compile?

kriztioan, 2024-01-24:
> @kriztioan Thanks. What's the config I should build to check things at least compile?

kriztioan ;-)

rbergen, 2024-01-24:
Well, that's another lesson learned. I had checked out your i596 branch and the PlatformIO plugin in VSCode did update, but did _not_ pick up that added config. I had to restart VSCode for that to happen.  
I'll remember that one as well.

Anyway, the kriztioan config builds on my machine. I'll open the PR on your fork in the next few minutes.

rbergen, 2024-01-24:
@kriztioan I opened kriztioan#1 with an alternative approach to declaring device presence via Improv. Please let me know if this makes a functional difference for you.

kriztioan, 2024-01-24:
@rbergen That fixed it! Thank you for looking into this! I'll let you deal with closing the issue etc.

rbergen, 2024-01-25:
Glad to hear that! I'll open a PR on main to propagate this "hello!" approach to the main tree, and close this issue with the merge of that. Thank you for raising this, and finding the exact steps to unblock the start-up; in the end that's what led us to the resolution of this.

rbergen, 2024-01-26:
Chances are good that I'm missing something, but I don't understand what a timer on its own would solve? If you start a timer and then make a blocking call, you're still stuck. For that to work you'd have to do the offending call in a separate thread, and I really, really don't want to kick off another one of those if we don't have to - if only because of the few KB of heap-allocated stack each of them chews up.

I believe the watchdog effectively only kicks in if one of the idle threads (one per core) doesn't have enough CPU breathing room to reset the watchdog timer. If the block happens in a different thread in a way that doesn't involve a busy wait, then the watchdog will happily conclude that the CPU is in its happy place - which it is.

I don't think Improv is the thing to blame as such in this case. All my devices happily thunder through the `ImprovSerial::setup()` code with or without USB buffers being drained, and I'm sure we would have already heard it if this was a common thing. I think there's something about @kriztioan's specific hardware configuration that triggers this. Undoubtedly whatever it is is not unique, but unusual.  
...which also means I wouldn't currently know how to properly generalize this to a class of failures, let alone in a reproducible way.

rbergen, 2024-01-26:
> Uhm, no. The timer interrupt fires, preempts the current (presumably spin-waiting) task, and then has a chance to recover. If it's cancelled by loop, we carry on. If the timer actually fires, we know we never got to loop. Timers run asynchronously to the scheduler.

Right, so I did miss something. :) I missed the part where the timer kicks off some code that could act.

> So you were never able to actually repro this and you just got lucky (that sounds insulting, but you hopefully know what I mean) with a brute force fix?

I know what you mean, and I think me getting lucky is a fair way to put it. Because no, I was never able to reproduce this.

> Maybe we should leave Pandora's Box closed on this one after all...but this is the kind of thing that makes me crazy. I don't *like* unexplained mystery crashes.

Same here, but sometimes I choose to count my blessings and move on. Who knows, it may come back to haunt us in a more persistent and reproducible way at a later time. At which time you're totally entitled to pull the "I told you so" card. :)

rbergen, 2024-01-26:
> Still, this kind of thing bugs me...

Really? Man, you really know how to hide that well! 😂

PR #599 (2024-01-26): Move Improv announcement from setup to loop
## Description

This fixes #596, which documents a case where the sending of the Improv state packet to indicate we're up blocks setup, and hence the starting of the blinking of LEDs. It does this by moving the one-time sending of the packet from `ImprovSerial::setup()` to `ImprovSerial::loop()`.

## Contributing requirements

* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I  [... cut]
closed 2024-01-26 by rbergen