PlummersSoftwareLLC/NightDriverStrip #598: E (2148) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

fanno, 2024-01-23:
I appoligize in advance if this is not where i should be looking for the "support" / guidance on this

## Bug report
Hello i am getting this in the log/consol after following the guide.

is the esp32 broken?

i install with:
ESP32-S3-DevKitC-1 / Panlee

```
ELF file SHA256: 5b8c8f93c30f4511

E (2148) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403799af
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
E (274) esp_core_dump_flash: No core dump partition found!
E (274) esp_core_dump_flash: No core dump partition found!
E (276) psram: PSRAM ID read erro2'��Replacing Idle Tasks with TaskManager...
(I) (PrintOutputHeader)(C1) NightDriverStrip
(I) 
(I) (PrintOutputHeader)(C1) ------------------------------------------------------------------------------------------------------------
(I) (PrintOutputHeader)(C1) M5STICKC: 0, USE_M5DISPLAY: 0, USE_OLED: 0, USE_TFTSPI: 1, USE_LCD: 0, USE_AUDIO: 0, ENABLE_REMOTE: 0
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: FAIL
(I) (PrintOutputHeader)(C1) Version 40: Wifi SSID: "" - ESP32 Free Memory: 301308, PSRAM:0, PSRAM Free: 0
(I) (PrintOutputHeader)(C1) ESP32 Clock Freq : 240 MHz
(I) (setup)(C1) Startup!
(I) (setup)(C1) Starting DebugLoopTaskEntry
>> Launching Debug Thread.  Mem: 301308, LargestBlk: 270324, PSRAM Free: 0/0, (W) (ReadWiFiConfig)(C1) Retrieved SSID and Password from NVS: "", "********"
(W) (setup)(C1) Starting ImprovSerial for ESP32-S3
[   630][E][vfs_api.cpp:182] remove(): /improv.log does not exists or is directory
(I) (setup)(C1) Sending Improv packet to declare we're up. Ignore any IMPROV lines that follow this one.
IMPROV�
>> Launching JSON Writer Thread.  Mem: 292548, LargestBlk: 270324, PSRAM Free: 0/0, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (setup)(C1) Creating TFT Screen
[  1075][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1080) gpio: gpio_set_level(227): GPIO output gpio_num error
[  1087][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1092) gpio: gpio_set_level(227): GPIO output gpio_num error
E (1104) gpio: gpio_set_level(227): GPIO output gpio_num error
E (1125) gpio: gpio_set_level(227): GPIO output gpio_num error
(W) (InitializeHardware)(C1) Allocating LEDStripGFX for channel 0
(I) (AddLEDsToFastLED)(C1) Adding LEDs to FastLED...
(I) (AddLEDsToFastLED)(C1) Adding 1152 LEDs to pin 5 from channel 0 on FastLED.
(I) (SetupBufferManagers)(C1) Could allocate 1231068 buffers but limiting it to 500
(I) 
(W) (SetupBufferManagers)(C1) Reserving 500 LED buffers for a total of 1744000 bytes...
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420116ad  PS      : 0x00060430  A0      : 0x820117cc  A1      : 0x3fceba80  
A2      : 0x0000006d  A3      : 0x00000010  A4      : 0x00000000  A5      : 0x3fcf7a60  
A6      : 0x00000001  A7      : 0x3fcf7a90  A8      : 0x820116a6  A9      : 0x3fceba60  
A10     : 0x00000000  A11     : 0x3fcdfb64  A12     : 0x3fcc3eec  A13     : 0x00000400  
A14     : 0x3fcdfa04  A15     : 0x3fcdfbfc  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0x420116aa:0x3fceba80 0x420117c9:0x3fcebac0 0x42011839:0x3fcebaf0 0x42011ac9:0x3fcebb20 0x420127fb:0x3fcebb50 0x42029932:0x3fcebc10
```

**Problem**

**Steps**
i have only tried with 1 ESP32 i dont wanna break others incase this is breaking the ESP
1.  using https://plummerssoftwarellc.github.io/NightDriverStrip/
2.  chose type ESP32-S3-DevKitC-1
3.  chose Panlee
4. cant get the wifi config to pop up (its posible i am doing some thing wrong)

I eraced the device when i installed the first time ? was that correct ?

**Example**
before going here i loaded a simple webserver on the ESP

**Notes**
This is my first time trying to get in to this project after getting the hardware.

smiliea, 2024-01-24:
I believe you are out of memory. Try reducing the amount of visualizations
and features. I seriously doubt your ESP-32 is broken.

fanno, 2024-01-24:
Hello both of you for replying

Thanks for taking your time to reply with feedback

@smiliea 
as to memory, its not something i "add" i followed the webinstaller guide but perhaps the board is just not supported properly due to some of the things that @robertlipe maybe brought up.

@robertlipe 
Currently nothing is hooped up to the dev board. (i am aware the dev board it self hooks things up to pins)
I know 1 pin is properly the on board neo pixel or what ever its called. pehaps that is it.

the board is a:
TZT ESP32-S3-DevKitC-1 ESP32-S3 WiFi Bluetooth-compatible BLE 5.0 Mesh Development Board ESP32 Wireless Module N16R8

that is why on:
https://plummerssoftwarellc.github.io/NightDriverStrip/
i chose : ESP32-S3-DevKitC-1 / Panlee (only option)

If something NEED to be hooked up that is fine. and i will need to read further, i just started guide and tried the web installer.

Other than trying to use the web installer i have not done anything. (started from the top of the README)

I figured web-installer would be a good start off point. Perhaps that is not a option with the board i got..and i need to dig a little more. and combine and such my self that is eventually my goal i just wanted to do "baby" steps.

in case i missed something i will comment on the points further:
1. I am not sure what this means. If you think its relevant to this.
I will have to read on it more.  but form complete read it sounds like 2, 4 and 5 may be more of interest ? as its web installed and no config change ?

2. Could be because the dev board has something hooped up to a pin that cause a conflict?
2a. i have no comment on this one, other that i tried to look a bit at the link you provided (did not read it all yet) other than the psram parts i could find there

3. if its something i "need" to do i would have to build my self i would assume since i used web-installer.

4. I am clueless about this one,

5. I did not my self attempt to "add" the led's but here is where i guess perhaps my board is simply not supported by the web-installer ? or something and i need to work building my self or what not. ?

Thanks again for your time and long and impressive reply.

fanno, 2024-01-24:
Hello

I have managed to compile in vscode took some learning but thats the fun part.

I read a bit on the PSRAM after trying to disable it it is working.

in [env:panlee]
build_flags
i added:
                  -DENABLE_WIFI=1
                  -DENABLE_WEBSERVER=1

and removed:
${psram_flags.build_flags}

and that made it all compile and run (no actual strip connected yet)

> For modules that include Octal PSRAM (any module that has 8MB PSRAM) you MUST NOT use GPIO35, GPIO36 or GPIO37.

in  [env:panlee] i do not see any of this pins used

i got a magnifuing glass my eyes are not what they used to be on the cover plade of the esp32 it says:
ESP32-S3-WROVER-I
as i said earlier the webside listing was:
TZT ESP32-S3-DevKitC-1 ESP32-S3 WiFi Bluetooth-compatible BLE 5.0 Mesh Development Board ESP32 Wireless Module N16R8

from googling ESP32-WROVER-I (NO S3) i can find models with psram

is it possible that this model just do NOT have psram ?

if so did i buy the "wrong" model ? is PSRAM recommended?

I can se HEAP and DMA are both around 95 %

fanno, 2024-01-25:
Just a thought, since this is a option that pehaps is board dependend could a option on the webinstalled be

PSRAM / noPSRAM ? so that the option is there for somone who might have same issue as me that maybe dont have the "skills/will" to compile manually ect.

just a idea. its something i could look in to my self aswell i just have something else i wanna look in to first.

fanno, 2024-01-25:
i read it wrong it was ESP32-WROOM-I i open another from package and the text was more clear.

it looks like WROOM is without psram ? is that correct ?

fanno, 2024-01-25:
I see your points, and again thanks for your time.

I am a bit confused. just realize i get this in the. compiler: (both CDS and in a new test project i made (from google sources))
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.5.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)

source code:
```
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  delay(5000);
  log_d("Total heap: %d", ESP.getHeapSize());
  log_d("Free heap: %d", ESP.getFreeHeap());
  log_d("Total PSRAM: %d", ESP.getPsramSize());
  log_d("Free PSRAM: %d", ESP.getFreePsram());
}
```

test1  config:
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
              -DBOARD_HAS_PSRAM
              -mfix-esp32-psram-cache-issue
```
test1  result:
```
E (84) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
[    83][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
[  5108][D][main.cpp:9] loop(): Total heap: 398468
[  5109][D][main.cpp:10] loop(): Free heap: 373192
[  5109][D][main.cpp:11] loop(): Total PSRAM: 0
[  5111][D][main.cpp:12] loop(): Free PSRAM: 0
```

test 2 config:
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
              -DBOARD_HAS_PSRAM
              -mfix-esp32-psram-cache-issue
board_build.arduino.memory_type = dio_opi      
```
test2  result:
```
[    85][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5104][D][main.cpp:9] loop(): Total heap: 396900
[  5104][D][main.cpp:10] loop(): Free heap: 371848
[  5104][D][main.cpp:11] loop(): Total PSRAM: 8386279
[  5107][D][main.cpp:12] loop(): Free PSRAM: 8386019
```

i have tried adding this to the NDS  but the webinterface atleast still show  NAN (fresh clone with nothing but secret.h and board_build.arduino.memory_type = dio_opi added to panlee)

Som i have comflicting info but that matches what you talk about in the previous post (about the eccosystem).

could the webreport NAN even tho its actuarly there? as my demo app suggest its there.

BOARD_HAS_PSRAM and USE_PSRAM appear to be on in the active config so i suspect is is on

what is strange tho is that the app no longer crashes when trying to use PSRAM before i would get a the error that started this.
i tried doing a fresh clone and just schose panlee that cause the issue earlier i cant recreate it anymore

can changes have be made to the esp32 that is not overwritten when you you send "firmware" ?

from the "clean buuld with only board_build.arduino.memory_type = dio_opi       added
>> Launching Drawing Thread.  Mem: 149204, LargestBlk: 139252, PSRAM Free: 0/0, (W) (DrawLoopTaskEntry)(C1) >> DrawLoopTaskEntry

Some images as "prof" and explain a bit. i hope.
[image]
[image]

Once again sorry if i am "spamming", and feel free to let me know if i should be posting some where else. (i am trying to learn)

fanno, 2024-01-26:
Thanks once again for taking the time to help i cant stress enough how much i appreciate it.

You said -DDBOARD_HAS_PSRAM, i assume its type and shuld be -DBOARD_HAS_PSRAM is this correct or should there be Dx2

JSON FILE/BOARD
when trying to use the board you provided.
I created the json file you providede inside (esp32-s3-devkitc-1-n16r8v.json)
c:\Users\MOHU\.platformio\platforms\espressif32\boards\

from reading https://docs.platformio.org/en/stable/platforms/creating_board.html
I am getting: Error: BoardConfig: Board is not defined

or is this folder for something else, or do i need to do something different with the file.

when trying it with the nightdriverstrip
```
[env:demo2]
extends         = dev_yd-esp32-s3
build_flags     = -DDEMO=1
                  -DENABLE_WIFI=1
                  -DBOARD_HAS_PSRAM 
                  -DUSE_PSRAM
                  ${dev_yd-esp32-s3.build_flags}
board_build.arduino.memory_type = dio_opi          
```

A bit more experimentation on the tiny demo app i made, i changed the code with 1 line to test:
```
void loop() {
  delay(5000);
  log_d("Total heap: %d", ESP.getHeapSize());
  log_d("Free heap: %d", ESP.getFreeHeap());
  log_d("Total PSRAM: %d", ESP.getPsramSize());
  log_d("Free PSRAM: %d", ESP.getFreePsram());

  byte* psdRamBuffer = (byte*)ps_malloc(1000);

}
```
result:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    85][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5104][D][main.cpp:9] loop(): Total heap: 396900
[  5104][D][main.cpp:10] loop(): Free heap: 371848
[  5104][D][main.cpp:11] loop(): Total PSRAM: 8386279
[  5107][D][main.cpp:12] loop(): Free PSRAM: 8386019 <<<<<<<<<<<<
[ 10112][D][main.cpp:9] loop(): Total heap: 396900
[ 10112][D][main.cpp:10] loop(): Free heap: 371848
[ 10112][D][main.cpp:11] loop(): Total PSRAM: 8386263
[ 10115][D][main.cpp:12] loop(): Free PSRAM: 8385003 <<<<<<<<<<<<
[ 15119][D][main.cpp:9] loop(): Total heap: 396900
[ 15119][D][main.cpp:10] loop(): Free heap: 371848
[ 15119][D][main.cpp:11] loop(): Total PSRAM: 8386247
[ 15122][D][main.cpp:12] loop(): Free PSRAM: 8383987 <<<<<<<<<<<<
```
I added the ps_malloc, and i can see everytime it loops it Free PSRAM goes down by a bit over 1000 (i assume the overhead above 1000 has to do with referance data/pointer or something to the data.)

so maybe i am confused or asking the wrong question. if it appear its working in the tiny app i made why would the board_build.arduino.memory_type = dio_opi  not work with say enc:demo when adding -DBOARD_HAS_PSRAM,- DUSE_PSRAM

you also asked about the nastygram from init is there anything special you need for that ? created demo2 below
```
[... 81 lines cut ...]
(I) (begin)(C1) Embedded timezones file size: 18359
(I) (begin)(C1) Connecting Web Endpoints
(I) (begin)(C1) HTTP server started
(I) (ConnectToWiFi)(C1) Web Server begin called!
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 36084, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 21 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 007%, 019%, FreeDraw: 0.020
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 19 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 000%, 002%, FreeDraw: 0.009
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 003%, FreeDraw: 0.016
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 36748, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 19 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 002%, FreeDraw: 0.030
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 18 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 002%, FreeDraw: 0.009
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 22 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 002%, FreeDraw: 0.009
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 23 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 000%, 002%, FreeDraw: 0.015
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 19 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 002%, FreeDraw: 0.006
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 1
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 19 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 000%, 002%, FreeDraw: 0.016
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 20 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 002%, FreeDraw: 0.031
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 15 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 001%, 001%, FreeDraw: 0.030
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.87.123, Mem: 37556, LargestBlk: 26100, PSRAM Free: 0/0, LED FPS: 15 LED Bright: 100%, LED Watts: 30, Buffer: 0/37, CPU: 000%, 002%, FreeDraw: 0.012
```

I am sure i have not addressed everything in your previous message. i figured i would put this out as i go throw everytihng in your reply further.

-Morten

fanno, 2024-01-26:
Ok i think i founed something. (maybe)

in main.cpp -> void PrintOutputHeader()
```
    #if USE_PSRAM
        debugI("ESP32 PSRAM Init: %s", psramInit() ? "OK" : "FAIL");
    #endif
```

when i edit the file and have selected the env:demo2 slightly modifyed config off to and added PSRAM as show in previous message
ESP32 PSRAM Init: ...... do not appear in the log so USE_PSRAM must be 0 ?

This would be why the crash do not happen anymore as it dont try to use it. ? and why its showing 0/0?

How can this be ? i was under the impressin such things cant be redefined "runtime" or could there be something else at foot i am missing.

fanno, 2024-01-26:
Now i am really confused.

i added: (in PrintOutputHeader)
```
    debugI("TestTestTest"); <<<<<<<<<<<<<<<<

    #if USE_PSRAM
        debugI("ESP32 PSRAM Init: %s", psramInit() ? "OK" : "FAIL");
    #endif
```
this also do not appear. i also tried changing some of the exsisting ones. 

Can i make changes without them getting compiled ? i have tried clean and build then upload no chance.

this also explains some how why the i am not able to recreate the issue i guess.

fanno, 2024-01-27:
I did test "demo" earlier (a while ago), so if you by /dev/.... refere to a image fine on the esp or something then yes i guess it could be there

I did try to run the clean and i also but i did try delete content of build and build_cache
I not tried rename the whole .pio

The command was/is: (for the demo2 of cause.)
Executing task: C:\Users\MOHU\.platformio\penv\Scripts\platformio.exe run --target upload --environment demo2

I Did look in to "esptool erase flash" but then i stumbel on ability to:
Hold reset 3 sec > power on > hole for another 5 sec > release (resulting in a reset)
BuT in looking back its obvious that it did not work.

I ran: pio run --target erase --environment demo2
then i recompiled and it was updated.

!!! SUCCESS SUCCESS SUCCESS SUCCESS !!!
```
[env:demo2]
extends         = dev_esp32-s3
build_flags     = -DDEMO=1
                  -DCORE_DEBUG_LEVEL=5
                  ${dev_yd-esp32-s3.build_flags}
                  ${psram_flags.build_flags}
```
results in:
```
(I) (PrintOutputHeader)(C1) TestTestTest <<<<<<<<<<<<<<<
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: FAIL
...
>> Launching JSON Writer Thread.  Mem: 342240, LargestBlk: 311284, PSRAM Free: 0/0, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
```

test 2:
```
[env:demo2]
extends         = dev_esp32-s3
build_flags     = -DDEMO=1
                  -DCORE_DEBUG_LEVEL=5
                  ${dev_yd-esp32-s3.build_flags}
                  ${psram_flags.build_flags}
board_build.arduino.memory_type = dio_opi <<< only change
```
results in:
```
(I) (PrintOutputHeader)(C1) TestTestTest
(I) (PrintOutputHeader)(C1) ESP32 PSRAM Init: OK <<<<<<<<<<<<<<<
...
>> Launching JSON Writer Thread.  Mem: 341080, LargestBlk: 311284, PSRAM Free: 8385827/8386231, (W) (DeviceConfig)(C1) DeviceConfig could not be loaded from JSON, using defaults
```

conclusion so far:
board_build.arduino.memory_type = dio_opi
Would most likely have been the only thing i really needed to add. (with ofcause ${psram_flags.build_flags})

as to cause i am a bit curious maybe you have some insights.
I did Play around with OTA flash at some point. i wonder if that could be a "cause" for needing the erase.. 

I could/would close the issue now, but i would like you ask you one more thing if i may.

Ofcause if using OTA you cant erase the or it would remove the OTA feature. (because its some how part of the existing image on the device in running in the Loop)
But in deve face where the esp32 is not installed "in the addict" or anything like that.
Is there a way to set it up so it always do the erase first. 

Now to the question.
To at least ensure this wont ever happen again in a dev face. or is your experiance that this in general should not happen.
It was a fun tho of cause also frustrating issue and wasted your time. (tho i have to say i enjoy a challenge)

Is there a way i can make it always erase when preforming a upload again i am aware i could properly not do this when using OTA uploads.

But when i am using USB it would be nice i don't have to do it manually now that i know it "could" be needed i kinda just always wanna do it so i don't do something similar again (lesson learned).

If that is not the case, is there a way to "auto" increment a build "number" that i could then put in the to debug so i could with that verify if its old or new. (some kind of sanity check)

Thanks again for you patience
I also thank you for putting up with my "spam" and my bad spelling i am dyslectic and then English is not my first language

Iceing on the cake:
[image]
(after adding webserver)

P.S.
I hope i can in the near future fine something to contribute to the project.

fanno, 2024-01-27:
You are correct i am in my happy place.

Lesson learned i just wont use OTA unless i am putting the device up in the addict. (or some other place its hard to get to)

closed 2024-01-27 by fanno

rbergen, 2024-01-27:
@fanno I'm only commenting on this typo because it's particularly unfortunate, but I'm sure you mean "up in the attic".

fanno, 2024-01-28:
lol i had to look it up i see how thats a problem, thanks and yes that is what i mean.