PlummersSoftwareLLC/NightDriverStrip #600: Multiple WiFi options with the same name

Crazysnakepop, 2024-01-26:
[image]

Issue I'm experiencing is multiple wifi SSID names showing up when WiFi configuration window is displayed.  Additionally, no choices selected work with the correct password.  

I've tried multiple boards including the HelTec HTIT-WB32 and the NODEMCU ESP-32S.  Both say they program seccuessfully using the web 
[esp-web-tools-logs.txt](https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/14069291/esp-web-tools-logs.txt)

I do not have any guest networks turned on for my router but I'm also seeing multiple networks for my neighbors as well.  I've attached the log which shows it's trying to connect but each time it fails, I select the next Rodeman network in the list and try again.  To ensure I have the password correct, I disconnected my laptop from the home network and forgot the network so it would prompt me for the login/password again.  The laptop connects without any issue.

## Bug report

**Problem**

**Steps**

1.  
2.  
3.  

**Example**

**Notes**

rbergen, 2024-02-09:
@Crazysnakepop The WiFi network list passed to the Web Installer is the exact same list returned by the WiFi library we use (the one embedded in [Arduino ESP-32](https://github.com/espressif/arduino-esp32)), which gets it from the WiFi chip on your device. From our perspective, this is a "black box".

At the Web Installer side, the whole WiFi connectivity flow - uncluding the part that shows the list - is part of a dependency we use for that ([ESP Web Tools](https://esphome.github.io/esp-web-tools/)). Which again is something we can't "look into".

With that I don't mean to say that this behavior isn't problematic, but it's also caused somewhere in one of two black boxes from our codebase's perspective. 

What the logging you provided shows is that the network layer that _is_ part of our codebase concludes that no valid set of WiFi credentials are passed, which means the SSID, the password or both are empty.

From a practical perspective you could try manually entering the WiFi SSID instead of selecting one from the list (the manual entry option should be at the bottom of the list of SSIDs you're looking at).

To see which "black box" (the WiFi stack or the Web Installer component) causes this, you could enable Improv logging for your configuration. That would allow us to see exactly what your devices get back from the WiFi stack, and also what credentials get passed down from the Web Installer.  
This would require you to set up a local build environment (install PlatformIO, download or clone the source, etc.), make a small code modification to whatever configuration you're using and then build and flash that custom image.  
I'm happy to provide further pointers for the code modification, but I think it makes sense for you to first get to the point that your local build environment is set up and confirmed working - if you're so inclined, of course.

rbergen, 2024-07-17:
@Crazysnakepop Could you retry with the current version of the installer?

rbergen, 2024-07-27:
Closing this for lack of response from OP. Can be reopened if deemed appropriate at any time.

closed 2024-07-27 by rbergen