PlummersSoftwareLLC/NightDriverStrip #609: Incoming wifi data stops drawing after a few minutes

woodward54, 2024-02-08:
## Bug report

**Problem**

I am sending led data over wifi from Unity. It will work great for 2-3 minutes then hang up. Through some debugging, I found that every time effectmanager.h NextEffect() is called, all incoming wifi data stops drawing on the pixels. Im not sure if the call to NextEffect() is the problem, or just a downstream effect of no data being received. The socket remains connected according to unity. If I restart unity (re-establishing the socket), the pixels start rendering again properly. 

Do I possibly have something misconfigured? 

**Steps**

1.  
2.  
3.  

**Example**

Here is my custom setup

    #ifndef PROJECT_NAME
    #define PROJECT_NAME            "Led Ceiling"
    #endif

    #define MATRIX_WIDTH            240
    #define MATRIX_HEIGHT           1
    #define NUM_LEDS                (MATRIX_WIDTH*MATRIX_HEIGHT)
    #define NUM_CHANNELS            1
    #define LED_PIN0                15
    

    #define ENABLE_WIFI             1   // Connect to WiFi
    #define WAIT_FOR_WIFI           1   // Hold in setup until we have WiFi - for strips without effects
    #define TIME_BEFORE_LOCAL       2   // How many seconds before the lamp times out and shows local content
    #define ENABLE_WEBSERVER        0   // Turn on the internal webserver
    #define ENABLE_NTP              1   // Set the clock from the web
    #define ENABLE_OTA              1   // Accept over the air flash updates
    #define ENABLE_REMOTE           0
    #define ENABLE_AUDIO            0

    #define INCOMING_WIFI_ENABLED       1   // Accepting incoming color data and commands
    #define COLORDATA_SERVER_ENABLED    0
    #define MIN_BUFFERS                 1
    #define MAX_BUFFERS                 30

    #define POWER_LIMIT_MW              (1 * 5 * 1000)         // 10A x 5V (50W)

    #define DEFAULT_EFFECT_INTERVAL 0

**Notes**

Here are the packets being sent (192.168.1.172 = esp, 192.168.1.67 = Unity). Right when the LEDs stop drawing is when I start seeing all the red ZeroWindow errors. 
[image]

woodward54, 2024-02-09:
Sometimes, Ill get this message when it stops drawing.

(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 56080, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 20/132, CPU: 005%, 005%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 56088, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 24/132, CPU: 005%, 004%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 52920, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 14/132, CPU: 004%, 000%, FreeDraw: 1.000
(W) (ReadUntilNBytesReceived)(C1) ERROR: -1 bytes read in ReadUntilNBytesReceived trying to read 5
(W) 
(W) (ProcessIncomingConnectionsLoop)(C1) Read error in getting header.
(W) 
(W) (SocketServerTaskEntry)(C1) Socket connection closed.  Retrying...
(W) 
(I)

Sometimes it will stop drawing with no error message. (See below buffers drop to 0)

(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 59152, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 25/132, CPU: 003%, 004%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 55984, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 23/132, CPU: 008%, 005%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 59152, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 27/132, CPU: 006%, 006%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 52832, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 27/132, CPU: 007%, 005%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 44792, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 24 LED Bright: 100%, LED Watts: 1, Buffer: 0/132, CPU: 002%, 000%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 44792, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 62 LED Bright: 100%, LED Watts: 1, Buffer: 0/132, CPU: 002%, 001%, FreeDraw: 0.009
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 44516, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 62 LED Bright: 100%, LED Watts: 1, Buffer: 0/132, CPU: 001%, 001%, FreeDraw: 0.009

woodward54, 2024-02-09:
Update: I pulled a fresh copy of NightDriverStrip, using the base LEDSTRIP setup and seeing the same issue. It will work fine for about 2-minutes, then about to the default local effect. 

Is anyone having issues sending data over wifi?

rbergen, 2024-02-09:
Not as such (I don't tend to do that myself), but I do notice something when looking at the second log snippet you included in your previous message: the buffers dropping to 0 correlates with free RAM dropping from the 55 to 59K range all the way down to 44K. As documented in multiple issues and discussions, we've learned that at that level of free RAM things start breaking, with networking services suffering first/most. We've recently disabled certain networking services in certain configurations for exactly this reason. (We then reenabled a subset of those later, when it turned out that (a) newer version(s) of some core dependencies were less memory-hungry than the ones before.)

What causes this memory drop to happen in your case I can't say looking at just the logging you provided.

rbergen, 2024-02-09:
> Hard to say if that's cause or effect. If you receive more data, you'll have more buffers in use and your free RAM will drop. But if  free ram dropped because ... gremlins, then we're surely into that mystery space Rutger wisely describes.

That's a good point. I based my cause/effect conclusion on the fact that OP states the problem starts after about two minutes. **If** the buffer count has been stable at ~25 for that long, I'm not sure why they would suddenly consume another 10-15K. But yeah, I did start that sentence with "if".

woodward54, 2024-02-09:
I did some more experiments 

Tried disabling NTP: frames became very choppy, pixels still stopped drawing after a few minutes
Same with disabling OTA

Changing the buffer to 24 reduced my framerate, and used about the same about of memory as 30 (see below, hovering around 130-140k consistently). 

Heres a video for reference (in this video MAX_BUFFER = 30, smooth fps and mem stays above 120k). Green pallet effect is the local fallback effect, unity is sending the blue square with red spinner. Unity is still showing the socket as connected. 

Cutout happens at 1:56, then it reverts to the local effect. 

[image]

Here are the logs from that video (note the cutout when the buffers suddenly drop to 0
`(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 139360, LargestBlk: 94196, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 13/30, CPU: 003%, 002%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 142400, LargestBlk: 102388, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 23/30, CPU: 005%, 003%, FreeDraw: 0.000
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 139376, LargestBlk: 98292, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 21/30, CPU: 007%, 003%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 139400, LargestBlk: 98292, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 13/30, CPU: 005%, 003%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 139396, LargestBlk: 94196, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 11/30, CPU: 004%, 003%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 139396, LargestBlk: 98292, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 15/30, CPU: 006%, 004%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 132728, LargestBlk: 90100, PSRAM Free: 0/0, LED FPS: 24 LED Bright:  52%, LED Watts: 19, Buffer: 0/30, CPU: 002%, 002%, FreeDraw: 0.001
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 128360, LargestBlk: 86004, PSRAM Free: 0/0, LED FPS: 4 LED Bright: 100%, LED Watts: 6, Buffer: 0/30, CPU: 001%, 000%, FreeDraw: 0.025
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 127872, LargestBlk: 86004, PSRAM Free: 0/0, LED FPS: 30 LED Bright: 100%, LED Watts: 6, Buffer: 0/30, CPU: 001%, 003%, FreeDraw: 0.025
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 127872, LargestBlk: 86004, PSRAM Free: 0/0, LED FPS: 30 LED Bright:  98%, LED Watts: 10, Buffer: 0/30, CPU: 001%, 003%, FreeDraw: 0.025
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 127872, LargestBlk: 86004, PSRAM Free: 0/0, LED FPS: 30 LED Bright: 100%, LED Watts: 6, Buffer: 0/30, CPU: 002%, 003%, FreeDraw: 0.025
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.175, Mem: 127592, LargestBlk: 86004, PSRAM Free: 0/0, LED FPS: 30 LED Bright: 100%, LED Watts: 6, Buffer: 0/30, CPU: 001%, 003%, FreeDraw: 0.025`

woodward54, 2024-02-09:
As for getting a board with more PSRAM, I ordered the 8MB to test.

However, this is for a project and we already have 50 esp-WROOM-32s on hand we'd like to use (no PSRAM to my knowledge). Reordering 50 more 8MB boards would add another few hundred dollars to our project. Would be ideal if we can use the esps we already have on hand.

It _seems_ like from the above test, memory is not the issue as its not dipping below 100k.

woodward54, 2024-02-10:
Using [ESP32-WROOM-32](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf) board

4MB RAM, 0 PSRAM, ESP32-D0WDQ6 chip

woodward54, 2024-02-10:
[This](https://www.alibaba.com/product-detail/Esp32-Esp32esp32-Esp32-Development-Board-Wifi_1600682238074.html?spm=a2700.galleryofferlist.p_offer.d_image.418d5b1due2qRL&s=p) is the exact board we ordered. I don't see any 8 pin PSRAM chips soldered on the physical boards.

woodward54, 2024-02-10:
[image]

woodward54, 2024-02-10:
Ill try to do some more detailed debugging tomorrow. 

How can I enable the debugV logs to print?

rbergen, 2024-02-10:
Me being from the past I'm late responding, but for the record:

> It _seems_ like from the above test, memory is not the issue as its not dipping below 100k.

I concur. Thanks for the work towards establishing that.

Looking at the new logging lines of buffers dropping to 0, I do notice that around the time they do, memory usage again increases with ~10K. This being the second time we see this, we may be seeing some sort of correlation between the two conditions at least.

woodward54, 2024-02-10:
Alright figured it out!

Turns out Unity was not programmed to accept the response packet from the esp32 (my guess is the unity project is outdated). The TCP response buffer was filling up and shutting down. The answer was in the wireshark dump all along [TCP Window Full]

I set bSendResponsePacket = false (socketserver.cpp:208) to disable response and that fixed the issue.

If I get around to it, Ill implement the response packet on the Unity side and make a PR

closed 2024-02-10 by woodward54

woodward54, 2024-02-12:
Might be a good idea to add a global var SEND_RESPONSE so this feature can be enabled/disabled down the road. But for now Im happy and moving on.