PlummersSoftwareLLC/NightDriverStrip #618: New M5stickC plus2 wont work?

platini76, 2024-04-08:
I tried to upload project as old M5stickc plus and did not work,
don't reach wifi config. someone has experience ?

rbergen, 2024-04-09:
@platini76 Personally, I have no experience with the M5StickC Plus2. However, to narrow the field of investigation into what it is you're running into, it would be helpful if you could provide answers to the following questions:

1. Do I understand correctly that you are using the WebInstaller to upload the image?
2. What project/configuration are you trying to flash? I know the device you mentioned is the M5StickC Plus, but I'm not yet sure which of the projects that are available for that device.
3. Do you see any activity on the device's built-in LCD screen after the upload?

rbergen, 2024-04-09:
As far as I'm aware, you **must** select a project before you can try to flash. This is the relevant section on the WebInstaller page:

[image]

As you can see, you have to select a device in the first drop-down - I understand from your first message that you selected "M5StickC Plus" there, as shown in the picture. You then have to select a project in the second drop-down; if you don't the CONNECT button you have to push to start the flash doesn't even show underneath. So, my question is what project you selected there when you got stuck.

I can't really recommend a project without knowing what it is you are trying to do - i.e. what you are flashing the your M5StickC Plus2 for.

platini76, 2024-04-09:
I select m5stick plus... it flashed... correctly, and when arrivate to next page... it say again ..... flash... dont recognize any flashed devices for wifi config

rbergen, 2024-04-09:
Can you maybe provide screen shots of what you're seeing?

platini76, 2024-04-09:
sure!

[image]
[image]
[image]
[image]
[image]
[image]

rbergen, 2024-04-09:
Thank you, that's very helpful. Simply put, the last screenshot shows me that the M5StickC Plus2 isn't booting the image you flashed - or even trying to. I think this means the M5StickC Plus2 cannot be made to work using an image for a "plain" M5StickC Plus. Which means we'll have to wait for someone to add support for the Plus2 to the project.

platini76, 2024-04-09:
thank you... hope someone will do it quickly...

rbergen, 2024-04-10:
@platini76 I've done some more Googling about this, and it seems that the first hurdle is adding support for the M5StickC Plus2 to PlatformIO. There is an issue on their espressif32 project repo requesting exactly that: platformio/platform-espressif32#1346.

rbergen, 2024-07-08:
@platini76 It turns out that @davepl found a way to add support for the Plus2 to NightDriverStrip without requiring "official" support in PlatformIO, and made the non-trivial effort required to apply the necessary changes to the codebase.

He discusses this in #641. As he also mentions, he has now added this support to his fork (at <https://github.com/davepl/NightDriverStrip>); you can pull and test the Plus2 support from there for now.

A merge of the relevant code into the main tree here will follow at a later time.

PR #642 (2024-07-16): Add M5 Unified support (M5 Stick C Plus 2, M5 Stack Core 2)
## Description
Moves all M5-based projects to the unified M5 stack

Fixes #582. Fixes #618.

* [X] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [X] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [X] I selected `main` as the target branch.
* [X] All code herein is subjected to the license terms in [COPYING.txt](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/COPYING.txt).
closed 2024-07-16 by rbergen