PlummersSoftwareLLC/NightDriverStrip #624: Async WebServer failure to start begin()

JimmyRoasts, 2024-04-24:
## Bug report

Thanks very much to whoever takes the time to read this. I've been trying to get the ESP32 webserver to work and I'm running into issues. Any advice, tips, things to try are greatly appreciated. Also whatever is wrong is likely something I did, I'm just trying to work out how exactly I can put it right.

**Problem**

Hi, the bug is in the demo environment, uploading for ESP32 microcontroller. (ESP-WROOM32). It seems that, (based on the debug output from the microcontroller) it is failing to start the AsyncTCP function. I'm guessing failure to start this service causes the webserver to not function correctly which results in not being able to access the webserver. The debug output is as follows:

> 
> (I) (begin)(C1) Web UI URL pathnames enabled
> **[  9082][E][AsyncTCP.cpp:1251] begin(): failed to start task**
> (I) (begin)(C1) HTTP server started
> (I) (ConnectToWiFi)(C1) Web Server begin called!
> (I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.54, Mem: 53864, LargestBlk: 36852, PSRAM Free: 0/0, LED FPS: 30 LED Bright:  31%, LED Watts: 3, CPU: 002%, 001%, FreeDraw: 0.019

performing an nmap on 192.168.1.54 yields the following:

> Nmap scan report for 192.168.1.54
> Host is up (0.0044s latency).
> Not shown: 998 closed ports
> PORT      STATE SERVICE
> 23/tcp    open  telnet
> 12000/tcp open  cce4x

I can see that the webserver port (80) has not been opened. Navigating to the webserver also fails. The built in site also has a 'connection refused' error when trying to used the rest API.

globals.h is defined as:

```
    #define PROJECT_NAME            "Demo"
    #endif

    #define MATRIX_WIDTH            120
    #define MATRIX_HEIGHT           1
    #define NUM_LEDS                (MATRIX_WIDTH*MATRIX_HEIGHT)
    #define NUM_CHANNELS            1
    #define ENABLE_AUDIO            0

    // Once you have a working project, selectively enable various additional features by setting
    // them to 1 in the list below.  This DEMO config assumes no audio (mic), or screen, etc.

    #ifndef ENABLE_WIFI
        #define ENABLE_WIFI             1   // Connect to WiFi
    #endif

    #define INCOMING_WIFI_ENABLED   0   // Accepting incoming color data and commands
    #define TIME_BEFORE_LOCAL       0   // How many seconds before the lamp times out and shows local content
    #define ENABLE_NTP              0   // Set the clock from the web
    #define ENABLE_OTA              0   // Accept over the air flash updates

    #if M5STICKC || M5STICKCPLUS || M5STACKCORE2
        #define LED_PIN0 32
    #elif LILYGOTDISPLAYS3
        #define LED_PIN0 21
    #else
        #define LED_PIN0 5
    #endif

    // The webserver serves files that are baked into the device firmware. When running you should be able to
    // see/select the list of effects by visiting the chip's IP in a browser.  You can get the chip's IP by
    // watching the serial output or checking your router for the DHCP given to a new device; often they're
    // named "esp32-" followed by a seemingly random 6-digit hexadecimal number.

    #ifndef ENABLE_WEBSERVER
        #define ENABLE_WEBSERVER        1   // Turn on the internal webserver
    #endif
```

espaddr.jsx is defined as:

> 
> const httpPrefix = process.env.NODE_ENV === "development" ? "http://192.168.1.54" : undefined;
> export default httpPrefix;

**Steps**

1.  build for environment "demo"
2.  upload to esp32 microcontroller
3.  attempt to connect over port 80 to access webserver

**Example**

I'm also getting an error on CMakeLists.txt for the ESP Async WebServer component.

[image]

**Notes**

No additional notes. Just in case Dave has a look, just wanted to say. I read your book as I've recently been diagnosed myself (35). I found it to be one of the most useful books I've ever read. We share a lot of the same experiences, good and bad. That issue you had with Microsofts legal team was particularly poignant for me as I had similar legal issues and handled them just about the same. Now I've read the book, I'll be better prepared. If you haven't read the book and think you might be autistic, I highly reccomend. Please let me know if you need additional information and it will be cheerfly provided. Again, thank you for taking time to read through this.

davepl, 2024-04-24:
I think your issue might be as simple as the DEMO project does not include the web server!  On line 298 or so of globals.h you’ll see:

    #ifndef ENABLE_WEBSERVER
        #define ENABLE_WEBSERVER        0   // Turn on the internal webserver
    #endif

So it explicity disables it, since the other definition only kicks in if its NOT defined.

Change that to a 1, I bet it works!

 -Dave

closed 2024-04-24 by JimmyRoasts

davepl, 2024-04-24:
> Thanks Dave, for your quick response and for the building the project.
>
> I've had a look and I think I have it enabled correctly as in your post.
>
> image.png (view on web) <https://github.com/PlummersSoftwareLLC/NightDriverStrip/assets/51733768/46e2fad2-c872-41ee-8c59-f49b12d4cac2>
> I noticed on line 1142, there is also an ENABLE _WEBSERVER and I've set that 1 as well.
>
> image.png (view on web) <https://github.com/PlummersSoftwareLLC/NightDriverStrip/assets/51733768/06a87986-ca41-43f0-b22a-428c3120ead1>
> I get the following output when connecting over serial:
>
> (W) (ConnectToWiFi)(C1) Connected to AP with BSSID: "20:C9:D0:1C:A4:73", received IP: 192.168.1.54
> (I) (ConnectToWiFi)(C1) Starting Web Server...
> (I) (begin)(C1) Embedded html file size: 507
> (I) (begin)(C1) Embedded jsx file size: 15320
> (I) (begin)(C1) Embedded ico file size: 6471
> (I) (begin)(C1) Embedded timezones file size: 18359
> (I) (begin)(C1) Connecting Web Endpoints
> (I) (begin)(C1) Web UI URL pathnames enabled

With 36K free you’re right on the ragged edge of stuff not working.  One approach would be to set the MAX_BUFFERS to a smaller numberm or if you don’t need to control 1140 LEDs, set NUM_LEDS to a smaller value.  That will save a lof of RAM.

- Dave

JimmyRoasts, 2024-04-24:
The ragged edge is an all too familar place and better sometimes to just pay the $10 and play well away from it. I've got a little bit fixated with this project recently, and that does sound like an interesting exercise. I have a 120 LED strip, so nowhere close to 1140, so I'll definitely try that and report back if it works. Also I'll let it run for a bit and see if it crashes. Thanks so much Dave, not just for the help, but for all the tips on living on the spectrum too, it's seriously really helped.

JimmyRoasts, 2024-04-24:
To anyone who experience similar. I did get this working with the following configuration for demo (located in globals.h)

```
#elif DEMO

    // This is a simple demo configuration.  To build, simply connect the data lead from a WS2812B
    // strip to pin 5 or other pin marked PIN0 below.  This does not use the OLED, LCD, or anything fancy, it simply drives the
    // LEDs with a simple rainbow effect as specified in effects.cpp for DEMO.
    //
    // Please ensure you supply sufficent power to your strip, as even the DEMO of 144 LEDs, if set
    // to white, would overload a USB port.
    #ifndef PROJECT_NAME
    #define PROJECT_NAME            "Demo"
    #endif

    #define MATRIX_WIDTH            120
    #define MATRIX_HEIGHT           1
    #define NUM_LEDS                (MATRIX_WIDTH*MATRIX_HEIGHT)
    #define NUM_CHANNELS            1
    #define ENABLE_AUDIO            0

    // Once you have a working project, selectively enable various additional features by setting
    // them to 1 in the list below.  This DEMO config assumes no audio (mic), or screen, etc.

    #ifndef ENABLE_WIFI
        #define ENABLE_WIFI             1   // Connect to WiFi
    #endif

    #define INCOMING_WIFI_ENABLED   1   // Accepting incoming color data and commands
    #define TIME_BEFORE_LOCAL       0   // How many seconds before the lamp times out and shows local content
    #define ENABLE_NTP              1   // Set the clock from the web
    #define ENABLE_OTA              1   // Accept over the air flash updates
    #define USE_PSRAM               0   // Use PSRAM if available

    #if M5STICKC || M5STICKCPLUS || M5STACKCORE2
        #define LED_PIN0 32
    #elif LILYGOTDISPLAYS3
        #define LED_PIN0 21
    #else
        #define LED_PIN0 5
    #endif

    // The webserver serves files that are baked into the device firmware. When running you should be able to
    // see/select the list of effects by visiting the chip's IP in a browser.  You can get the chip's IP by
    // watching the serial output or checking your router for the DHCP given to a new device; often they're
    // named "esp32-" followed by a seemingly random 6-digit hexadecimal number.

    #define ENABLE_WEBSERVER        1   // Turn on the internal webserver

    #define MIN_BUFFERS             120
    #define MAX_BUFFERS             150

```
The solution is to add a line to define the BUFFERS. It does work (the TCP process now starts, the port opens and I can access it through a browser. I can also verify that the nightdriver website can connect to the ESP.

[image]

davepl, 2024-04-24:
Let us know if reducing the LED count solves it!

JimmyRoasts, 2024-04-25:
Hi Dave, 

Sorry, I should have been more clear in my previous post. I can confirm that reducing the LED count works. I've been running the code for about 24 hours without issue and I added a handful of effects for testing. The fire effects are so cool!

[image]

JimmyRoasts, 2024-04-27:
Wow there are knee slappers in there, @robertlipe, man's got jokes for days! You're dead right at the strip, its a 2m, 120 LED strip. So far, I've got a lot of the effects working just by copying at pasting out of the other builds. Every now and then I jump in and just see if I can get one upload and see what it does. My son (14 months), loves "bouncing balls" effect. He loves pushing buttons, so I had an idea to whack them in a 3D printed cube or something and have hooked to a few buttons so he can cycle the effects. Ha! the old 12v on the rail by accident trick has stung me before! Took me a sec to realise what had happened and it was like taking a penalty at hockey. Shame for 2 minutes, then - get free. The board is still running from a few days ago. I might have a tinker then upload the configuration somewhere in case anyone just wants a "kitchen sink" build for one of these low end chips. Fire effects, palette, rainbowfill, colorcycle all work beautifully with no modification.

davepl, 2024-04-27:
Glad you enjoyed the book!  Check out my draft of the sequel at nonvisiblepartofthespectrum.com <http://nonvisiblepartofthespectrum.com/>, and shoot me any feedback please!

- Dave